riscv-collab/riscv-gnu-toolchain #1636: Unable to use llvm-bolt to instrument RISC-V program

JunGaos, 2024-12-10:
The compilation command I used is: 
`riscv64-unknown-linux-gnu-gcc` program.c -Wl,--emit-relocs -fno-reorder-blocks-and-partition -static -o program `
program.c is a simple program.
Next, when I use llvm-bolt for instrumentation, the command I used is: 
`llvm-bolt program -instrument -o instrumented_program -instrumentation-file=without_pgo.fdata -instrumentation-sleep-time=2`
the following error message appears:
Log:
`BOLT-INFO: Target architecture: riscv64
BOLT-INFO: BOLT version: a4bf6cd7cfb1a1421ba92bca9d017b49936c55e4
BOLT-INFO: first alloc address is 0x10000
BOLT-INFO: creating new program header table at address 0x210000, offset 0x200000
BOLT-WARNING: debug info will be stripped from the binary. Use -update-debug-sections to keep it.
BOLT-INFO: enabling relocation mode
BOLT-INFO: static input executable detected
BOLT-INFO: forcing -jump-tables=move for instrumentation
BOLT-WARNING: Failed to analyze 37 relocations
BOLT-INFO: 0 out of 887 functions in the binary (0.0%) have non-empty execution profile
not implemented
UNREACHABLE executed at /home/gaojun/diskWD/llvm-project/bolt/include/bolt/Core/MCPlusBuilder.h:2001!
 #0 0x00005556dc8e2edf llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xbe4edf)
 #1 0x00005556dc8e0714 SignalHandler(int) Signals.cpp:0:0
 #2 0x00007f4bec171420 __restore_rt (/lib/x86_64-linux-gnu/libpthread.so.0+0x14420)
 #3 0x00007f4bebc3e00b raise /build/glibc-LcI20x/glibc-2.31/signal/../sysdeps/unix/sysv/linux/raise.c:51:1
 #4 0x00007f4bebc1d859 abort /build/glibc-LcI20x/glibc-2.31/stdlib/abort.c:81:7
 #5 0x00005556dc87dace (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xb7face)
 #6 0x00005556dd0a9c62 (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0x13abc62)
 #7 0x00005556dcfe3e4e llvm::bolt::Instrumentation::createAuxiliaryFunctions(llvm::bolt::BinaryContext&) (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0x12e5e4e)
 #8 0x00005556dcfe4b62 llvm::bolt::Instrumentation::runOnFunctions(llvm::bolt::BinaryContext&) (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0x12e6b62)
 #9 0x00005556dc9c7ec4 llvm::bolt::BinaryFunctionPassManager::runPasses() (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xcc9ec4)
#10 0x00005556dc9cb1fd llvm::bolt::BinaryFunctionPassManager::runAllPasses(llvm::bolt::BinaryContext&) (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xccd1fd)
#11 0x00005556dc96711d llvm::bolt::RewriteInstance::runOptimizationPasses() (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xc6911d)
#12 0x00005556dc9c4cb4 llvm::bolt::RewriteInstance::run() (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0xcc6cb4)
#13 0x00005556dc01e69d main (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0x32069d)
#14 0x00007f4bebc1f083 __libc_start_main /build/glibc-LcI20x/glibc-2.31/csu/../csu/libc-start.c:342:3
#15 0x00005556dc0b77de _start (/home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt+0x3b97de)
PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.
Stack dump:
0.  Program arguments: /home/gaojun/diskWD/llvm-project/build/bin/llvm-bolt program -instrument -o instrumented_program -instrumentation-file=without_pgo.fdata -instrumentation-sleep-time=2
Aborted (core dumped)`
Are there issues with my compilation options?

TommyMurphyTM1234, 2024-12-10:
> `PLEASE submit a bug report to https://github.com/llvm/llvm-project/issues/ and include the crash backtrace.`

This is what you need to do. Report the issue upstream.

JunGaos, 2024-12-10:
Alright, I have already raised my issue in the llvm-project, but perhaps someone here might also help me solve the problem?

TommyMurphyTM1234, 2024-12-10:
llvm-bolt is crashing so the appropriate place to seek assistance is upstream in the LLVM project. This isn't a `riscv-gnu-toolchain` issue per se.

JunGaos, 2024-12-10:
OK, I'll close this issue.

closed 2024-12-10 by JunGaos