riscv-collab/riscv-gnu-toolchain #1637: Failed to build in container with no error message

yoaveshel, 2024-12-10:
I'm trying build the `riscv32-unknown-elf-gcc` toolchain in a podman container. Here is the ContainerFile:
```
FROM ubuntu:latest

RUN apt-get update && apt-get install -y \
    build-essential \
    curl \
    git \
    autoconf \
    automake \
    autotools-dev \
    python3 \
    libmpc-dev \
    libmpfr-dev \
    libgmp-dev \
    gawk \
    bison \
    flex \
    texinfo \
    gperf \
    libtool \
    patchutils \
    bc \
    zlib1g-dev \
    libexpat-dev \
    ninja-build

ENV CARGO_HOME /root/.cargo 
RUN curl https://sh.rustup.rs -sSf | sh -s -- -y
ENV PATH="/root/.cargo/bin:${PATH}"
RUN rustup target add riscv32i-unknown-none-elf
RUN cargo install cargo-binutils && rustup component add llvm-tools-preview

WORKDIR /app

RUN git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
ENV PATH="/opt/riscv/bin:${PATH}"
RUN (cd riscv-gnu-toolchain; ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d)
RUN (cd riscv-gnu-toolchain; make linux)
```
But the last step fails for no apparent reason. The last few lines of the output are:
```
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-1.o -MT insn-emit-1.o -MMD -MP -MF ./.deps/insn-emit-1.TPo insn-emit-1.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-2.o -MT insn-emit-2.o -MMD -MP -MF ./.deps/insn-emit-2.TPo insn-emit-2.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-3.o -MT insn-emit-3.o -MMD -MP -MF ./.deps/insn-emit-3.TPo insn-emit-3.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-4.o -MT insn-emit-4.o -MMD -MP -MF ./.deps/insn-emit-4.TPo insn-emit-4.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-5.o -MT insn-emit-5.o -MMD -MP -MF ./.deps/insn-emit-5.TPo insn-emit-5.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-6.o -MT insn-emit-6.o -MMD -MP -MF ./.deps/insn-emit-6.TPo insn-emit-6.cc
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-7.o -MT insn-emit-7.o -MMD -MP -MF ./.deps/insn-emit-7.TPo insn-emit-7.cc
g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [Makefile:1198: insn-emit-7.o] Error 1
make[2]: Leaving directory '/app/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[1]: *** [Makefile:4673: all-gcc] Error 2
make[1]: Leaving directory '/app/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:472: stamps/build-gcc-linux-stage1] Error 2
```

TommyMurphyTM1234, 2024-12-10:
> I'm trying build the `riscv32-unknown-elf-gcc` toolchain in a podman container. 

In that case why are you building the Linux toolchain?

> ```
> RUN (cd riscv-gnu-toolchain; ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d)
> RUN (cd riscv-gnu-toolchain; make linux)
> ```

I'm not familiar with Podman - can you provide full details on how to reproduce the build in such a container, ideally on Ubuntu?

What is this stuff for? 

> ```
> ENV CARGO_HOME /root/.cargo 
> RUN curl https://sh.rustup.rs -sSf | sh -s -- -y
> ENV PATH="/root/.cargo/bin:${PATH}"
> RUN rustup target add riscv32i-unknown-none-elf
> RUN cargo install cargo-binutils && rustup component add llvm-tools-preview
> ```

The CI builds are performed successfully using GitHub containers so containerised builds definitely can be made to work. 

- https://github.com/riscv-collab/riscv-gnu-toolchain/tree/master/.github/workflows

Are you sure that your container isn't running out of resources such as memory or disk space perhaps? 

> ```
> g++: fatal error: Killed signal terminated program cc1plus
> compilation terminated.
> make[2]: *** [Makefile:1198: insn-emit-7.o] Error 1
> make[2]: Leaving directory [...]

Maybe try giving your container more resources?

yoaveshel, 2024-12-10:
> In that case why are you building the Linux toolchain?
I was testing to see if it gives the same error and forgot to remove before posting here. Thanks!

> I'm not familiar with Podman - can you provide full details on how to reproduce the build in such a container, ideally on Ubuntu?
Sure. To install you run `apt-get -y install podman` and to build the container you can run `podman build -t my_container .` from the same directory as the ContainerFile.

> What is this stuff for?
Installing rust. Shouldn't be relevant for my question

> Are you sure that your container isn't running out of resources such as memory or disk space perhaps?
I'm pretty sure that's not the case. AFAIK podman doesn't limit memory on containers by default. You can add a (4 GB) limit by running `podman build --memory 4g -t my_container .` which gives the same output. Usually when a container runs out of memory it gives an OOM error

TommyMurphyTM1234, 2024-12-10:
> I'm pretty sure that's not the case. AFAIK podman doesn't limit memory on containers by default. You can add a (4 GB) limit by running podman build --memory 4g -t my_container . which gives the same output. Usually when a container runs out of memory it gives an OOM error

What about disk space? 

Can you run Podman containers interactively like you can with Docker as that may help with diagnosing the failure?

yoaveshel, 2024-12-10:
> What about disk space?

I think it is also not limited by default

>Can you run Podman containers interactively like you can with Docker as that may help with diagnosing the failure?

Yes. Podman is supposed to be CLI compatible with docker (some people say that transitioning from docker is as simple as `alias docker=podman`) so you can run `podman run -it my_container /bin/bash`

TommyMurphyTM1234, 2024-12-10:
> Yes. Podman is supposed to be CLI compatible with docker (some people say that transitioning from docker is as simple as `alias docker=podman`) so you can run `podman run -it my_container /bin/bash`

You should do this so and see if you can check things like running out of disk space etc.

cmuellner, 2024-12-10:
I believe opening a second terminal window and typing `journalctl -f` will tell you more.

TommyMurphyTM1234, 2024-12-10:
I can't reproduce the problem.

On Ubuntu 24.04 running on WSL2 on Windows 10:
```
# On host
sudo apt install podman
podman -it ubuntu:latest

# In Container
apt update
apt upgrade
apt install git

git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

.github/setup-apt.sh

./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
make 2>&1 | tee build.log

df -H
Filesystem      Size  Used Avail Use% Mounted on
overlay         1.1T  162G  865G  16% /
tmpfs            68M     0   68M   0% /dev
tmpfs           829M  6.1M  823M   1% /etc/hosts
shm              66M     0   66M   0% /dev/shm
tmpfs           4.2G     0  4.2G   0% /sys/fs/cgroup
tmpfs           4.2G     0  4.2G   0% /proc/acpi
tmpfs           4.2G     0  4.2G   0% /sys/firmware
tmpfs           4.2G     0  4.2G   0% /sys/dev/block
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/18079246/build.log.zip)

TommyMurphyTM1234, 2024-12-11:
Any update @yoaveshel?

TommyMurphyTM1234, 2024-12-13:
No update and no reproducible test case/scenario so I am closing this issue.

closed 2024-12-13 by TommyMurphyTM1234