riscv-collab/riscv-gnu-toolchain #1638: toolchain + spike + pk run rvv program error

LittleWaters, 2024-12-10:
Recently I am learning about the vector instruction set of RISCV. Then i found that it can run vector through spike + pk, So i do the following work.

1.  Build riscv-gnu-toolchain
```c
git clone https://github.com/riscv-collab/riscv-gnu-toolchain.git
git submodule update --init --recursive
cd riscv-gnu-toolchain
mkdir build
cd build
../configure --prefix=/riscv-gnu-toolchain/build --with-arch=rv64gcv --with-abi=lp64d
make -j4
``` 

2. Add /riscv-gnu-toolchain/build into PATH
```c
sudo gedit ~/.bashrc
export PATH=$PATH:/riscv-gnu-toolchain/build/bin
source ~/.bashrc
``` 

3. Build spike
```c
git clone https://github.com/riscv-software-src/riscv-isa-sim.git
cd riscv-isa-sim
mkdir build
cd build
../configure --prefix=/riscv-isa-sim/build
make -j4
``` 

4. Build pk
```c
git clone https://github.com/riscv-software-src/riscv-pk.git
cd riscv-pk
mkdir build
cd build
../configure --prefix=/riscv-pk/build --host=riscv64-unknown-elf
cp pk /riscv-isa-sim/build
``` 

5. Compile rvv test program
```c
touch hello.c

#include <stdio.h>
int main(void)
{
    printf("hello, world!\n");

    return 0;
}

riscv64-unknown-elf-gcc -o hello hello.c

``` 

6. Run spike + pk + hello
```c
./spike --isa=rv64gcv pk hello
(printf nothing)
``` 

Finally I found out， When i build riscv-gnu-toolchian, i use
```c
../configure --prefix=/riscv-gnu-toolchain/build --with-arch=rv64gc --with-abi=lp64d
``` 
remove vector, and use this toolchian build spike + pk, at the same time, build rvv test program(hello) use toolchain with **rv64gcv**, Then run
```c
./spike --isa=rv64gcv pk hello
hello, world!
``` 
It can run.

So, can tell me  this fo why?

best regards!

TommyMurphyTM1234, 2024-12-10:
I can't reproduce your problem:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1638
cd issue-1638
./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d --with-sim=spike
make
make build-sim

cd installed-tools/bin

./riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1638/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/issue-1638/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1638/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1638/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 ()

./spike 
./spike --help
Spike RISC-V ISA Simulator 1.1.1-dev
...

# Create test.c

// test.c
#include <stdio.h>
int main(void)
{
    printf("hello, world!\n");

    return 0;
}

./riscv64-unknown-elf-gcc -o test test.c

./spike --isa=rv64gcv pk64 test
hello, world!
```

LittleWaters, 2024-12-11:
> I can't reproduce your problem:
> 
> ```
> # Clone, configure and build riscv-gnu-toolchain (for rv64gcv/lp64d) and Spike
> 
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1638
> cd issue-1638
> ./configure --prefix=`pwd`/installed-tools --with-arch=rv64gcv --with-abi=lp64d --with-sim=spike
> make
> make build-sim
> 
> # Check toolchain and Spike version info
> 
> cd installed-tools/bin
> 
> ./riscv64-unknown-elf-gcc -v
> Using built-in specs.
> COLLECT_GCC=./riscv64-unknown-elf-gcc
> COLLECT_LTO_WRAPPER=/home/user/issue-1638/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
> Target: riscv64-unknown-elf
> Configured with: /home/user/issue-1638/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1638/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1638/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gcv --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
> Thread model: single
> Supported LTO compression algorithms: zlib
> gcc version 14.2.0 ()
> 
> ./spike --help
> Spike RISC-V ISA Simulator 1.1.1-dev
> ...
> 
> # Create test.c
> 
> // test.c
> #include <stdio.h>
> int main(void)
> {
>     printf("hello, world!\n");
> 
>     return 0;
> }
> 
> # Compile test.c
> 
> ./riscv64-unknown-elf-gcc -o test test.c
> 
> # Run test using Spike
> 
> ./spike --isa=rv64gcv pk64 test
> hello, world!
> ```

Thanks for your reply!
I just tried following the steps you posted，it works！
And i think there's some difference, like when i compile spike and pk , i clone spike and pk git address, not use riscv-gnu-toolchian  option to build it, is this the difference? if use my steps when compile pk finish it get "pk", while use riscv-gnu-toolchain option, it get "pk64".
However I am still grateful for your suggestion. I can use spike and pk directly according to your suggestion.

TommyMurphyTM1234, 2024-12-11:
> Thanks for your reply! I just tried following the steps you posted，it works！ 

You're welcome. Glad it's working for you now.

> And i think there's some difference, like when i compile spike and pk , i clone spike and pk git address, not use riscv-gnu-toolchian option to build it, is this the difference? if use my steps when [...]

I don't know about the differences but the way that I'm doing it is letting the top level `riscv-gnu-toolchain` `Makefile` build everything which is the recommended way for using this repo. If you build certain components such as Spike/pk separately/manually then you may get different results.

Since there is no issue with the `riscv-gnu-toolchain` approach I'll close this issue.

closed 2024-12-11 by TommyMurphyTM1234

LittleWaters, 2024-12-11:
> You're welcome. Glad it's working for you now.
>
>
> I don't know about the differences but the way that I'm doing it is letting the top level `riscv-gnu-toolchain` `Makefile` build everything which is [...]

Yeah, I see the riscv-gnu-toolchian Makefile, When build pk, it use --with-arch=rv64gc, So the problem maybe is when build pk, Should use rv64gc.
Thanks！

TommyMurphyTM1234, 2024-12-11:
I don't know how exactly this differs from building pk manually but the top-level `riscv-gnu-toolchain` `Makefile` builds 32 and/or 64 bit versions of pk depending on the toolchain/multilib configuration:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/7b5adbdc5cc84cda61086a5e504005a901fb474b/Makefile.in#L228-L239
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/7b5adbdc5cc84cda61086a5e504005a901fb474b/Makefile.in#L1095-L1119

and then, when using Spike with pk, you use the appropriate version for your target RISC-V XLEN.

TommyMurphyTM1234, 2024-12-11:
This worked fine for me:
```
cd 
pwd
/home/user

# Clone, configure and build toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1638
cd issue-1638
./configure --prefix=/home/user/issue-1638/installed-tools --with-multilib-generator="rv64gcv-lp64d--"
make

# Clone, configure and build Spike
git submodule init spike
git submodule update spike
cd spike 
./configure --prefix=/home/user/issue-1638/installed-tools
make
make install

# Clone, configure and build pk
cd ..
git submodule init pk
git submodule update pk
cd pk
mkdir build
cd build
export PATH=/home/user/issue-1638/installed-tools/bin:$PATH
../configure --prefix=/home/user/issue-1638/installed-tools --host=riscv64-unknown-elf
make
make install

# Create test.c
cd /home/user/issue-1638/installed-tools/bin
vi test.c

> #include <stdio.h>
> int main(void)
> {
>     printf("hello, world!\n");
> 
>     return 0;
> }

# Compile and run test for rv64gc/lp64d
 ./riscv64-unknown-elf-gcc test.c -o test
spike pk test
hello, world!

# Compile and run test for rv64gcv/lp64d
./riscv64-unknown-elf-gcc test.c -o test -march=rv64gcv -mabi=lp64d
./spike --isa=rv64gcv pk test
hello, world!
```