riscv-collab/riscv-gnu-toolchain #1640: automatic parallelization in LLVM?

aaaaala, 2024-12-11:
Are there compilation options in Clang/LLVM for automatic parallelization, similar to GCC's `-ftree-parallelize-loops=` option, which automatically splits simple loops into different threads?

Additionally, how can we find more compilation options in Clang/LLVM, such as options like -mllvm -force-vector-width, which do not appear in the `clang -help` output

TommyMurphyTM1234, 2024-12-11:
You need to ask these questions upstream in the LLVM/Clang project. 

- https://llvm.org/
  - https://discourse.llvm.org/
  - https://discord.com/invite/xS7Z362
  - ...

TommyMurphyTM1234, 2024-12-12:
Closing since this is an LLVM/Clang and not a `riscv-gnu-toolchain` issue/question.

closed 2024-12-12 by TommyMurphyTM1234