riscv-collab/riscv-gnu-toolchain #1641: make error

Thomas-J-Kidd, 2024-12-12:
Hello I am getting an error when I run

```bash
./configure --prefix ./
make
```
```bash
g++  -fno-PIE -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H -fno-PIE -I. -I. -I../.././gcc/gcc -I../.././gcc/gcc/. -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o insn-emit-10.o -MT insn-emit-10.o -MMD -MP -MF ./.deps/insn-emit-10.TPo insn-emit-10.cc
/tmp/ccYJd6Fo.s: Assembler messages:
/tmp/ccYJd6Fo.s: Internal error (Segmentation fault).
Please report this bug.
double free or corruption (out)
/tmp/ccYJd6Fo.s: Internal error (Aborted).
Please report this bug.
make[2]: *** [Makefile:1198: insn-emit-10.o] Error 1
make[2]: Leaving directory './riscv-gnu-toolchain/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4689: all-gcc] Error 2
make[1]: Leaving directory './riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:651: stamps/build-gcc-newlib-stage1] Error 2
```

TommyMurphyTM1234, 2024-12-12:
Please provide the full set of commands that you're using (`git clone`, `configure`, `make`), your full build log (zip and attach it rather than copying and pasting it if necessary), and details of the host on which you're trying to build. 

Does it make any difference if you use a different prefix directory? E.g. `./configure --prefix=`pwd`/installed-tools`.

TommyMurphyTM1234, 2024-12-12:
> ```
> ./configure --prefix ./
> ```

This doesn't work for me:
```
./configure --prefix ./
configure: error: expected an absolute directory name for --prefix: .
```
Neither does this:
```
./configure --prefix=./
configure: error: expected an absolute directory name for --prefix: .
```
Are you sure that you're accurately describing how you're configuring the toolchain?

TommyMurphyTM1234, 2024-12-13:
Any update @Thomas-J-Kidd?

If you're still having problems then please provide sufficient detail to allow the issue to be reproduced.

Thomas-J-Kidd, 2024-12-13:
Sorry I have been swamped with grading finals... 

Let me replicate my error and upload it! Thanks for all the messages though

TommyMurphyTM1234, 2024-12-16:
No update. Open again if necessary and if there's a reproducible test case.

closed 2024-12-16 by TommyMurphyTM1234