riscv-collab/riscv-gnu-toolchain #1642: Submodule fetch failure: binutils

Vikrius, 2024-12-15:
After trying to build using make, I encountered errors with each of the submodules within the riscv-gnu-toolchain repository. 

Example:
`Fetched in submodule path 'binutils', but it did not contain beb2cdbcda911764b2bed5e57921fe90493260bd. Direct fetching of that commit failed.`

Initially, I tried running:

`git pull --recurse-submodules`

Which worked for most of the submodules. However, binutils was still giving me the same error.

So, I then tried running:

`git submodule update --force --recursive --init --remote`

Which gave this output:

`Unable to find current origin/binutils-2_43 revision in submodule path 'binutils'`

I tried modifying the script so that it would use different branches from the sourceware repository, to no avail.

I would greatly appreciate any help in this matter.

Thanks!

TommyMurphyTM1234, 2024-12-15:
You'll need to provide the build log for a clean `clone`/`configure`/`make` - e.g.
```
cd ~/Downloads
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1642
cd issue-1642
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log
...
```
Please zip and upload the `build.log` rather than copying and pasting it.

Are you sure that the issue isn't down to some networking issue at your end?
If you try to clone the individual submodule repos and branches/tags manually do you have problems?

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.gitmodules

For what it's worth all of these worked fine for me and I cannot reproduce any problems with fetching any of the submodule repos/branches/tags...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1642 --recursive
Cloning into 'issue-1642'...
remote: Enumerating objects: 9501, done.
remote: Counting objects: 100% (495/495), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 9501 (delta 374), reused 399 (delta 334), pack-reused 9006 (from 1)
Receiving objects: 100% (9501/9501), 6.44 MiB | 4.65 MiB/s, done.
Resolving deltas: 100% (4734/4734), done.
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Submodule 'dejagnu' (https://git.savannah.gnu.org/git/dejagnu.git) registered for path 'dejagnu'
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Submodule 'llvm' (https://github.com/llvm/llvm-project.git) registered for path 'llvm'
Submodule 'musl' (https://git.musl-libc.org/git/musl) registered for path 'musl'
Submodule 'newlib' (https://sourceware.org/git/newlib-cygwin.git) registered for path 'newlib'
Submodule 'pk' (https://github.com/riscv-software-src/riscv-pk.git) registered for path 'pk'
Submodule 'qemu' (https://gitlab.com/qemu-project/qemu.git) registered for path 'qemu'
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Submodule 'uclibc-ng' (https://github.com/wbx-github/uclibc-ng.git) registered for path 'uclibc-ng'
Cloning into '/home/user/issue-1642/binutils'...
remote: Enumerating objects: 370302, done.
remote: Counting objects: 100% (370302/370302), done.
remote: Compressing objects: 100% (104742/104742), done.
remote: Total 370302 (delta 320407), reused 306115 (delta 262392), pack-reused 0
Receiving objects: 100% (370302/370302), 289.50 MiB | 4.65 MiB/s, done.
Resolving deltas: 100% (320407/320407), done.
Checking connectivity: 370302, done.
Cloning into '/home/user/issue-1642/dejagnu'...
...
remote: Enumerating objects: 16281, done.
remote: Counting objects: 100% (10669/10669), done.
remote: Compressing objects: 100% (2927/2927), done.
remote: Total 9138 (delta 6864), reused 8177 (delta 6073), pack-reused 0 (from 0)
Receiving objects: 100% (9138/9138), 1.80 MiB | 3.94 MiB/s, done.
Resolving deltas: 100% (6864/6864), completed with 501 local objects.
From https://github.com/riscv-software-src/riscv-isa-sim
 * branch              88fc84ded155a9e01987c4dfb7a77800e69b232b -> FETCH_HEAD
Submodule path 'spike': checked out '88fc84ded155a9e01987c4dfb7a77800e69b232b'
Submodule path 'uclibc-ng': checked out 'ecbc412d0d93c8b7b77bec6539b290c9da5401cf'
```
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1642
Cloning into 'issue-1642'...
remote: Enumerating objects: 9501, done.
remote: Counting objects: 100% (493/493), done.
remote: Compressing objects: 100% (161/161), done.
remote: Total 9501 (delta 373), reused 397 (delta 332), pack-reused 9008 (from 1)
Receiving objects: 100% (9501/9501), 6.46 MiB | 6.39 MiB/s, done.
Resolving deltas: 100% (4729/4729), done.
cd issue-1642/

[... 11 lines cut ...]
Submodule 'spike' (https://github.com/riscv-software-src/riscv-isa-sim.git) registered for path 'spike'
Submodule 'uclibc-ng' (https://github.com/wbx-github/uclibc-ng.git) registered for path 'uclibc-ng'
Cloning into '/home/user/issue-1642/binutils'...
Cloning into '/home/user/issue-1642/dejagnu'...
...
From https://gitlab.com/qemu-project/qemu
 * branch                  ae35f033b874c627d81d51070187fbf55f0bf1a7 -> FETCH_HEAD
Submodule path 'qemu': checked out 'ae35f033b874c627d81d51070187fbf55f0bf1a7'
remote: Enumerating objects: 22057, done.
remote: Counting objects: 100% (22056/22056), done.
remote: Compressing objects: 100% (6262/6262), done.
remote: Total 21204 (delta 14758), reused 20378 (delta 14048), pack-reused 0 (from 0)
Receiving objects: 100% (21204/21204), 5.26 MiB | 4.55 MiB/s, done.
Resolving deltas: 100% (14758/14758), completed with 410 local objects.
From https://github.com/riscv-software-src/riscv-isa-sim
 * branch              88fc84ded155a9e01987c4dfb7a77800e69b232b -> FETCH_HEAD
Submodule path 'spike': checked out '88fc84ded155a9e01987c4dfb7a77800e69b232b'
Submodule path 'uclibc-ng': checked out 'ecbc412d0d93c8b7b77bec6539b290c9da5401cf'
```
I'm also in the process of doing a build as follows and will post the build log when finished but so far there have been no problems with accessing/retrieving any of the submodule repos/branches/tags...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1642
make 2>&1 | tee build.log
...
```

TommyMurphyTM1234, 2024-12-15:
> I'm also in the process of doing a build as follows and will post the build log when finished but so far there have been no problems with accessing/retrieving any of the submodule [...]
This also worked fine.

- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/18141015/build.log.zip)

Vikrius, 2024-12-15:
Oddly enough, I think it may have been some networking issue on my end. I have no idea what I did differently from yesterday, but just as a sanity check I did a fresh pull of the repository and ran make, after which everything worked fine.

Sorry to bother, thank you for your time and patience in this matter.

closed 2024-12-15 by Vikrius