riscv-collab/riscv-gnu-toolchain #1645: Question: How to properly replace the libgcc with custom library functions

muxiff, 2024-12-19:
Hi,
I wrote a library function to calculate the multiplication of two uint64 integers under rv32，
```
typedef unsigned long long uint64_t ;

uint64_t mul_u64(uint64_t a, uint64_t b)
{
  uint64_t Product;
  Product = 0;
  while (a) {
    if (a & 1) {
      Product += b;
    }
    a >>= 1;
    b <<= 1;
  }
  return Product;
}
```
For compilation information, see https://godbolt.org/z/hM74MvYcs

At this piece of assembly code,
```
...
.L3:
        mv      a2,s3
        mv      a3,s1
        andi    a0,s0,1
        li      a1,0
        call    __muldi3
        add     a0,s5,a0
        sltu    a5,a0,s5
...
```
The compiler will always call `__muldi3` in libgcc, and I expect to keep calling `mul_u64` recursively, is there any way to do that?

Thanks

TommyMurphyTM1234, 2024-12-19:
> The compiler will always call `__muldi3` in libgcc, and I expect to keep calling `mul_u64` recursively

Why would you expect this to happen automatically?

> is there any way to do that?

If you mean that you want the toolchain to use your function instead of `__muldi3` then you'll probably need to modify libgcc and/or the toolchain itself. Or change your function name to `__muldi3` (if it has the appropriate prototype) and link that ahead of the libgcc version. 

This is not really a RISC-V or `riscv-gnu-toolchain` specific question/issue and how to do such toolchain/library modifications is way outside the scope of this repo. You should probably familiarise yourself with the GCC internals documentation and maybe ask about this in an appropriate upstream forum.

- https://gcc.gnu.org/onlinedocs/gccint/
- https://gcc.gnu.org/lists.html

TommyMurphyTM1234, 2024-12-21:
Closing since this isn't really a `riscv-gnu-toolchain` bug/issue.

closed 2024-12-21 by TommyMurphyTM1234