riscv-collab/riscv-gnu-toolchain #1646: --disable-debug-info does nothing

jefftrull, 2024-12-20:
I am building toolchains for use in Docker containers and I find that the `--disable-debug-info` configure option does nothing - debug info is always present in the toolchain binaries, causing my container images to be very large. For example:

```
/configure --disable-gdb --disable-debug-info --with-arch=rv32imc --with-abi=ilp32 --with-cmodel=medlow --prefix=`pwd`/installed
make
...
file installed/bin/*
```
shows all the binaries as `with debug_info, not stripped`

I can see that `--enable-debug-info` does add `-g` to the compile flags, but it seems like it must be there by default in gcc (at least) and what we need is some way of turning that off.

TommyMurphyTM1234, 2024-12-20:
Doesn't `--disable-debug-info` only apply to the standard library? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/43536acae8791de5fc93acad51d0c03dda9f903e/configure#L1383-L1384

Maybe you need to use something like `make CFLAGS_FOR_TARGET_EXTRA=-g0 ...`?

E.g. see here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1530#issue-2459536719

jefftrull, 2024-12-21:
That's a great point re: `--disable-debug-info`. However it does seem as if that too does not work. In the `lib` directory all the `.so` files are `with debug_info, not stripped`. There are some `.a` files as well and when I extract the `.o` files from them they also have debug symbols.

Looking at the configure code I think that this switch can only _add_  debug symbols, and it seems that the underlying repos (gcc etc.) already have symbols enabled by default somehow.

I will try your `-g0` suggestion.

TommyMurphyTM1234, 2024-12-21:
Probably worth looking at the build log to see what flags are being used. It'll be a day or two before I can try a build myself to check this out.

jefftrull, 2024-12-21:
Thanks for looking at this. I should be able to make some progress with these tips.

Would you mind clarifying about the extra variables? Judging by the fact that `CFLAGS_FOR_TARGET_EXTRA` gets combined with `-mcmodel=medlow` in the Makefile, it seems like it applies to the binaries generated by the toolchain, rather than the toolchain itself (which is at issue in this case). Am I confused? If not, is there a different variable I can use?

TommyMurphyTM1234, 2024-12-21:
If you want to affect the toolchain binaries themselves then maybe simply `make CFLAGS=-g0 ...` or similar? I'm AFK so can't really check at the moment unfortunately.

jefftrull, 2024-12-21:
That was the ticket (for the binaries!). Leaving this open for the libraries.

TommyMurphyTM1234, 2024-12-21:
> That was the ticket (for the binaries!). Leaving this open for the libraries.

Does a combination of `configure --disable-debug-info ...` and `make ASFLAGS_FOR_TARGET_EXTRA=-g0 CFLAGS_FOR_TARGET_EXTRA=-g0 CXXFLAGS_FOR_TARGET_EXTRA=-g0 ...` not eliminate debug info for the libraries?

TommyMurphyTM1234, 2024-12-21:
FWIW, as far as I can see the `riscv-gnu-toolchain` top-level `Makefile[.in]` always uses `make install` (or equivalent) rather than `make install-strip` (or equivalent where available) and there is no facility to opt for the stripped install. And, by default, in the absence of `configure --disable-debug-info ...` and/or `make ASFLAGS_FOR_TARGET_EXTRA=-g0 CFLAGS_FOR_TARGET_EXTRA=-g0 CXXFLAGS_FOR_TARGET_EXTRA=-g0 ...` and/or `make ASFLAGS_FOR_TARGET_EXTRA=-g0 CFLAGS_FOR_TARGET_EXTRA=-g0 CXXFLAGS_FOR_TARGET_EXTRA=-g0 ...` etc., everything is built (and installed) with debug information.

I presume that the issue for you @jefftrull is that you want the toolchain to be as lean as possible and not unnecessarily carrying any additional baggage such as debug info/symbols?

Just tagging @ilg-ul to ask if he does something different with his xPack build in order to produce a toolchain distro that bundles some or all components already stripped?

ilg-ul, 2024-12-21:
> Just tagging @ilg-ul to ask if he does something different with his xPack build in order to produce a toolchain distro that bundles some or all components already stripped?

Can you be more specific? There are a lot of configuration options, both for GCC and binutils.

TommyMurphyTM1234, 2024-12-21:
> Can you be more specific? There are a lot of configuration options, both for GCC and binutils.

Are some, most or all cross development host and RISC-V target binaries/libraries/object files in the xPack RISC-V GCC toolchain stripped (of debug information/symbols)? I just haven't had a chance to check.

ilg-ul, 2024-12-21:
I did not check either, but, as far as I remember, the xPack build scripts strip all native executables and libraries of debug information. There are options to prevent this for development builds.

As for the cross libraries... I don't know, but generally running debug sessions with the GCC runtime and newlib libraries is not possible.

jefftrull, 2024-12-21:
> > That was the ticket (for the binaries!). Leaving this open for the libraries.
> 
> Does a combination of `configure --disable-debug-info ...` and `make ASFLAGS_FOR_TARGET_EXTRA=-g0 CFLAGS_FOR_TARGET_EXTRA=-g0 CXXFLAGS_FOR_TARGET_EXTRA=-g0 ...` not eliminate debug info for the libraries?

Unfortunately not. `libgcc.a`, `libstdc++.a`, and `libc.a` all contain object files with debug info. I used the following command lines:

```
./configure --disable-gdb --disable-debug-info --with-arch=rv32imc --with-abi=ilp32 --with-cmodel=medlow --prefix=`pwd`/installed
GCC_EXTRA_CONFIGURE_FLAGS=--enable-target-optspace BINUTILS_TARGET_FLAGS_EXTRA=--enable-target-optspace NEWLIB_TARGET_FLAGS_EXTRA=--enable-target-optspace  CFLAGS=-g0 CXXFLAGS=-g0 ASFLAGS_FOR_TARGET_EXTRA=-g0 CFLAGS_FOR_TARGET_EXTRA=-g0 CXXFLAGS_FOR_TARGET_EXTRA=-g0 gmake -j4
```

It looks to me as though `--disable-debug-info` does nothing, because "disabled" is the default state. Some casual Googling has led me to believe that gcc etc. binaries have debug info _by default_.  If I'm right about this, and a good combination of options can be determined, perhaps `--disable-debug-info` can be made to select them for the user.

TommyMurphyTM1234, 2024-12-21:
@jefftrull, what is the problem with debug information in the libraries? In a bare-metal deployment scenario the debug info sections are never actually loaded on the target. They just exist in the ELF file on the development host to facilitate debugging. 

What is the exact problem that you are trying to address here? 

Other than the fact (it seems) that `--[dis|en]able-debug-info` may not be working as expected?

ilg-ul, 2024-12-21:
The xPack build scripts run lots of post-processing actions, including stripping debug infos.

jefftrull, 2024-12-21:
> @jefftrull, what is the problem with debug information in the libraries? In a bare-metal deployment scenario the debug info sections are never actually loaded on the target. They just exist in the [...]

I am creating Docker container images with the toolchain, for continuous integration and other purposes. The images produced at the moment are surprisingly large, consuming resources on the host machines as well as taking up time for data transfer. I also discovered that it increases build time (I assume for writing all this data to disk).

jefftrull, 2024-12-22:
I've made an interesting discovery.  `--disable-debug-info` _enables_ debug symbols. That is, it sets `enable_debug_info` to "no", which is considered a true value by the check [here](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/43536acae8791de5fc93acad51d0c03dda9f903e/configure#L3944).

So when I tried to remove debug symbols from the toolchain binaries by using `--disable-debug-info` it ended up enabling them in the libraries!

So maybe the proper procedure is:

1. leave `debug-info` alone - libraries will not get debug info
2. use `CFLAGS=-g0 CXXFLAGS=-g0` with make to remove debug info from binaries

ilg-ul, 2024-12-22:
> it sets enable_debug_info to "no", which is considered a true value by the check ...

If you identified a bug in the configure script, please report it to GCC bugzilla.

jefftrull, 2024-12-22:
That's the riscv-gnu-toolchain configure script :)

TommyMurphyTM1234, 2024-12-22:
> I've made an interesting discovery. `--disable-debug-info` _enables_ debug symbols. That is, it sets `enable_debug_info` to "no", which is considered a true value by the check [here](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/43536acae8791de5fc93acad51d0c03dda9f903e/configure#L3944).

I don't see any anomaly here:
```
./configure --prefix=`pwd`/installed-tools --disable-gdb
cp Makefile Makefile.default

./configure --prefix=`pwd`/installed-tools --disable-gdb --disable-debug-info

diff Makefile Makefile.default
18c18
< DEBUG_INFO := -g
---
> DEBUG_INFO :=

grep DEBUG_INFO Makefile
DEBUG_INFO := -g
CFLAGS_FOR_TARGET := $(CFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
CXXFLAGS_FOR_TARGET := $(CXXFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
ASFLAGS_FOR_TARGET := $(ASFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO) -mcmodel=medlow

grep DEBUG_INFO Makefile.default
DEBUG_INFO :=
CFLAGS_FOR_TARGET := $(CFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
CXXFLAGS_FOR_TARGET := $(CXXFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
ASFLAGS_FOR_TARGET := $(ASFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO) -mcmodel=medlow
```

TommyMurphyTM1234, 2024-12-22:
Looks to me like maybe the option `--disable-debug-info` was provided in the past but then this was changed to `--enable-debug-info` but this change left behind some configuration code that confuses the issue?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1305
- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1306

Maybe @kito-cheng, @cmuellner or @jeremybennett might know more about this?

TommyMurphyTM1234, 2024-12-22:
For now I think that you need to do this:
```
# Build libs without debug info
./configure --prefix=...

grep DEBUG_INFO Makefile
DEBUG_INFO :=
CFLAGS_FOR_TARGET := $(CFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
CXXFLAGS_FOR_TARGET := $(CXXFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
ASFLAGS_FOR_TARGET := $(ASFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO) -mcmodel=medlow

# Build libs with debug info
./configure --prefix=... --enable-debug-info

grep DEBUG_INFO Makefile
DEBUG_INFO := -g
CFLAGS_FOR_TARGET := $(CFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
CXXFLAGS_FOR_TARGET := $(CXXFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO)  -mcmodel=medlow
ASFLAGS_FOR_TARGET := $(ASFLAGS_FOR_TARGET_EXTRA) $(DEBUG_INFO) -mcmodel=medlow
```

TommyMurphyTM1234, 2024-12-22:
And to build the toolchain binaries (that run on the cross development host system) with/without debug info:
```
# Build toolchain binaries with debug info
./configure --prefix=...
make

# Build toolchain binaries without debug info?
./configure --prefix=...
make ASFLAGS=-g0 CFLAGS=-g0 CXXFLAGS=-g0
```

TommyMurphyTM1234, 2024-12-22:
FWIW...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1646
cd issue-1646

# Build 1: Toolchain host binaries with debug info/not stripped, target libs with debug info
./configure --prefix=`pwd`/installed-tools --enable-debug-info
make 
du -hd0 installed-tools
2.2G    installed-tools

# Build 2: Toolchain host binaries with debug info/not stripped, target libs without debug info
make distclean && rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools
make 
du -hd0 installed-tools
2.1G    installed-tools

# Build 3: Toolchain host binaries without debug info/not stripped, target libs without debug info
make distclean && rm -rf installed-tools
make ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0
du -hd0 installed-tools
493M    installed-tools

# Build 4: Toolchain host binaries without debug info/stripped, target libs without debug info
strip installed-tools/bin/*
strip: installed-tools/bin/riscv64-unknown-elf-gdb-add-index: file format not recognized
strip installed-tools/lib/*
strip: Warning: 'installed-tools/lib/bfd-plugins' is a directory
strip: Warning: 'installed-tools/lib/gcc' is a directory
strip: installed-tools/lib/libcc1.la: file format not recognized
du -hd0 installed-tools
467M    installed-tools
```

demin-han, 2025-04-28:
> FWIW...
>
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1646
> cd issue-1646
>
> # Build 1: Toolchain host binaries with debug info/not stripped, target libs with debug info [...]

Hi,
It can't work when use --enable-linux.
glibc can't compile.  libc-symbols.h:77:3: error: #error "glibc cannot be compiled without optimization"
make ASFLAGS=-g0 CFLAGS=-g0 CXXFLAGS=-g0 will override glibc CFLAGS.

any suggestion for this situation?

TommyMurphyTM1234, 2025-04-30:
> It can't work when use --enable-linux. 

I only tried it for the bare-metal Newlib toolchain.

> glibc can't compile. libc-symbols.h:77:3: error: #error "glibc cannot be compiled without optimization" make ASFLAGS=-g0 

From that message it seems that Glibc must be compiled with optimization enabled?

> any suggestion for this situation?

Figure out how to hack the build scripts/`Makefile[s]` to skip disabling optimisation for Glibc maybe?

demin-han, 2025-05-22:
> > It can't work when use --enable-linux.
> 
> I only tried it for the bare-metal Newlib toolchain.
> 
> > glibc can't compile. libc-symbols.h:77:3: error: #error "glibc cannot be compiled without optimization" make ASFLAGS=-g0
> 
> From that message it seems that Glibc must be compiled with optimization enabled?
> 
> > any suggestion for this situation?
> 
> Figure out how to hack the build scripts/`Makefile[s]` to skip disabling optimisation for Glibc maybe? Or build with Musl instad of Glibc perhaps?
> 
> * https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#installation-linux

I think the best sulutions is to replace "make install" with "make install-strip".
binutils and gcc both have this make target.

TommyMurphyTM1234, 2025-05-22:
> I think the best sulutions is to replace "make install" with "make install-strip".

That will strip debug symbolic information but will it compile everything with optimisation enabled?

demin-han, 2025-05-22:
> > I think the best sulutions is to replace "make install" with "make install-strip".
> 
> That will strip debug symbolic information but will it compile everything with optimisation enabled?

yes, the compile flags is "-g -O2".
"make install-strip" is post-process, I think it doesn't affect compiling process.

shenki, 2025-05-22:
I have some changes that change the build system to strip host binaries (gcc, qemu) by default:

https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1724

demin-han, 2025-07-19:
#1743  can solve this issue with --enable-strip configure option.

demin-han, 2025-07-22:
> > [#1743](https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1743) can solve this issue with --enable-strip configure option.
> 
> Thanks [@demin-han](https://github.com/demin-han) - but can you explain this more clearly please? I presume that the referenced PR doesn't fix the actual configuration anomaly raised in this issue?
> 
> * [--disable-debug-info enables debug information #1646 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1646#issuecomment-2558454552)

#1743 only strips host binaries.
AFAIK, config options don't have a opposite for this repo.
It has a default value, if you want enable/disable a feature, then add --enable/--disable, **don't use opposite option**.
for example, --enable-debug-info and --disable-debug-info both add `-g`.