riscv-collab/riscv-gnu-toolchain #1647: Backward jump by -2048 does not generate c.j

davidharrishmc, 2024-12-24:
I have a backward jump of -2048.  I expect RISC-V GCC would produce c.j -2048 (0xB001).  Instead, it gives an uncompressed jump by -2048 (0x801ff06f).

Is this the intended behavior, or is there a code generation bug?

This can be reproduced with the following code.  The unexpected uncompressed instruction is at 11804:

```
0000000000011800 <forward2048>:
   11800:	003100b3          	add	ra,sp,gp
   11804:	801ff06f          	j	11004 <back2048>

------ code

# cj.S
# Test case to illustrate GCC not generating compresed backward branches by -2048
# compile with
# riscv64-unknown-elf-gcc -march=rv64gc cj.S -o cj.elf
# riscv64-unknown-elf-objdump -D cj.elf > cj.elf.objdump

.global main

main: 
    j forward2048
.align 2
back2048:   
    j done

.align 11
forward2048: # address 80000800
    add x1, x2, x3 # dummy 4-byte instruction
    # j back2048 is a backward branch by a distance -2048 
    # I expect this should assemble into c.j -2048 (0xB001)
    # Instead, GCC gives  11804:	801ff06f          	j	11004 <back2048>
    j back2048

done:
    add x4, x5, x6
    
self_loop:
    j self_loop
```

davidharrishmc, 2024-12-26:
c.jal has the same behavior.

I would have expected this backward jal by 0x800 to produce machine language code 3001 rather than 801ff0ef.

80003800:	801ff0ef          	jal	80003000 <b11_c.jal>

TommyMurphyTM1234, 2024-12-31:
> c.jal has the same behavior.

This is probably not surprising since `c.j offset` expands to `c.jal x0, offset`?

> I would have expected this backward jal by 0x800 to produce machine language code 3001 rather than 801ff0ef.
> 
> 80003800: 801ff0ef jal 80003000 <b11_c.jal>

This suggests that LLVM/Clang may also expect this?

- https://reviews.llvm.org/D45385
- [image]

And that encoding certainly looks correct:

- https://luplab.gitlab.io/rvcodecjs/#q=0xB001&abi=false&isa=AUTO

Probably worth asking this question or logging a bug upstream since RISC-V maintenance/enhancement is done there rather than in this repo these days:

- https://gcc.gnu.org/lists.html
- https://gcc.gnu.org/bugs/

Perhaps it's a GCC bug with the edge case encoding?

For what it's worth:
```
# Pad to 2040 = (1024 + (127 x 8))
.align 10
# repeat the following .dword x 127
.dword 0xdeadbeefdeadbeef
...
forward2048: 

# Generated code
0000000000010ff8 <forward2048>:
   10ff8:       003100b3                add     ra,sp,gp
   10ffc:       b021                    j       10804 <back2048>
```
```
# Pad to 2044 = (1024 + (127 * 8) + 4)
.align 10
# repeat the following .dword x 127
.dword 0xdeadbeefdeadbeef
...
.word 0xdeadbeef
forward2048: 

# Generated code
0000000000010ffc <forward2048>:
   10ffc:       003100b3                add     ra,sp,gp
   11000:       805ff06f                j       10804 <back2048>
```

TommyMurphyTM1234, 2025-01-01:
Just in case this was something that has been addressed in the latest upstream GCC sources I built the toolchain as follows, unfortunately, get the same results.

I think that the issue should be reported upstream.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1647
cd issue-1647
git clone https://github.com/gcc-mirror/gcc gcc-upstream-master --depth=1
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-gcc-src=`pwd`/gcc-upstream-master
make

./installed-tools/bin/riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1647/installed-tools/libexec/gcc/riscv64-unknown-elf/15.0.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/issue-1647/gcc-upstream-master/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1647/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gf8b559626 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1647/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/issue-1647/gcc-upstream-master --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 15.0.0 20250101 (experimental) (gf8b559626)
```

TommyMurphyTM1234, 2025-01-03:
Quick follow up...

I modified the test code so that the backward jumps of -2040, -2042, -2044, -2046 and -2048 could be tested and the comments show the results if at most one of the "Goto +..." code blocks is uncommented. In summary, up to and including -2042 generates a 16-bit compressed `c.j` while -2044, -2046 and -2048 generate a 32-bit `jal x0, <offset>`.

I'll try to get this logged in the upstream GCC bugbase.

```
# cj.S
# Test case to illustrate GCC not generating compresed backward branches by -2048
# compile with
# riscv64-unknown-elf-gcc -march=rv64gc cj.S -o cj.elf
# riscv64-unknown-elf-objdump -D cj.elf > cj.elf.objdump

.global main

main: 
    j forward
.align 2
back:   
    j done

# Skip forward by 1024 bytes
.align 10
# Skip forward by 127 x 8 = 1016 bytes
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
.dword 0xdeadbeefdeadbeef
# Now at +2040 (0x7f8) - generates 16-bit c.j -2040

# Uncomment at most one of the following code blocks to test the relevant offset

# Goto +2042 (0x7fa) - generates 16-bit c.j -2042
# .hword 0xdead

# Goto +2044 (0x7fc) - generates 32-bit jal x0, -2044
# .word 0xdeadbeef

# Goto +2046 (0x7fe) - generates 32-bit jal x0, -2046
# .word 0xdeadbeef
# .hword 0xdead

# Goto +2048 (0x800) - generates 32-bit jal x0, -2048
# .dword 0xdeadbeefdeadbeef

forward:
    add x1, x2, x3 # dummy 4-byte instruction
    j back

done:
    add x4, x5, x6
    
self_loop:
    j self_loop

```

TommyMurphyTM1234, 2025-01-03:
I've logged the issue here:

- https://gcc.gnu.org/bugzilla/show_bug.cgi?id=118284

davidharrishmc, 2025-01-03:
+James Stine

Tommy,

Thank you for logging this.  

I’ve been searching the GCC code for this code generation, but seem to either be looking in the wrong place or at the wrong repo.  I suspect this will be easy for somebody who knows what they are doing, although your discovery about -2044 and -2046 throws a wrinkle into my theory that it just wasn’t handling the 2’s complement asymmetry between +/-2048.  

On further experimentation, there is some weird behavior on forward jumps as well.  I will follow up on the bugzilla issue.

David

----

I've logged the issue here:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=118284

TommyMurphyTM1234, 2025-01-03:
> On further experimentation, there is some weird behavior on forward jumps as well. 

Do you mean...

- jump +2040 -> `c.j`
- jump +2042 or more -> `jal x0, <offset>`

davidharrishmc, 2025-01-03:
It’s erratic.  

For example, the following code produces uncompressed jumps for offsets of 0x800 to 0x76c, and then compressed jumps for offsets less than 76c.  Doing other experiments gave different cutoffs for uncompressed jumps but I don’t have a consistent pattern yet.

main:
    .rept 550
    j forward
    .endr
.align 10
forward:

-----------------

main:
    .rept 550
    j forward
    80000000:	0010006f          	j	80000800 <forward>
    80000004:	7fc0006f          	j	80000800 <forward>
...
    80000090:	7700006f          	j	80000800 <forward>
    80000094:	76c0006f          	j	80000800 <forward>
    80000098:	a7a5                	j	80000800 <forward>
    8000009a:	a79d                	j	80000800 <forward>
...
    80000494:	a6b5                	j	80000800 <forward>
    80000496:	a6ad                	j	80000800 <forward>
    80000498:	00000013          	nop
...
    800007fc:	00000013          	nop

0000000080000800 <forward>:

 
.align 10
forward:
    j done
    80000800:	a009                	j	80000802 <done>

TommyMurphyTM1234, 2025-01-03:
> It’s erratic. For example, the following code produces uncompressed jumps for offsets of 0x800 to 0x76c

That doesn't seem to tally with my previous comments.
I've only tried the top end of the range but definitely get `c.j` for offset +2040 (0x7f8).

In case it matters (I doubt it) I'm using GCC 15.0.0:
```
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1647/installed-tools/libexec/gcc/riscv64-unknown-elf/15.0.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/issue-1647/gcc-upstream-master/configure --target=riscv64-unknown-elf --prefix=/home/user/issue-1647/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=gf8b559626 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1647/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/home/user/issue-1647/gcc-upstream-master --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 15.0.0 20250101 (experimental) (gf8b559626)
```

pinskia, 2025-01-03:
> +James Stine Tommy, Thank you for logging this. I’ve been searching the GCC code for this code generation, but seem to either be looking in the wrong place or at the wrong repo. I suspect this will [...]

Most likely because it is the GNU binutils project which handles the assembler/linker where the compression in this case happens. rather than GCC.

TommyMurphyTM1234, 2025-01-03:
> It’s erratic. For example, the following code produces uncompressed jumps for offsets of 0x800 to 0x76c, and then compressed jumps for offsets less than 76c. 

FWIW this is what I get for selected offsets related to your range above:

- +2048 (0x800): `jal x0, 2048` 
- +1900 (0x76c): `c.j 1900`
- +1888 (0x760): `c.j 1888`

davidharrishmc, 2025-01-04:
Here is my full input and output. This is with GCC 14.2.0.

Different experiments have given me different patterns of compressed vs. uncompressed, so it appears that the decision is not being made strictly on the distance.  

David Harris

# Test case to illustrate GCC not generating compresed backward branches by -2048
# compile with
# riscv64-unknown-elf-gcc -march=rv64gc cjforward.S -o cjforward.elf
# riscv64-unknown-elf-objdump -D cjforward.elf > cjforward.elf.objdump

.global main

main:
    .rept 550
    j forward
    .endr

 
.align 10
forward:
    j done

done:
    add x4, x5, x6
    
self_loop:
    j self_loop

 ￼
￼

TommyMurphyTM1234, 2025-01-04:
> Here is my full input and output. 
> ...
>  ￼ ￼

There seems to be something missing from your reply?
Maybe because you're using email to reply to the thread instead of posting directly here?

BTW - I have logged the backwads offset issue against Binutils now:

- https://sourceware.org/bugzilla/show_bug.cgi?id=32516

TommyMurphyTM1234, 2025-01-04:
> ```
> .rept 550
> ```

Could have saved myself some copying and pasting if I had been aware of that directive! :-D 

- https://ftp.gnu.org/old-gnu/Manuals/gas/html_chapter/as_toc.html#TOC116

davidharrishmc, 2025-01-04:
Uploading file example that didn't come through from email.

[cjforward.tar.gz](https://github.com/user-attachments/files/18307099/cjforward.tar.gz)

davidharrishmc, 2025-01-04:
The last example could be complicated for the compiler because it doesn't know whether the repeated jumps will be compressed or not, so it doesn't know where to place code and hence the offsets might change.  

Here's another example using NOP instead of repeated jumps, to make the code generation easier. GCC still never produces compressed forward jumps when I use .align 11, even with a bunch of NOPs before hand.  It happily produces a compressed jump with .align 10.  I think there is something more complicated going on here than whether binutils locally optimizes compressed jumps into uncompressed.

[cj2.tar.gz](https://github.com/user-attachments/files/18307228/cj2.tar.gz)

For reference, here is the code in plain text:
```
# Test case to illustrate GCC not generating compresed backward branches by -2048
# compile with
# riscv64-unknown-elf-gcc -march=rv64gc cj2.S -o cj2.elf
# riscv64-unknown-elf-objdump -D cj2.elf > cj2.elf.objdump

.global main

main:
    c.nop
    c.j forward2046  # GCC produces uncompressed jump even though it should be able to emit c.j
 
.align 11
forward2046:
    c.nop
    c.nop
    c.j forward2044  # GCC produces uncompressed jump even though it should be able to emit c.j

.align 11
forward2044:
    .rept 600
    nop
    .endr
    c.j forwarda  # GCC produces uncompressed jump even though it should be able to emit c.j

.align 11
forwarda:
    .rept 1000
    nop
    .endr
    c.j forwardb  # GCC produces uncompressed jump even though it should be able to emit c.j

.align 11
forwardb:
    c.j forwardc  # GCC produces c.j (note align 10 below)

.align 10
forwardc:
    add t0, t0, t1

done:
    add x4, x5, x6
    
self_loop:
    j self_loop
```
and the interesting disassembly
```

cj2.elf:     file format elf64-littleriscv

0000000000011000 <main>:
   11000:	0001                	nop
   11002:	7fe0006f          	j	11800 <forward2046>
   11006:	00000013          	nop
 ...
   117fe:	0001                	nop

0000000000011800 <forward2046>:
   11800:	0001                	nop
   11802:	0001                	nop
   11804:	7fc0006f          	j	12000 <forward2044>
   11808:	00000013          	nop
...
   11ffc:	00000013          	nop

0000000000012000 <forward2044>:
   12000:	0001                	nop
...
   124ac:	0001                	nop
   124ae:	0001                	nop
   124b0:	3500006f          	j	12800 <forwarda>
   124b4:	00000013          	nop
...
   127fc:	00000013          	nop

0000000000012800 <forwarda>:
   12800:	0001                	nop
...
   12fce:	0001                	nop
   12fd0:	0300006f          	j	13000 <forwardb>
...
   12ffc:	00000013          	nop

0000000000013000 <forwardb>:
   13000:	a101                	j	13400 <forwardc>
   13002:	0001                	nop
   13004:	00000013          	nop
...
   133fc:	00000013          	nop

0000000000013400 <forwardc>:
   13400:	929a                	add	t0,t0,t1

0000000000013402 <done>:
   13402:	00628233          	add	tp,t0,t1

0000000000013406 <self_loop>:
   13406:	a001                	j	13406 <self_loop>
```

TommyMurphyTM1234, 2025-01-13:
@davidharrishmc, if you have specific self contained reproducible tests cases for forward jump anomalies then you might want to add them to the bug that I logged or open a separate one specifically for the forward jump issues (the former seems like the more appropriate approach if the subject line of my bug is edited appropriately?):

- https://sourceware.org/bugzilla/show_bug.cgi?id=32516

Once that is done I would prefer to close this issue here as it is then redundant with the focus moving upstream.

davidharrishmc, 2025-01-13:
I’ve got too many balls in the air to put my last anomaly into Bugzilla properly.  Let’s let it drop.

The -2048 jump is the key one for the ability to complete functional testing on our core.  The others are missing optimizations.

Thank you for guiding this and please close the issue when you are satisfied.

David

TommyMurphyTM1234, 2025-01-13:
Thanks  @davidharrishmc - FWIW I've linked from the Binutils bug report to your previous post in case it's of any use with further investigation/analysis.

- https://sourceware.org/bugzilla/show_bug.cgi?id=32516#c1

closed 2025-01-13 by TommyMurphyTM1234