riscv-collab/riscv-gnu-toolchain #1649: Huge binary sizes

MrHighVoltage, 2024-12-30:
Hi everyone,

first of all, Merry Christmas (to all who celebrate) and all the best for 2025!

Now, to my issue: We are integrating the RISC-V Toolchain into our Docker-based distribution for open-source chip design ([IIC-OSIC-Tools](https://github.com/iic-jku/iic-osic-tools)). As of now, we are working on shrinking the image, to make it more accessible to people with limited bandwidth and storage available.
I found out, that there are a few binaries in the toolchain, that are huge, compared to the other <10 MiB binaries. Here is a list:

```
400M riscv64-unknown-elf-lto-dump
177M spike
168M spike-log-parser
167M riscv64-unknown-elf-gdb
165M xspike
165M termios-xspik
```

Is there a way to get more compact binaries in this case?

Thanks for your help and BR,

Georg

TommyMurphyTM1234, 2024-12-30:
Maybe you need to build the toolchain without debug information and maybe also strip the binaries?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1646#issuecomment-2558581458

TommyMurphyTM1234, 2024-12-30:
FWIW - here's a way to reduce the size of the core toolchain and its host based binaries (I'll look at Spike anon):
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1649
cd issue-1649

./configure --prefix=`pwd`/installed-tools --with-sim=spike

# Default build - host binaries include debug info
make

du -hd0 installed-tools
2.0G    installed-tools

ls -hl installed-tools/bin
total 730M
-rwxr-xr-x 1 user user 5.8M Dec 30 16:54 riscv64-unknown-elf-addr2line
-rwxr-xr-x 2 user user 6.0M Dec 30 16:54 riscv64-unknown-elf-ar
-rwxr-xr-x 2 user user 8.1M Dec 30 16:54 riscv64-unknown-elf-as
-rwxr-xr-x 2 user user 8.9M Dec 30 17:08 riscv64-unknown-elf-c++
-rwxr-xr-x 1 user user 5.8M Dec 30 16:54 riscv64-unknown-elf-c++filt
-rwxr-xr-x 1 user user 8.8M Dec 30 17:08 riscv64-unknown-elf-cpp
-rwxr-xr-x 1 user user 127K Dec 30 16:54 riscv64-unknown-elf-elfedit
-rwxr-xr-x 2 user user 8.9M Dec 30 17:08 riscv64-unknown-elf-g++
-rwxr-xr-x 2 user user 8.8M Dec 30 17:08 riscv64-unknown-elf-gcc
-rwxr-xr-x 2 user user 8.8M Dec 30 17:08 riscv64-unknown-elf-gcc-14.2.0
-rwxr-xr-x 1 user user 159K Dec 30 17:08 riscv64-unknown-elf-gcc-ar
-rwxr-xr-x 1 user user 159K Dec 30 17:08 riscv64-unknown-elf-gcc-nm
-rwxr-xr-x 1 user user 159K Dec 30 17:08 riscv64-unknown-elf-gcc-ranlib
-rwxr-xr-x 1 user user 6.3M Dec 30 17:08 riscv64-unknown-elf-gcov
-rwxr-xr-x 1 user user 5.1M Dec 30 17:08 riscv64-unknown-elf-gcov-dump
-rwxr-xr-x 1 user user 5.2M Dec 30 17:08 riscv64-unknown-elf-gcov-tool
-rwxr-xr-x 1 user user 162M Dec 30 17:10 riscv64-unknown-elf-gdb
-rwxr-xr-x 1 user user 4.6K Dec 30 17:10 riscv64-unknown-elf-gdb-add-index
-rwxr-xr-x 1 user user 6.4M Dec 30 16:54 riscv64-unknown-elf-gprof
-rwxr-xr-x 4 user user 9.2M Dec 30 16:54 riscv64-unknown-elf-ld
-rwxr-xr-x 4 user user 9.2M Dec 30 16:54 riscv64-unknown-elf-ld.bfd
-rwxr-xr-x 1 user user 398M Dec 30 17:08 riscv64-unknown-elf-lto-dump
-rwxr-xr-x 2 user user 5.9M Dec 30 16:54 riscv64-unknown-elf-nm
-rwxr-xr-x 2 user user 6.6M Dec 30 16:54 riscv64-unknown-elf-objcopy
-rwxr-xr-x 2 user user 9.4M Dec 30 16:54 riscv64-unknown-elf-objdump
-rwxr-xr-x 2 user user 6.0M Dec 30 16:54 riscv64-unknown-elf-ranlib
-rwxr-xr-x 2 user user 4.2M Dec 30 16:54 riscv64-unknown-elf-readelf
-rwxr-xr-x 1 user user 9.4M Dec 30 17:10 riscv64-unknown-elf-run
-rwxr-xr-x 1 user user 5.8M Dec 30 16:54 riscv64-unknown-elf-size
-rwxr-xr-x 1 user user 5.8M Dec 30 16:54 riscv64-unknown-elf-strings
-rwxr-xr-x 2 user user 6.6M Dec 30 16:54 riscv64-unknown-elf-strip

file installed-tools/bin/riscv64-unknown-elf-gcc
installed-tools/bin/riscv64-unknown-elf-gcc: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=daa38bcf9edf3a945c60e87425222ce3d44bd4f2, for GNU/Linux 3.2.0, with debug_info, not stripped

# Custom build - host binaries with no debug info
make distclean
rm -rf installed-tools

make ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0

du -hd0 installed-tools
493M    installed-tools

ls -hl installed-tools/bin
total 162M
-rwxr-xr-x 1 user user 1.7M Dec 30 18:43 riscv64-unknown-elf-addr2line
-rwxr-xr-x 2 user user 1.7M Dec 30 18:43 riscv64-unknown-elf-ar
-rwxr-xr-x 2 user user 2.5M Dec 30 18:43 riscv64-unknown-elf-as
-rwxr-xr-x 2 user user 3.8M Dec 30 19:06 riscv64-unknown-elf-c++
-rwxr-xr-x 1 user user 1.7M Dec 30 18:43 riscv64-unknown-elf-c++filt
-rwxr-xr-x 1 user user 3.8M Dec 30 19:06 riscv64-unknown-elf-cpp
-rwxr-xr-x 1 user user  58K Dec 30 18:43 riscv64-unknown-elf-elfedit
-rwxr-xr-x 2 user user 3.8M Dec 30 19:06 riscv64-unknown-elf-g++
-rwxr-xr-x 2 user user 3.8M Dec 30 19:06 riscv64-unknown-elf-gcc
[... 45 lines cut ...]
-rwxr-xr-x 1 user user  43K Dec 30 19:13 riscv64-unknown-elf-gcc-ar
-rwxr-xr-x 1 user user  43K Dec 30 19:12 riscv64-unknown-elf-gcc-nm
-rwxr-xr-x 1 user user  43K Dec 30 19:13 riscv64-unknown-elf-gcc-ranlib
-rwxr-xr-x 1 user user 1.6M Dec 30 19:13 riscv64-unknown-elf-gcov
-rwxr-xr-x 1 user user 1.4M Dec 30 19:12 riscv64-unknown-elf-gcov-dump
-rwxr-xr-x 1 user user 1.4M Dec 30 19:13 riscv64-unknown-elf-gcov-tool
-rwxr-xr-x 1 user user  15M Dec 30 19:13 riscv64-unknown-elf-gdb
-rwxr-xr-x 1 user user 4.6K Dec 30 19:08 riscv64-unknown-elf-gdb-add-index
-rwxr-xr-x 1 user user 1.6M Dec 30 19:13 riscv64-unknown-elf-gprof
-rwxr-xr-x 4 user user 2.6M Dec 30 19:13 riscv64-unknown-elf-ld
-rwxr-xr-x 4 user user 2.6M Dec 30 19:13 riscv64-unknown-elf-ld.bfd
-rwxr-xr-x 1 user user  71M Dec 30 19:13 riscv64-unknown-elf-lto-dump
-rwxr-xr-x 2 user user 1.6M Dec 30 19:13 riscv64-unknown-elf-nm
-rwxr-xr-x 2 user user 1.7M Dec 30 19:13 riscv64-unknown-elf-objcopy
-rwxr-xr-x 2 user user 2.7M Dec 30 19:13 riscv64-unknown-elf-objdump
-rwxr-xr-x 2 user user 1.6M Dec 30 19:13 riscv64-unknown-elf-ranlib
-rwxr-xr-x 2 user user 1.3M Dec 30 19:12 riscv64-unknown-elf-readelf
-rwxr-xr-x 1 user user 2.5M Dec 30 19:13 riscv64-unknown-elf-run
-rwxr-xr-x 1 user user 1.6M Dec 30 19:13 riscv64-unknown-elf-size
-rwxr-xr-x 1 user user 1.6M Dec 30 19:12 riscv64-unknown-elf-strings
-rwxr-xr-x 2 user user 1.7M Dec 30 19:13 riscv64-unknown-elf-strip

file installed-tools/bin/riscv64-unknown-elf-gcc
installed-tools/bin/riscv64-unknown-elf-gcc: ELF 64-bit LSB executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=34c4f1713b10db5f2ea6e143908f7cb42f61dc35, for GNU/Linux 3.2.0, stripped
```

TommyMurphyTM1234, 2024-12-30:
Regarding Spike...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1649
cd issue-1649

./configure --prefix=`pwd`/installed-tools --with-sim=spike

# Build toolchain with no debug info
make ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0

# Strip the binaries
find installed-tools/bin -type f -executable -exec strip {} \;
strip: installed-tools/bin/riscv64-unknown-elf-gdb-add-index: file format not recognized

# Try to build Spike with no debug info
make build-sim ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0

# For some reason this fails when trying to build pk64
# I don't know why yet
...
make[1]: Entering directory '/home/user/issue-1646/build-pk64'
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/file.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/syscall.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/handlers.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/frontend.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/elf.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/console.c
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/mmap.c
/home/user/issue-1646/pk/pk/mmap.c: In function 'init_early_alloc':
/home/user/issue-1646/pk/pk/mmap.c:555:47: error: 'MEM_START' undeclared (first use in this function); did you mean 'KVA_START'?
  555 |   free_pages = (mem_size - (first_free_page - MEM_START)) / RISCV_PGSIZE;
      |                                               ^~~~~~~~~
      |                                               KVA_START
/home/user/issue-1646/pk/pk/mmap.c:555:47: note: each undeclared identifier is reported only once for each function it appears in
/home/user/issue-1646/pk/pk/mmap.c: In function 'pk_vm_init':
/home/user/issue-1646/pk/pk/mmap.c:566:33: error: 'MEM_START' undeclared (first use in this function); did you mean 'KVA_START'?
  566 |   __map_kernel_range(KVA_START, MEM_START, mem_size, PROT_READ|PROT_WRITE|PROT_EXEC);
      |                                 ^~~~~~~~~
      |                                 KVA_START
make[1]: *** [Makefile:336: mmap.o] Error 1
make[1]: Leaving directory '/home/user/issue-1646/build-pk64'
make: *** [Makefile:1112: stamps/build-pk64] Error 2

# So let's try to build without -g0 first
rm stamps/build-spike stamps/build-pk64

make build-sim

# Now just rebuild Spike (not pk64) with -g0
rm stamps/build-spike 

# And strip the Spike binaries
find installed-tools/bin -type f -executable -exec strip {} \;
strip: installed-tools/bin/riscv64-unknown-elf-gdb-add-index: file format not recognized

du -hd0 installed-tools
515M    installed-tools

ls -hl installed-tools/bin/*spike*
-rwxr-xr-x 1 user user 8.4M Dec 30 21:27 installed-tools/bin/spike
-rwxr-xr-x 1 user user 300K Dec 30 21:27 installed-tools/bin/spike-dasm
-rwxr-xr-x 1 user user 8.1M Dec 30 21:27 installed-tools/bin/spike-log-parser
-rwxr-xr-x 1 user user 8.0M Dec 30 21:27 installed-tools/bin/termios-xspike
-rwxr-xr-x 1 user user 8.0M Dec 30 21:27 installed-tools/bin/xspike

file installed-tools/bin/spike
installed-tools/bin/spike: ELF 64-bit LSB pie executable, x86-64, version 1 (GNU/Linux), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=7decb0f1ed703f5a8fd9de4d5bebd2889f702c8c, for GNU/Linux 3.2.0, stripped
```
Hope this all helps for you to built a leaner toolchain plus Spike.

TommyMurphyTM1234, 2024-12-30:
Or, more succinctly...
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1649
cd issue-1649

./configure --prefix=`pwd`/installed-tools --with-sim=spike

make ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0
make stamps/build-pk64
make build-sim ASFLAGS==g0 CFLAGS=-g0 CXXFLAGS=-g0

find installed-tools/bin -type f -executable -exec strip {} \;
strip: installed-tools/bin/riscv64-unknown-elf-gdb-add-index: file format not recognized

du -hd0 installed-tools
521M    installed-tools
```
So that's 512MB versus the original default size of 3GB - about 1/6th the size:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1649
cd issue-1649

./configure --prefix=`pwd`/installed-tools --with-sim=spike

make 
make build-sim

du -hd0 installed-tools
3.0G    installed-tools
```

TommyMurphyTM1234, 2024-12-31:
Regarding the pk64 build failure when using `make build-sim ASFLAGS=g0 CFLAGS=-g0 CXXFLAGS=-g0` it looks like, for example, the `CFLAGS` specified on the command line completely override those specified in the pk64 `Makefile`:
```
# make build-sim
riscv64-unknown-elf-gcc -MMD -MP -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE  -march=rv64gc -mabi=lp64d -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE -DBBL_PAYLOAD=\"bbl_payload\" -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/mmap.c

# make build-sim ASFLAGS=g0 CFLAGS=-g0 CXXFLAGS=-g0
riscv64-unknown-elf-gcc -MMD -MP -g0 -I. -I/home/user/issue-1646/pk/pk -I/home/user/issue-1646/pk/bbl -I/home/user/issue-1646/pk/softfloat -I/home/user/issue-1646/pk/dummy_payload -I/home/user/issue-1646/pk/machine -I/home/user/issue-1646/pk/util -c /home/user/issue-1646/pk/pk/mmap.c
/home/user/issue-1646/pk/pk/mmap.c: In function 'init_early_alloc':
/home/user/issue-1646/pk/pk/mmap.c:555:47: error: 'MEM_START' undeclared (first use in this function); did you mean 'KVA_START'?
  555 |   free_pages = (mem_size - (first_free_page - MEM_START)) / RISCV_PGSIZE;
      |                                               ^~~~~~~~~
      |                                               KVA_START
/home/user/issue-1646/pk/pk/mmap.c:555:47: note: each undeclared identifier is reported only once for each function it appears in
/home/user/issue-1646/pk/pk/mmap.c: In function 'pk_vm_init':
/home/user/issue-1646/pk/pk/mmap.c:566:33: error: 'MEM_START' undeclared (first use in this function); did you mean 'KVA_START'?
  566 |   __map_kernel_range(KVA_START, MEM_START, mem_size, PROT_READ|PROT_WRITE|PROT_EXEC);
      |                                 ^~~~~~~~~                                                                                                                                                 |                                 KVA_START                                                                                                                                           make[1]: *** [Makefile:336: mmap.o] Error 1                                                                                                                                                 make[1]: Leaving directory '/home/user/issue-1646/build-pk64'
make: *** [Makefile:1112: stamps/build-pk64] Error 2                                      
```
even though the `Makefile` seems to be structured to allow additional `CFLAGS` to be specified in this way:
```
# Extract from pk64 Makefile
CFLAGS        := -Wall -Werror -D__NO_INLINE__ -mcmodel=medany -O2 -std=gnu99 -Wno-unused -Wno-attributes -fno-delete-null-pointer-checks -fno-PIE $(CFLAGS) $(march) $(mabi) -DBBL_LOGO_FILE=\"bbl_logo_file\" -DMEM_START=0x80000000 -fno-stack-protector -U_FORTIFY_SOURCE
```
I'm not sure why this happens and would have to dig into the build logs and `Makefile`s further.

MrHighVoltage, 2025-01-01:
Thanks for the deep insights! I'm already very happy with the first part, building the whole toolchain with debugging symbols disabled, reducing the size significantly. 

Thanks again for the quick and easy solution!

closed 2025-01-01 by MrHighVoltage

TommyMurphyTM1234, 2025-01-01:
Hi @MrHighVoltage - glad it helped. I've flagged the issue previously here so maybe at some point the default will be to build with `-g0` and maybe also to strip the toolchain host binaries:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1608#issuecomment-2558582358