riscv-collab/riscv-gnu-toolchain #1650: configure: line 3495: riscv32-unknown-linux-gnu-gcc: command not found, when I think that's what I am trying to _create_

DHDOWNS, 2024-12-31:
I am following this:
https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#readme
to build on Ubuntu 24.04.1 LTS. The install for riscv64-unknown-elf-gcc works. 
But I want 32. So I try:
%> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
%> gmake linux
and it fails. In build-glibc-linux-rv32gc-ilp32d/config.log I see the following, and if that is the actual cause of failure I don't get it (is not riscv32-unknown-linux-gnu-gcc (one of the things) I am trying to _create_?  ):
...
configure:2969: result: riscv32-unknown-linux-gnu
configure:3007: checking for riscv32-unknown-linux-gnu-gcc
configure:3040: result: riscv32-unknown-linux-gnu-gcc
configure:3445: checking for C compiler version
configure:3454: riscv32-unknown-linux-gnu-gcc  --version >&5
/home/dan/riscv/riscv-gnu-toolchain/glibc/configure: line 3456: riscv32-unknown-linux-gnu-gcc: command not found
configure:3465: $? = 127
configure:3454: riscv32-unknown-linux-gnu-gcc  -v >&5
/home/dan/riscv/riscv-gnu-toolchain/glibc/configure: line 3456: riscv32-unknown-linux-gnu-gcc: command not found
configure:3465: $? = 127 
...

If someone can point out my fatal flaw in thinking here I'd appreciate it. I do not know all what you need but here's where it all went downhill (again, I do have the 64bit toolchain working).

The build fails when I gmake linux
%> ~/riscv/riscv-gnu-toolchain
%> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
%> gmake linux
rm -rf stamps/build-glibc-linux-rv32gc-ilp32d build-glibc-linux-rv32gc-ilp32d
mkdir build-glibc-linux-rv32gc-ilp32d
cd build-glibc-linux-rv32gc-ilp32d && \
	CC="riscv32-unknown-linux-gnu-gcc " \
	CXX="this-is-not-the-compiler-youre-looking-for" \
	CFLAGS="   -mcmodel=medlow -O2 " \
	CXXFLAGS="   -mcmodel=medlow -O2 " \
	ASFLAGS="  -mcmodel=medlow " \
	/home/dan/riscv/riscv-gnu-toolchain/glibc/configure \
	--host=riscv32-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/dan/riscv/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv32-unknown-linux-gnu
checking for riscv32-unknown-linux-gnu-gcc... riscv32-unknown-linux-gnu-gcc 
checking for suffix of object files... configure: error: in '/home/dan/riscv/riscv-gnu-toolchain/build-glibc-linux-rv32gc-ilp32d':
configure: error: cannot compute suffix of object files: cannot compile
See 'config.log' for more details
gmake: *** [Makefile:444: stamps/build-glibc-linux-rv32gc-ilp32d] Error 1

TommyMurphyTM1234, 2024-12-31:
> The install for riscv64-unknown-elf-gcc works. But I want 32.
> So I try: 
> ```
> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d 
> gmake linux 
> ```
> and it fails. 

If you have built the `riscv64-unknown-elf` bare-metal/Newlib toolchain but want the 32 bit version why are you doing `gmake linux`?

Please do a clean build in case the previous build is impacting the new build and if it still fails then upload the zipped build log - e.g.
```
make distclean && rm -rf /opt/riscv
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
[g]make linux
```

DHDOWNS, 2024-12-31:
Thanks. I will clean and retry.

As for your
If you have built the riscv64-unknown-elf bare-metal/Newlib toolchain but
want the 32 bit version why are you doing gmake linux?

question, the answer is that's this hardware engineer's attempt to follow
directions by backing up to the

The build defaults to targeting RV64GC (64-bit) with glibc, even on a
32-bit build environment. To build the 32-bit RV32GC toolchain, use:
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d make
linux

step without properly (apparently) comprehending what is built/made at
each step. So I appreciate the help.

TommyMurphyTM1234, 2024-12-31:
> Thanks. I will clean and retry.

Generally it's better to do a `make distclean` and maybe also an `rm -rf <prefix-dir>` when changing the configuration, especially the `arch/abi`, otherwise build problems such as you mention can occur.

> As for your If you have built the riscv64-unknown-elf bare-metal/Newlib toolchain but want the 32 bit version why are you doing gmake linux? question, the answer is that's this hardware engineer's [...]

I don't really understand this. But if you do this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1650
cd issue-1650
./configure --prefix=...
make
```
then that builds the `rv64gc/lp64d` bare-metal Newlib toolchain and not the Linux/Glibc toolchain.
If you specified `--enable-linux` to `configure` then it would build the Linux toolchain.

TommyMurphyTM1234, 2024-12-31:
Also, bear in mind that any RISC-V toolchain configuration can generate code for both RV32 and RV64.
You don't necessarily need a `riscv32-unknown-...` toolchain to target RV32.
If you want a single toolchain that can do both (RV32 and RV64) then you can maybe use a multilib toolchain.
E.g. this will build a `riscv64-unknown-...` toolchain that can target both and bundles multilibs for a selected set of RV32 and RV64 `arch/abi`'s.
```
./configure ... --enable-multilib
make [linux]
```

TommyMurphyTM1234, 2025-01-01:
FWIW this worked fine for me:
```
# Check distro version
cat /etc/os-release
PRETTY_NAME="Ubuntu 24.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.1 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1650
cd issue-1650

# Install prerequisites
sudo .github/setup-apt.sh

# Configure and build rv64gc/lp64d bare-metal Newlib toolchain
./configure --prefix=`pwd`/installed-tools
make

# Configure and build rv32gc/ilp32d Linux/Glibc toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
make linux

# Both worked!
ls installed-tools/bin/
riscv32-unknown-linux-gnu-addr2line   riscv32-unknown-linux-gnu-gcov           riscv32-unknown-linux-gnu-ranlib   riscv64-unknown-elf-g++            riscv64-unknown-elf-ld.bfd
riscv32-unknown-linux-gnu-ar          riscv32-unknown-linux-gnu-gcov-dump      riscv32-unknown-linux-gnu-readelf  riscv64-unknown-elf-gcc            riscv64-unknown-elf-lto-dump
riscv32-unknown-linux-gnu-as          riscv32-unknown-linux-gnu-gcov-tool      riscv32-unknown-linux-gnu-run      riscv64-unknown-elf-gcc-14.2.0     riscv64-unknown-elf-nm
riscv32-unknown-linux-gnu-c++         riscv32-unknown-linux-gnu-gdb            riscv32-unknown-linux-gnu-size     riscv64-unknown-elf-gcc-ar         riscv64-unknown-elf-objcopy
riscv32-unknown-linux-gnu-c++filt     riscv32-unknown-linux-gnu-gdb-add-index  riscv32-unknown-linux-gnu-strings  riscv64-unknown-elf-gcc-nm         riscv64-unknown-elf-objdump
riscv32-unknown-linux-gnu-cpp         riscv32-unknown-linux-gnu-gfortran       riscv32-unknown-linux-gnu-strip    riscv64-unknown-elf-gcc-ranlib     riscv64-unknown-elf-ranlib
riscv32-unknown-linux-gnu-elfedit     riscv32-unknown-linux-gnu-gprof          riscv64-unknown-elf-addr2line      riscv64-unknown-elf-gcov           riscv64-unknown-elf-readelf
riscv32-unknown-linux-gnu-g++         riscv32-unknown-linux-gnu-ld             riscv64-unknown-elf-ar             riscv64-unknown-elf-gcov-dump      riscv64-unknown-elf-run
riscv32-unknown-linux-gnu-gcc         riscv32-unknown-linux-gnu-ld.bfd         riscv64-unknown-elf-as             riscv64-unknown-elf-gcov-tool      riscv64-unknown-elf-size
riscv32-unknown-linux-gnu-gcc-14.2.0  riscv32-unknown-linux-gnu-lto-dump       riscv64-unknown-elf-c++            riscv64-unknown-elf-gdb            riscv64-unknown-elf-strings
riscv32-unknown-linux-gnu-gcc-ar      riscv32-unknown-linux-gnu-nm             riscv64-unknown-elf-c++filt        riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-strip
riscv32-unknown-linux-gnu-gcc-nm      riscv32-unknown-linux-gnu-objcopy        riscv64-unknown-elf-cpp            riscv64-unknown-elf-gprof
riscv32-unknown-linux-gnu-gcc-ranlib  riscv32-unknown-linux-gnu-objdump        riscv64-unknown-elf-elfedit        riscv64-unknown-elf-ld
```

TommyMurphyTM1234, 2025-01-01:
Hi @DHDOWNS - as shown above I cannot reproduce your problem.
If you are still experiencing it then please provide full details needed to reproduce it - i.e. all relevant steps and commands from the starting `git clone` through the `configure` and `make`. 
Ideally this should be done on a clean host setup if possible - e.g. a clean Docker container or virtual machine - in order to eliminate the possibility that previous activity on the host and in the environment is interfering with the build.

DHDOWNS, 2025-01-01:
Thanks again.
I just tried this:

> make distclean && rm -rf /opt/riscv
> ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
> [g]make linux

from your first reply and eventually I see this:
   /opt/riscv/riscv32-unknown-linux-gnu
but I can't find the gcc build anywhere. For example:

%>$ ls -l /opt/riscv/riscv32-unknown-linux-gnu/bin/
total 76820
-rwxr-xr-x 2 root root  6204288 Jan  1 11:11 ar
-rwxr-xr-x 2 root root  8383752 Jan  1 11:11 as
-rwxr-xr-x 4 root root 11985480 Jan  1 11:11 ld
-rwxr-xr-x 4 root root 11985480 Jan  1 11:11 ld.bfd
-rwxr-xr-x 2 root root  6094216 Jan  1 11:11 nm
-rwxr-xr-x 2 root root  6848584 Jan  1 11:11 objcopy
-rwxr-xr-x 2 root root  9768696 Jan  1 11:11 objdump
-rwxr-xr-x 2 root root  6204312 Jan  1 11:11 ranlib
-rwxr-xr-x 2 root root  4317320 Jan  1 11:11 readelf
-rwxr-xr-x 2 root root  6848584 Jan  1 11:11 strip
and no amount of 'find'ing can locate a `riscv32-unknown-linux-gnu-gcc|g++|c++` anything

I'll start over again with your penulitmate reply (FWIW this worked fine for me) and try again.

Cheers (on Jan 1st)

TommyMurphyTM1234, 2025-01-01:
> Thanks again. I just tried this:
> 
> > make distclean && rm -rf /opt/riscv
> > ./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
> > [g]make linux
> 
> from your first reply and eventually I see this: /opt/riscv/riscv32-unknown-linux-gnu but I can't find the gcc build anywhere. 

Your toolchain binaries should be in `/opt/riscv/bin`.
E.g.:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1650
cd issue-1650
./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
make linux
cd installed-tools
tree > tree.log
less tree.log
.
├── bin
│   ├── riscv32-unknown-linux-gnu-addr2line
│   ├── riscv32-unknown-linux-gnu-ar
│   ├── riscv32-unknown-linux-gnu-as
│   ├── riscv32-unknown-linux-gnu-c++
│   ├── riscv32-unknown-linux-gnu-c++filt
│   ├── riscv32-unknown-linux-gnu-cpp
│   ├── riscv32-unknown-linux-gnu-elfedit
│   ├── riscv32-unknown-linux-gnu-g++
│   ├── riscv32-unknown-linux-gnu-gcc
│   ├── riscv32-unknown-linux-gnu-gcc-14.2.0
│   ├── riscv32-unknown-linux-gnu-gcc-ar
│   ├── riscv32-unknown-linux-gnu-gcc-nm
│   ├── riscv32-unknown-linux-gnu-gcc-ranlib
│   ├── riscv32-unknown-linux-gnu-gcov
│   ├── riscv32-unknown-linux-gnu-gcov-dump
│   ├── riscv32-unknown-linux-gnu-gcov-tool
│   ├── riscv32-unknown-linux-gnu-gdb
│   ├── riscv32-unknown-linux-gnu-gdb-add-index
│   ├── riscv32-unknown-linux-gnu-gfortran
│   ├── riscv32-unknown-linux-gnu-gprof
│   ├── riscv32-unknown-linux-gnu-ld
│   ├── riscv32-unknown-linux-gnu-ld.bfd
│   ├── riscv32-unknown-linux-gnu-lto-dump
│   ├── riscv32-unknown-linux-gnu-nm
│   ├── riscv32-unknown-linux-gnu-objcopy
│   ├── riscv32-unknown-linux-gnu-objdump
│   ├── riscv32-unknown-linux-gnu-ranlib
│   ├── riscv32-unknown-linux-gnu-readelf
│   ├── riscv32-unknown-linux-gnu-run
│   ├── riscv32-unknown-linux-gnu-size
│   ├── riscv32-unknown-linux-gnu-strings
│   └── riscv32-unknown-linux-gnu-strip
├── include
│   ├── gdb
│   │   └── jit-reader.h
│   └── sim
│       ├── callback.h
│       └── sim.h
├── lib
│   ├── bfd-plugins
│   │   └── libdep.so
│   ├── gcc
│   │   └── riscv32-unknown-linux-gnu
│   │       └── 14.2.0
│   │           ├── crtbegin.o
│   │           ├── crtbeginS.o
...
```
- `tree.log`: [tree.log](https://github.com/user-attachments/files/18286488/tree.log)

So, after doing a clean build (`make distclean && rm -rf /opt/riscv`) you no longer get the original toolchain build failure?

DHDOWNS, 2025-01-01:
Answering your last question first, yes. I did NOT see the build failure after doing that clean build.

But the installed tools/bin did no have those contents. They were not there (in that location) nor anywhere.
However, from your FWIW reply, this just completed:
>  distclean and purge and begin anew ...
>   700  git clone https://github.com/riscv-collab/riscv-gnu-toolchain
>   701  cd riscv-gnu-toolchain/
>   702  sudo .github/setup-apt.sh
>   703  ./configure --prefix=`pwd`/installed-tools --with-arch=rv32gc --with-abi=ilp32d
>   704  make linux

and I see:

> $ find ../../ -name \*riscv32-unknown\*
> ../../riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu
>...
> ../../riscv/riscv-gnu-toolchain/installed-tools/bin/riscv32-unknown-linux-gnu-gcc
> ...
and I can compile and get 

> %> file test
> test: ELF 32-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv32-ilp32d.so.1, for GNU/Linux 5.4.0, not stripped

and it disassembles into what looks like RISCV.
So it looks like that worked. 
Thank you so much for your help so far. If I may impose with two questions ...
Q1: It does seem I have an environment variable set differently that you since I no longer have anything in /opt/<nothing to see here>. Have you an recommendation there as I would like it all to end up in /opt/riscv/bin?
(FWIW, /opt/ is empty because this is a brand new laptop I have for only this one little project, so it's clean).
Q2: I will want to compile without the 16-bit/c/compressed extension. Will what I've built here allow that? (given I pass the compiler the correct switches of course).

Cheers.

TommyMurphyTM1234, 2025-01-01:
> Answering your last question first, yes. I did NOT see the build failure after doing that clean build.
> 
> But the installed tools/bin did no have those contents. 

Then it looks like your toolchain did not build properly.
The `<prefixdir>/bin` directory should contain the main `riscv32-unknown-linux-gnu-*` binaries when the toolchain has built correctly and only those ones should be used for compiling code.

You probably need to capture and upload your (zipped) build log to understand better what's happening. E.g.:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
sudo make linux 2>&1 | tee build.log
```

And, as I said before, it would be better if you could try everything from scratch on a clean host (e.g. Docker container or virtual machine) to avoid problems with previous work done on the host or a polluted environment etc.

> They were not there (in that location) nor anywhere. However, from your FWIW reply, this just completed:
>
>
> and I see:

Why `../../`?

> > ../../riscv/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv32-unknown-linux-gnu
> > ...
> > ../../riscv/riscv-gnu-toolchain/installed-tools/bin/riscv32-unknown-linux-gnu-gcc
> > ...
> > and I can compile and get

I don't understand what directories you are referring to here and cannot correlate them to the prefix directory that you specified to `configure`. It's very confusing and you probably need to state more clearly exactly what you're doing and why and upload (zipped) logs where applicable.

Building the toolchain works much more easily than all this so I can only assume that when it doesn't build and install properly then you are doing something wrong or your host/environment is somehow causing something unexpected/untoward to happen.

> > %> file test
> > test: ELF 32-bit LSB executable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-riscv32-ilp32d.so.1, for GNU/Linux 5.4.0, not stripped
> 
> and it disassembles into what looks like RISCV. 

Ditto - I'm totally confused.

> Q1: It does seem I have an environment variable set differently that you since I no longer have anything in /opt/. Have you an recommendation there as I would like it all to end up in /opt/riscv/bin?

If you want the built toolchain in `/opt/riscv` then you need to do what you were doing originally - usually you need to `sudo make ...` in order to be able to install the tools to `/opt/riscv`:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
sudo make linux

# And then to use the toolchain
export PATH=/opt/riscv/bin:$PATH
riscv32-unknown-linux-gnu-gcc -v
```
> Q2: I will want to compile without the 16-bit/c/compressed extension. Will what I've built here allow that? (given I pass the compiler the correct switches of course).

You mean that you want to compile for `rv32g/ilp32d`?
In that case why not build the toolchain for that `arch/abi` using `configure --prefix=... --with-arch=rv32g --with-abi=ilp32d`?

The `rv32gc/ilp32d` can compile for `rv32g/ilp32d` but may not link depending on what startup objects and/or libraries that get linked.

DHDOWNS, 2025-01-02:
Thank you for the suggestion to back up completely and retry what I was doing originally. I now have stuff I recognize, _viz_:
%> tree /opt/riscv/
/opt/riscv/
├── bin
│   ├── riscv32-unknown-linux-gnu-addr2line
...
│   ├── riscv32-unknown-linux-gnu-gcc
...
│   └── riscv32-unknown-linux-gnu-strip
├── include
│   ├── gdb
│   │   └── jit-reader.h
│   └── sim
│       ├── callback.h
│       └── sim.h
...
Thank your for your patience and help.

You asked a number of questions, and if you want answers I'm not too embarrassed to provide them.
Q(ish) ... Docker.
A: That's something else a naive hardware engineer would have to look into doing correctly, as this is my first go at installing tools such as these.

Q: Why ../../?
A: Happened to be where I was (in the dir). No reason.

Q:    "me: and it disassembles into what looks like RISCV."
Ditto - I'm totally confused.
A: Hardware engineers never understand anything unless they can see the bits, so I "confirm" I have a RISCV cross-compiler for RV32 by looking at the disassembled file. Pathetic perhaps, but it's what I need to see. 

Q: In that case why not build the toolchain for that arch/abi using configure --prefix=... --with-arch=rv32g --with-abi=ilp32d?
A: Because this hardware engineer (see above) is mostly unfamiliar with what the possible options are here.

So you've been dealing with an idiot, and you did marvelously.  Thanks again.

Cheers.

TommyMurphyTM1234, 2025-01-03:
> Thank you for the suggestion to back up completely and retry what I was doing originally. I now have stuff I recognize,

OK - so your toolchain build is working fine now, the original problem is no longer evident, and this issue can be closed?

DHDOWNS, 2025-01-03:
Yes. It can be closed.

closed 2025-01-03 by TommyMurphyTM1234