riscv-collab/riscv-gnu-toolchain #1651: unrecognized opcode `ld.aq a6,0(a4)'

TonyXinPlus, 2025-01-03:
Use riscv64-unknown-linux-gnu-gcc to compile ISA-Rel-Acq.s file. The inline assembly in ISA-Rel-Acq.c contains ld.aq a6,0(a4), which the riscv toolchain cannot recognize
The compilation command is as follows：
**riscv64-unknown-linux-gnu-gcc -Wall -std=gnu99 -march=rv64imafdc_zifencei -mabi=lp64d -O2 -pthread  -o ISA-Rel-Acq.exe outs.o utils.o litmus_rand.o ISA-Rel-Acq.s --static**

riscv64-unknown-linux-gnu-gcc -v is as follows
**riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-linux-gnu/13.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-pkgversion=gc891d8dc23e-dirty --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=riscv-gnu-toolchain/gcc --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc_zifencei --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 13.2.0 (gc891d8dc23e-dirty)
**

kito-cheng, 2025-01-03:
Maybe what you want is `lr.w.aq`, I don't see any instruction mnemonics is `ld.aq`?

TonyXinPlus, 2025-01-03:
asm __volatile__ (
"\n"
"#START _litmus_P1\n"
"#_litmus_P1_0\n\t"
"ori %[x6],x0,1\n"
"#_litmus_P1_1\n\t"
"amoswap.d %[x10],%[x6],(%[x9])\n"
"#_litmus_P1_2\n\t"
"ld.aq %[x11],0(%[x9])\n"
"#_litmus_P1_3\n\t"
"ld %[x12],0(%[x8])\n"
"#END _litmus_P1\n"
:[x12] "=&r" (_a->out_1_x12[_i]),[x11] "=&r" (_a->out_1_x11[_i]),[x10] "=&r" (_a->out_1_x10[_i]),[x6] "=&r" (trashed_x6)
:[x8] "r" (&_a->x[_i]),[x9] "r" (&_a->y[_i])
:"cc","memory"
);

TonyXinPlus, 2025-01-03:
> Maybe what you want is `lr.w.aq`, I don't see any instruction mnemonics is `ld.aq`?

asm __volatile__ (
"\n"
"#START _litmus_P1\n"
"#_litmus_P1_0\n\t"
"ori %[x6],x0,1\n"
"#_litmus_P1_1\n\t"
"amoswap.d %[x10],%[x6],(%[x9])\n"
"#_litmus_P1_2\n\t"
"ld.aq %[x11],0(%[x9])\n"
"#_litmus_P1_3\n\t"
"ld %[x12],0(%[x8])\n"
"#END _litmus_P1\n"
:[x12] "=&r" (_a->out_1_x12[_i]),[x11] "=&r" (_a->out_1_x11[_i]),[x10] "=&r" (_a->out_1_x10[_i]),[x6] "=&r" (trashed_x6)
:[x8] "r" (&_a->x[_i]),[x9] "r" (&_a->y[_i])
:"cc","memory"
);

TommyMurphyTM1234, 2025-01-03:
> `asm **volatile** ( "\n" "#START _litmus_P1\n" "#_litmus_P1_0\n\t" "ori %[x6],x0,1\n" "#_litmus_P1_1\n\t" "amoswap.d %[x10],%[x6],(%[x9])\n" "#_litmus_P1_2\n\t" "ld.aq %[x11],0(%[x9])\n" [...]

> Use riscv64-unknown-linux-gnu-gcc to compile ISA-Rel-Acq.s file.

Where does this source file originate from? 
Maybe you need to ask whoever created it why they seem to be using a non-standard instruction mnemonic?

TommyMurphyTM1234, 2025-01-03:
Where in the RISC-V specifications can you see where `ld.aq` is a valid instruction mnemonic?

- https://lf-riscv.atlassian.net/wiki/spaces/HOME/pages/16154769/RISC-V+Technical+Specifications#ISA-Specifications

Doesn't this:

[image]

suggest that @kito-cheng is correct [above](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1651#issuecomment-2568719593):

> Maybe what you want is lr.w.aq, I don't see any instruction mnemonics is ld.aq?

For what it's worth this (completely artifical) example assembles OK. I'm using the bare-metal/Newlib toolchain here but the Linux/Glibc toolchain should behave similarly:
```
riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools/libexec/gcc/riscv64-unknown-elf/14.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /home/user/ch32v003fun/riscv-gnu-toolchain/gcc/configure --target=riscv64-unknown-elf --prefix=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/ch32v003fun/riscv-gnu-toolchain/installed-tools/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 14.2.0 ()

# test.c
void f(void)
{
asm volatile("lr.w.aq x1,0(x2)");
}

riscv64-unknown-elf-gcc -march=rv64imafdc_zifencei -c test.c

riscv64-unknown-elf-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <f>:
   0:   1141                    addi    sp,sp,-16
   2:   e406                    sd      ra,8(sp)
   4:   e022                    sd      s0,0(sp)
   6:   0800                    addi    s0,sp,16
   8:   140120af                lr.w.aq ra,(sp)
   c:   0001                    nop
   e:   60a2                    ld      ra,8(sp)
  10:   6402                    ld      s0,0(sp)
  12:   0141                    addi    sp,sp,16
  14:   8082                    ret
```

> Use `riscv64-unknown-linux-gnu-gcc` to compile `ISA-Rel-Acq.s` file.

> ```
> asm volatile (
>     "\n"
>     "#START _litmus_P1\n"
>     "#_litmus_P1_0\n\t"
>     "ori %[x6],x0,1\n"
>     "#_litmus_P1_1\n\t"
>     "amoswap.d %[x10],%[x6],(%[x9])\n" [...]

Where does this code come from?

Maybe not relevant but you are not using the latest toolchain:

> ```
> gcc version 13.2.0 (gc891d8dc23e-dirty)
> ```

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1531

However I tried GCC 14.2 and it gives the same error.

closed 2025-01-04 by TonyXinPlus

aswaterman, 2026-01-13:
These instructions were introduced in the recently ratified Zalasr extension.  IIRC, that happened late last year.