riscv-collab/riscv-gnu-toolchain #1661: Adding a new isntruction to RISCV

TVS2004Chaitanya, 2025-02-01:
I want to create a custom instruction and use the custom instruction without the help of inline assembly. For example I created a new instruction mod, which computes modulus, But i should use inline assembly to access this new instruction if not used the % symbol will revoke the existing remw instruction in RV64GC architecture to compute modulus. But I want the compiler to revoke my custom instruction instead of the existing one.

TommyMurphyTM1234, 2025-02-01:
This is not really a `riscv-gnu-tookchain` question. It's a more general question about how to customise the GCC toolchain. 

You should familiarise yourself with the GCC internals documentation and maybe ask about this in an upstream GCC forum:

- https://gcc.gnu.org/onlinedocs/gccint/
- https://gcc.gnu.org/

There are also several third party blogs that provide examples of how to add custom instructions to the RISC-V toolchain.

Existing open and closed issue in the repo also contain possibly useful information.

closed 2025-02-01 by TommyMurphyTM1234

TVS2004Chaitanya, 2025-02-03:
Dear Sir,

Thank you for the resources; I will go through them. I have also explored several third-party blogs that provide examples of how to add custom instructions to RISC-V. However, all these blogs explain how to add custom instructions that can only be accessed through inline assembly. I want to create a custom instruction for the RISC-V architecture that can be accessed without using inline assembly. Do you know of any resources that cover this?

Thank You,
TVS Chaitanya.

  *   https://gcc.gnu.org/onlinedocs/gccint/
  *   https://gcc.gnu.org/

TommyMurphyTM1234, 2025-02-03:
> However, all these blogs explain how to add custom instructions that can only be accessed through inline assembly. I want to create a custom instruction for the RISC-V architecture that can be accessed without using inline assembly. Do you know of any resources that cover this?

The GCC internals documentation should help with explaining how to add custom instructions that can be inferred by the compiler from high level language patterns or operators and do not require explicit use via inline assembly. I believe that there are other blogs/tutorials that cover this in general or specifically for RISC-V GCC but I haven't checked recently. If you need further assistance then you should look for it in the likes of the GCC mailing lists, StackOverflow etc.