riscv-collab/riscv-gnu-toolchain #1663: Make fails to clone submodules un Ubuntu 20.04

jordancarlin, 2025-02-03:
On Ubuntu 20.04 I get the following error message when running make:

```
error: Server does not allow request for unadvertised object 26f7004bf73c421c3fd5e5a6ccf470d05337b435
Fetched in submodule path 'newlib', but it did not contain 26f7004bf73c421c3fd5e5a6ccf470d05337b435. Direct fetching of that commit failed.
make: *** [Makefile:348: /opt/riscv/riscv-gnu-toolchain/newlib/.git] Error 1
error: Server does not allow request for unadvertised object beb2cdbcda911764b2bed5e57921fe90493260bd
Fetched in submodule path 'binutils', but it did not contain beb2cdbcda911764b2bed5e57921fe90493260bd. Direct fetching of that commit failed.
make: *** [Makefile:348: /opt/riscv/riscv-gnu-toolchain/binutils/.git] Error 1
error: Server does not allow request for unadvertised object 04696df09633baf97cdbbdd6e9929b9d472161d3
Fetched in submodule path 'gcc', but it did not contain 04696df09633baf97cdbbdd6e9929b9d472161d3. Direct fetching of that commit failed.
make: *** [Makefile:348: /opt/riscv/riscv-gnu-toolchain/gcc/.git] Error 1
```

It seems to be related to the version of git. Ubuntu 20.04 comes with git 2.25.1. If I build and install the latest version of git then this problem goes away. Seeing as Ubuntu 20.04 is still supported, it would be ideal if this worked with the built in version of git. 

The problem seems to have been introduced by #1608.

TommyMurphyTM1234, 2025-02-03:
I think that it's the same issue as discussed/explained here?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1654#issuecomment-2591195261

Note that Ubuntu 20.04 is no longer one of the distros used by the CI builds:

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/a33dac0251d17a7b74d99bd8fd401bfce87d2aed/.github/workflows/nightly-release.yaml#L45-L52

However, addressing the shallow checkout problem mentioned in the first link will probably help with 20.04?

myyerrol, 2025-02-05:
You can solve it by typing `git config --global protocol.version 2` in the terminal. For detailed reasons, please see: [Error Fetching Submodule in RISC-V GNU Toolchain: Server Does Not Allow Request for Unadvertised Object](https://stackoverflow.com/questions/79222723/error-fetching-submodule-in-risc-v-gnu-toolchain-server-does-not-allow-request).

jordancarlin, 2025-02-05:
Thank you for the information. Very informative link!

TommyMurphyTM1234, 2025-02-15:
> You can solve it by typing `git config --global protocol.version 2` in the terminal. For detailed reasons, please see: [Error Fetching Submodule in RISC-V GNU Toolchain: Server Does Not Allow [...]

That didn't make any difference for me:
```
git config --global protocol.version 2

git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1663

cd issue-1663

./configure --prefix=`pwd`/installed-tools

 make
cd /home/user/issue-1663 && \
flock `git rev-parse --git-dir`/config git submodule init /home/user/issue-1663/gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /home/user/issue-1663/gcc/
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Cloning into '/home/user/issue-1663/gcc'...
remote: Enumerating objects: 146196, done.
remote: Counting objects: 100% (146196/146196), done.
remote: Compressing objects: 100% (112468/112468), done.
remote: Total 146196 (delta 37715), reused 91057 (delta 31702), pack-reused 0
Receiving objects: 100% (146196/146196), 175.53 MiB | 9.56 MiB/s, done.
Resolving deltas: 100% (37715/37715), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: Server does not allow request for unadvertised object 04696df09633baf97cdbbdd6e9929b9d472161d3
Fetched in submodule path 'gcc', but it did not contain 04696df09633baf97cdbbdd6e9929b9d472161d3. Direct fetching of that commit failed.
make: *** [Makefile:348: /home/user/issue-1663/gcc/.git] Error 1
```

TommyMurphyTM1234, 2025-02-15:
> Ubuntu 20.04 comes with git 2.25.1. If I build and install the latest version of git then this problem goes away.

How did you do this?
I tried this to no avail:
```
wget https://www.kernel.org/pub/software/scm/git/git-2.48.1.tar.xz
tar xvf git-2.48.1.tar.xz
cd git-2.48.1/
./configure --prefix=`pwd`/installed-tools
make
...
    GEN git-mergetool--lib
    GEN git-sh-i18n
    GEN git-sh-setup
    CC scalar.o
    LINK scalar
    SUBDIR git-gui
GITGUI_VERSION = 0.21.0.139.gae633
    * new locations or Tcl/Tk interpreter
    GEN git-gui
    INDEX lib/
    * tclsh failed; using unoptimized loading
    MSGFMT    po/bg.msg make[1]: *** [Makefile:239: po/bg.msg] Error 127
make: *** [Makefile:2421: all] Error 2
```

TommyMurphyTM1234, 2025-02-18:
Closing due to lack of updates and given that it's effectively a duplicate of this: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669

closed 2025-02-18 by TommyMurphyTM1234