riscv-collab/riscv-gnu-toolchain #1664: Compile C code to RISC-V  RV32I instruction

toolchainman, 2025-02-07:
Hello, I did the install according to the README. What I want to do is to convert C code to RISCV RV32I, but it says that the command riscv64-unknown-elf-gcc does not exist. Can you tell me what I'm doing wrong and what else I need to install? 
Also, what if I want to compile several C code files like dhry.h(header file), dhry_1.c(main function), dhry_2.c(benchmark function) into RISCV RV32I instead of just one C code file? 

 Sincerely

pz9115, 2025-02-07:
1. Please check your install directory to make sure that you installed toolchain correctly（the default directory is /opt/riscv, you can modify it with `--prefix=` option）, and if for rv32i only, you need to assign the arch and abi option `--with-arch=rv32i --with-abi=ilp32`

2. To compile multiple C files (e.g., dhry.h, dhry_1.c, dhry_2.c) with rv32i, you can use the following command:
`riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -o output.elf dhry_1.c dhry_2.c` and make sure the dry.h is include correctly.
(here is a introduce book include what you want https://web.archive.org/web/20051215144142/http://www.network-theory.co.uk/docs/gccintro/gccintro_11.html)

TommyMurphyTM1234, 2025-02-07:
> Hello, I did the install according to the README. What I want to do is to convert C code to RISCV RV32I, but it says that the command riscv64-unknown-elf-gcc does not exist. Can you tell me what I'm doing wrong and what else I need to install?

You need to explain exactly how you configured and built your toolchain and how you are trying to use it. E.g. by providing all of the commands used to clone, configure, build and invoke the toolchain.

> Also, what if I want to compile several C code files like dhry.h(header file), dhry_1.c(main function), dhry_2.c(benchmark function) into RISCV RV32I instead of just one C code file?

You should probably read the GCC documentation as @pz9115 suggests and maybe also learn how to use a build system such as `make`:

- https://www.gnu.org/software/make/

Apart from the main documentation for both tools there are lots of third party tutorials available online.

TommyMurphyTM1234, 2025-02-08:
Any update @toolchainman?

TommyMurphyTM1234, 2025-02-11:
Closing due to lack of feedback.

closed 2025-02-11 by TommyMurphyTM1234

toolchainman, 2025-02-11:
My apologies for the late response. I was tied up with work.

I followed all the installation steps in the README, but I’m getting an error saying that riscv64-unknown-elf-gcc does not exist.

The command I want to run is:
riscv64-unknown-elf-gcc -march=rv32i -mabi=ilp32 -S test.c

Could you help me figure out what’s wrong?"

kito-cheng, 2025-02-11:
> I followed all the installation steps in the README, but I’m getting an error saying that riscv64-unknown-elf-gcc does not exist.

Do you see `riscv32-unknown-elf-gcc` in your `<install-path>/bin`? it will prefixed with `riscv32-` rather than `riscv64-` if the default arch is RV32.

toolchainman, 2025-02-11:
Thank you for your comment.

I couldn't find riscv32-unknown-elf-gcc in my riscv-gnu-toolchain folder(the folder exist after install README maunal) . Additionally, there is no bin folder inside riscv-gnu-toolchain. Instead, I see other folders like gcc, dejagnu, and build-binutils-linux.

What should I do next?

Thanks again for your help!

TommyMurphyTM1234, 2025-02-11:
> Thank you for your comment.
>
> I couldn't find riscv32-unknown-elf-gcc in my riscv-gnu-toolchain folder(the folder exist after install README maunal) . Additionally, there is no bin folder inside [...]

You need to provide the information that I outlined here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1664#issuecomment-2643266334

> You need to explain exactly how you configured and built your toolchain and how you are trying to use it. E.g. by providing all of the commands used to clone, configure, build and invoke the toolchain.

TommyMurphyTM1234, 2025-02-11:
Also, if you don't actually need to build the toolchain yourself and need one that supports `rv32i/ilp32` specifically (it's not really clear what `arch/abi`(s?) you need to target) then maybe a better idea might be to use a pre-build binary toolchain distribution like @ilg-ul's xPack one?

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/

It also supports many other `arch/abi`'s via many bundled multilibs. E.g.:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/blog/2024/12/06/riscv-none-elf-gcc-v14-2-0-3-released/#supported-libraries

Note that this distribution uses a different prefix to the "regular" one:

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/user/#miscellaneous

I regularly see many people struggling unnecessarily with trying to build the toolchain when they probably don't need to and could use that wasted time better elsewhere.

TommyMurphyTM1234, 2025-02-11:
If you do insist on building the toolchain yourself then this works (I'm using Ubuntu 24.04 LTS here):
```
# Clone repo
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1664
cd issue-1664

# Install prerequisites
sudo .github/setup-apt.sh

# Configure toolchain
./configure --prefix=`pwd`/installed-tools --with-arch=rv32i --with-abi=ilp32

# Build toolchain
make 2>&1 | tee build.log
```
Notes:

1. I'm assuming above that you need to target `rv32i/ilp32` specifically but if you need to target something else or need a multilib toolchain then adjust the `configure` command as needed.
2. I prefer to keep the installed toolchain local rather than putting it in a "system" directory but if you want it somewhere else then change the `--prefix` option or you can copy the `installed-tools` directory hierarchy to your preferred location.
3. Wherever you install the toolchain you may want to add the installed tools `bin` directory to your `PATH` for ease of access.
4. For managing builds of more than a single source file you should learn how to use something like `Make` or `CMake` etc.

TommyMurphyTM1234, 2025-02-11:
> I couldn't find riscv32-unknown-elf-gcc in my riscv-gnu-toolchain folder(the folder exist after install README maunal) . Additionally, there is no bin folder inside riscv-gnu-toolchain. Instead, I see other folders like gcc, dejagnu, and build-binutils-linux.

Only noticed this now and it means that you're looking in the wrong place for the `bin` directory and toolchain binaries. These should be in whatever `--prefix` directory was specified at configuration time. My previous post should clarify.

toolchainman, 2025-03-04:
Thank you very much for your response. 
I'm currently having issues with the hardware I designed, so I've put off benchmark testing for a bit. 
However, benchmarking is definitely something I will do in the future, so I will try to do it one step at a time as you suggested, thank you very much.

toolchainman, 2025-03-05:
To: tommymurphytm1234 @TommyMurphyTM1234 

My goal is to compile the Dhrystone benchmark for the RISC-V RV32I architecture.

The output I want to obtain is the assembly code generated for RISC-V RV32I.

ex) add x1, x2, x3
     beq x4, x2, x3
      ...... 

I have four files (dhry_1.c, dhry_1_orig.c, dhry_2.c, dhry.h), and I want to compile the entire Dhrystone benchmark at once.

@pz9115  said  use this command 
riscv32-unknown-elf-gcc -march=rv32i -mabi=ilp32 -o output.elf dhry_1.c dhry_2.c

When I run the command, I get the error "riscv32-unknown-elf-gcc: Command not found."

I installed the toolchain by following the installation guide from [this repository](https://github.com/riscv-collab/riscv-gnu-toolchain) and believe I executed all the installation commands correctly.

How can I generate the complete assembly code?

Thanks!

*I'm a beginner in Linux, so please take that into consideration.

TommyMurphyTM1234, 2025-03-05:
> I have four files (dhry_1.c, dhry_1_orig.c, dhry_2.c, dhry.h), and I want to compile the entire Dhrystone benchmark at once.

This has been explained already above by @kito-cheng.

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1664#issuecomment-2642041891

And it's not a question about `riscv-gnu-toolchain` but about how to use GCC to compile programs. As I already said, for such questions you should read the GCC documentation and maybe learn how to use a build tool such as Make or CMake.

> When I run the command, I get the error "riscv32-unknown-elf-gcc: Command not found."

This has also been covered and I've explained how to build the toolchain, where to find the binaries, and that you'll need to ensure that they're on your `PATH` if you're not explicitly specifying the location of the toolchain.

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1664#issuecomment-2650668248

> *I'm a beginner in Linux, so please take that into consideration.

That's fine but it's not an excuse for repeating the same questions when they have already been addressed and people have provided useful feedback on them which seems to be ignored.

toolchainman, 2025-03-06:
I'm really sorry if I seemed to ignore you. That wasn't my intention. 
I have some questions, but I'll go through what you told me step by step again.
 If I still have a problem after trying again, I'll ask you. 
Thank you for your response.

TommyMurphyTM1234, 2025-03-10:
Please don't use screenshots for errors as they are difficult to read for some and cannot be searched/processed.

What GNU/Linux distro are you using? The `setup-apt.sh` script is for Ubuntu and other distros that use `apt`. For other distros that use different package managers you will have to adapt the script and maybe the packages installed yourself.

toolchainman, 2025-03-10:
Thank you for your comment.

I'm using Rocky Linux, so I'll ask GPT how to install it.

Today, I understood what you explained.
So, I ran the --prefix option and executed the make command:

=>./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d
make install

then i ran the command below 

=> riscv32-unknown-linux-gnu-gcc -march=rv32i -mabi=ilp32 -o output.elf dhry_1.c dhry_2.c

But I'm still getting an error=> riscv32-unknown-linux-gnu-gcc: Command not found.
I ran this command in  /opt/riscv/bin 
and I also tried running it in   /home/ksh97(my ID)/riscv-gnu-toolchain

Am I still clueless about the make command or gcc?

*Okay, I won’t use any screenshots for others.

TommyMurphyTM1234, 2025-03-10:
> I ran this command in /opt/riscv/bin - I found the "riscv32-unknown-linux-gnu-gcc file" in /opt/riscv/bin
>
> Am I still clueless about the make command or gcc?

Well, whatever about today, you still seem to be ignoring useful assistance already provided - e.g.:

> 4\. Wherever you install the toolchain you may want to add the installed tools `bin` directory to your `PATH` for ease of access.

I can't really help further if you're not going to take on board all of the useful feedback already provided by different contributors.

toolchainman, 2025-03-10:
Does number 4 assistance mean copying "/opt/riscv/bin" directory to "/home/ksh97 (my ID)/riscv-gnu-toolchain" directory?
As I understand it, if I follow the instructions in comment #4, the command should work when executed in /opt/riscv/bin. However, I still encountered an error even when running it from that location.

What am I misunderstanding?

TommyMurphyTM1234, 2025-03-10:
> Does number 4 assistance mean copying "/opt/riscv/bin" directory to "/home/ksh97 (my ID)/riscv-gnu-toolchain" directory?

No - why on earth would you think that?

> What am I misunderstanding?

How to use the GNU/Linux `PATH` environment variable for one thing. E.g.:

- https://phoenixnap.com/kb/linux-add-to-path

None of this has anything to do with `riscv-gnu-toolchain` so I am going to cease giving further feedback now.