riscv-collab/riscv-gnu-toolchain #1666: Build libsantizers for RISCV

manivalaguru, 2025-02-10:
Hi,
I have riscv64-unknown-linux-gnu-gcc (GCC) 10.1.0 installed on my x86_64 GNU/Linux system, but it does not include libasan_preinit.o. I attempted to build the library from the riscv-gnu-toolchain source repository.
System Details:

    Target Architecture: RISC-V 64
    Host Architecture: x86_64 GNU/Linux

Configuration Command Used:
./configure --host=<target-architecture> --build=<build-architecture> --prefix=<install-path> CC=<cross-compiler-gcc> CXX=<cross-compiler-g++>

However, when running make with this configuration, I encountered multiple undefined reference errors. Attempting to resolve them led to further issues, including assembly instruction mismatches.
Request for Assistance:

Is it possible to successfully build libasan for RISC-V GCC 10.1.0 on an x86_64 machine? If so, could you provide a step-by-step guide on how to achieve this?

Your insights would be greatly appreciated.

kito-cheng, 2025-02-10:
Build libsanitizer out-of-tree or standalone build is not really recommended way, and here is an undocumented configure option `--enable-libsanitizer`, and the function is like its name, that only support RV64, so don't try to build that for linux toolchain with multi-lib.

TommyMurphyTM1234, 2025-02-10:
> I have riscv64-unknown-linux-gnu-gcc (GCC) 10.1.0 installed on my x86_64 GNU/Linux system

That's quite old. `riscv-gnu-toolchain` uses GCC 14 these days. But I'm not sure if that's of any relevance to your core issue.

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/a33dac0251d17a7b74d99bd8fd401bfce87d2aed/.gitmodules#L6-L10

> Configuration Command Used:
> ```
> ./configure --host= --build= --prefix= CC= CXX=<cross-compiler-g++>
> ```

This is not a valid configuration command.

> Build libsanitizer out-of-tree or standalone build is not really recommended way, and here is an undocumented configure option --enable-libsanitizer

@kito-cheng do you mean that this should be used? 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer
make linux
```

kito-cheng, 2025-02-10:
@TommyMurphyTM1234 
> do you mean that this should be used?

Yeah, and the libsantizers should be built after adding that

manivalaguru, 2025-02-10:
HI @kito-cheng  thanks it worked is it possible built it for 10.1.0 i dont see that release in here

TommyMurphyTM1234, 2025-02-10:
As I said, GCC 10 is quite old and you may have tricky challenges trying to build it from this repo due to changes in how submodule components are handled over the past few years. In particular, in the past this repo used forked versions of most or all of the submodules whereas more recently it uses upstream tags/releases. And the old versions of those forked repos may not be easy to locate. And, apart from all that, there have been so many improvements/enhancements to the RISC-V and general support in all of the tools since GCC 10 that it really doesn't make sense to use such an old toolchain in most cases.

manivalaguru, 2025-02-13:
hi thanks for the reply will try to work on latest release i have couple of questions

1.Is it possible enable MSAN in RISCV(in latest release)
2.I see that riscv64-unknown-elf not genrated by default while using this command ./configure --prefix=/home/manikandan/Downloads/riscv_tools/new/riscv-gnu-toolchain --enable-libsanitizer  how can i genrate them

TommyMurphyTM1234, 2025-02-13:
> 1.Is it possible enable MSAN in RISCV(in latest release)

What us MSAN?
Please clarify exactly what you mean.

> 2.I see that riscv64-unknown-elf not genrated by default while using this command ./configure --prefix=/home/manikandan/Downloads/riscv_tools/new/riscv-gnu-toolchain --enable-libsanitizer how can i genrate them

What happens?
Please capture, zip and upload your build log.

TommyMurphyTM1234, 2025-02-13:
Maybe this PR was incomplete in only adding the option for the Linux/Glibx toolchain?

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1062
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/a33dac0251d17a7b74d99bd8fd401bfce87d2aed/Makefile.in#L46
- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/a33dac0251d17a7b74d99bd8fd401bfce87d2aed/Makefile.in#L525

manivalaguru, 2025-02-13:
1.) Memory Sanitizer (MSan) is an efficient runtime tool designed to detect uninitialized memory usage in C and C++ programs.
I have built the toolchain completely
2.) In the libexec/gcc directory, I found only the riscv64-unknown-linux-gnu folder. Typically, there are two directories: riscv64-unknown-linux-gnu and riscv64-unknown-linux-elf. Am I missing a configuration or build step that generates the riscv64-unknown-linux-elf directory?

TommyMurphyTM1234, 2025-02-13:
> 2.) In the libexec/gcc directory, I found only the riscv64-unknown-linux-gnu folder. Typically, there are two directories: riscv64-unknown-linux-gnu and riscv64-unknown-linux-elf. Am I missing a configuration or build step that generates the riscv64-unknown-linux-elf directory?

I don't think that there's anything wrong here?
Using the latest CI build...

- https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2025.01.20
```
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2025.01.20/riscv64-glibc-ubuntu-24.04-gcc-nightly-2025.01.20-nightly.tar.xz
tar xvf riscv64-glibc-ubuntu-24.04-gcc-nightly-2025.01.20-nightly.tar.xz
tree riscv/libexec/
riscv/libexec/
└── gcc
    └── riscv64-unknown-linux-gnu
        └── 14.2.0
            ├── cc1
            ├── cc1plus
            ├── collect2
            ├── f951
            ├── g++-mapper-server
            ├── install-tools
            │   ├── fixinc.sh
            │   ├── fixincl
            │   ├── mkheaders
            │   └── mkinstalldirs
            ├── liblto_plugin.la
            ├── liblto_plugin.so
            ├── lto-wrapper
            ├── lto1
            └── plugin
                └── gengtype
```
BTW - FWIW I did a build as follows:
```
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer
make linux 2>&1 | tee build.log
./installed-tools/bin/riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=./installed-tools/bin/riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1666/installed-tools/libexec/gcc/riscv64-unknown-linux-gnu/14.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/user/issue-1666/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/home/user/issue-1666/installed-tools --with-sysroot=/home/user/issue-1666/installed-tools/sysroot --with-pkgversion=g04696df0963 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --enable-libsanitizer --disable-nls --disable-bootstrap --src=.././gcc --disable-default-pie --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 14.2.0 (g04696df0963)
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/18785814/build.log.zip)

```

TommyMurphyTM1234, 2025-02-13:
> 1.Is it possible enable MSAN in RISCV(in latest release)

> 1.) Memory Sanitizer (MSan) is an efficient runtime tool designed to detect uninitialized memory usage in C and C++ programs.
> I have built the toolchain completely

I still don't undestand. Is MSAN part of libsanitizer or something else completely?

TommyMurphyTM1234, 2025-02-18:
Any update @manivalaguru?

manivalaguru, 2025-02-20:
Hi,
MSAN,ASAN and TSAN are part of libsanitisers but i think MSAN is not full supported for RISCV architecture i can able to build properly with ASAN.

i have another question:
while building this binary is not genrated riscv64-unknown-elf-g++ do i have pass and other flag to get that binary?

TommyMurphyTM1234, 2025-02-20:
> Hi, MSAN,ASAN and TSAN are part of libsanitisers but i think MSAN is not full supported for RISCV architecture i can able to build properly with ASAN.

You'll have to ask about that in the upstream (libsanitizer?) project.

> i have another question: while building this binary is not genrated riscv64-unknown-elf-g++ do i have pass and other flag to get that binary?

It is generated for me when I do a bare-metal toolchain build. If it's not for you then you'll need to explain clearly how you're cloning, configuring and building the toolchain and capture, zip and upload the build log and maybe a listing of the binaries in your `<prefix-dir>/bin` directory.

manivalaguru, 2025-02-20:
Hi this is how i built the toolchain

$git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
$cd riscv-gnu-toolchain
$./configure --prefix=/home/econsys/Downloads/libasan/riscv-gnu-toolchain --enable-libsanitizer --with-arch=rv64imafdc --with-abi=lp64d --with-cmodel=medany
$Make

"This successfully builds the sanitizers, but it does not generate riscv64-unknown-elf-g++. I did not modify Makefile.in,. Could you help me understand why riscv64-unknown-elf-g++ is not being generated? Below is the list of binaries in my build."

riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-c++
riscv64-unknown-linux-gnu-c++filt
riscv64-unknown-linux-gnu-cpp
riscv64-unknown-linux-gnu-elfedit
riscv64-unknown-linux-gnu-g++
riscv64-unknown-linux-gnu-gcc
riscv64-unknown-linux-gnu-gcc-14.2.0
riscv64-unknown-linux-gnu-gcc-ar
riscv64-unknown-linux-gnu-gcc-nm
riscv64-unknown-linux-gnu-gcc-ranlib
riscv64-unknown-linux-gnu-gcov
riscv64-unknown-linux-gnu-gcov-dump
riscv64-unknown-linux-gnu-gcov-tool
riscv64-unknown-linux-gnu-gdb
riscv64-unknown-linux-gnu-gdb-add-index
riscv64-unknown-linux-gnu-gfortran
riscv64-unknown-linux-gnu-gp-archive
riscv64-unknown-linux-gnu-gp-collect-app
riscv64-unknown-linux-gnu-gp-display-html
riscv64-unknown-linux-gnu-gp-display-src
riscv64-unknown-linux-gnu-gp-display-text
riscv64-unknown-linux-gnu-gprof
riscv64-unknown-linux-gnu-gprofng
riscv64-unknown-linux-gnu-ld
riscv64-unknown-linux-gnu-ld.bfd
riscv64-unknown-linux-gnu-lto-dump
riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-run
riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-strip

TommyMurphyTM1234, 2025-02-20:
A few comments:

1. The `bin` directory listing in your post above shows the binarties generated for a build ot the Linux toolchain using, for example, `configure ... --enable-linux && make` or `configure ... && make linux` etc.
2. If you build the Linux toolchain then you are not going to get `riscv64-unknown-elf-*` binaries because they are only generated for a bare-metal toolchain build.
3. The instructions that you give at the top are for building the bare-metal/Newlib toolchain and do not match the results of the directory listing which are the Linux toolchain binaries - `make` with no other option builds the bare-metal toolchain by default.
4. As I have stated a few times already `--enable-libsanitizer` will not work for the bare-metal toolchain and the build will fail (the only way to exercise this right now is to manually edit the `Makefile[.in]` file.
5. If you are having toolchain build issues or confusion then you need to capture, zip and upload the build log as I have already explained.

manivalaguru, 2025-02-20:
Hi ,
Is this the bare metal tool chain build command if yes I have tired this too but I did not got riscv64-unknown-elf-g++
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer
make linux

If this is not the bare metal tool chain command kindly provide us the command I will try that and let you knoe

manivalaguru, 2025-02-20:
Kindly ignore libsanitisers for now first I will try to build for riscv in my x86 machine and then will try to add sanitisers

TommyMurphyTM1234, 2025-02-20:
> Hi , Is this the bare metal tool chain build command if yes I have tired this too but I did not got riscv64-unknown-elf-g++ git clone https://github.com/riscv-collab/riscv-gnu-toolchain cd riscv-gnu-toolchain ./configure --prefix=`pwd`/installed-tools --enable-libsanitizer make linux

No - obviously not. Why do you think that `make linux` would make the bare-metal toolchain?

> If this is not the bare metal tool chain command kindly provide us the command I will try that and let you knoe

It's all already covered by the README:

- Linux toolchain: https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#installation-linux
- Bare-metal toolchain: https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#installation-newlib

manivalaguru, 2025-02-20:
Then this is the right command right that's what i thought 
$git clone --recursive https://github.com/riscv/riscv-gnu-toolchain
$cd riscv-gnu-toolchain
$./configure --prefix=/home/econsys/Downloads/libasan/riscv-gnu-toolchain --enable-libsanitizer --with-arch=rv64imafdc --with-abi=lp64d --with-cmodel=medany
$Make
using this only i have got this 
riscv64-unknown-linux-gnu-addr2line
riscv64-unknown-linux-gnu-ar
riscv64-unknown-linux-gnu-as
riscv64-unknown-linux-gnu-c++
riscv64-unknown-linux-gnu-c++filt

i will try this again and let you know

manivalaguru, 2025-02-21:
I believe there might have been a misunderstanding. The following command sequence worked for me:

git clone --recursive https://github.com/riscv/riscv-gnu-toolchain  
cd riscv-gnu-toolchain  
./configure --prefix=/home/econsys/Downloads/libasan/riscv-gnu-toolchain \
    --enable-libsanitizer --with-arch=rv64imafdc --with-abi=lp64d --with-cmodel=medany  
make  

After running this, I successfully built the sanitizer libraries, including libasan_preinit.o and libtsan_preinit.o. Here’s the file output for verification:

file libasan_preinit.o  
libasan_preinit.o: ELF 64-bit LSB relocatable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), not stripped  

file libtsan_preinit.o  
libtsan_preinit.o: ELF 64-bit LSB relocatable, UCB RISC-V, RVC, double-float ABI, version 1 (SYSV), not stripped  

As you can see, these libraries were correctly built for RISC-V that why i dont understand this
[As I have stated a few times already --enable-libsanitizer will not work for the bare-metal toolchain. By default the Makefile[.in] only passes it to the Linux toolchain build and not to the bare-metal toolchain build. I simply edited the Makefile[.in] to see what would happen when it was passed to the latter. It fails/breaks so, right now, it cannot be used for the bare-metal toolchain build and I speculated that maybe the libsanitizer stuff has some dependency on OS capabilities/system calls or something like that so cannot be used in a bare-metal environment (at least without further work)?].

Additionally, here is the list of generated ELF binaries in my build:

riscv64-unknown-elf-addr2line  riscv64-unknown-elf-elfedit  riscv64-unknown-elf-nm       riscv64-unknown-elf-readelf  
riscv64-unknown-elf-ar         riscv64-unknown-elf-gprof    riscv64-unknown-elf-objcopy  riscv64-unknown-elf-size  
riscv64-unknown-elf-as         riscv64-unknown-elf-ld       riscv64-unknown-elf-objdump  riscv64-unknown-elf-strings  
riscv64-unknown-elf-c++filt    riscv64-unknown-elf-ld.bfd   riscv64-unknown-elf-ranlib   riscv64-unknown-elf-strip  

However, I noticed that riscv64-unknown-elf-gcc and riscv64-unknown-elf-g++ are missing, which I need for my BSP compilation. Could you help me understand why these are not being generated?

TommyMurphyTM1234, 2025-02-21:
You seem to be ignoring the advice that I've given you on how to build the two variants of the toolchain, what results to expect, and the fact that the bare-metal toolchain build simply cannot build the libsanitizer stuff. 

There is simply no way that a successful build of the bare-metal toolchain build can be building the libsanitizer stuff or is not building some of the `riscv64-unknown-elf-*` binaries.

If you want assistance then, as I've said several times, you need to be much clearer about what exact commands that you're using and to capture build log for the toolchain build **from scratch** and also detail any other commands that you are using (e.g. you seem to be suggesting that you are building libsanitizer stuff separately after building the bare-metal toolchain but you don't clarify exactly what you're doing).

E.g. you need to at least capture the build log for this: 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain
# Workaround for shallow clone problem: 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in
./configure ...

# Linux toolchain 
make distclean
rm -rf <prefix-dir>
make linux 2>&1 | tee build.log

# or... bare-metal toolchain 
make distclean
rm -rf <prefix-dir>
make 2>&1 | tee build.log
```
And also clarify what other separate build steps/commands that you're using and capture logs for those too.

TommyMurphyTM1234, 2025-02-21:
For clarity's sake... here are the results of my clean/from scratch build of the bare-metal toolchain:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1666
cd issue-1666
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer --with-arch=rv64imafdc --with-abi=lp64d --with-cmodel=medany
make 2>&1 | tee build.log

# Note that riscv64-unknown-elf-gcc and riscv64-unknown-elf-g++ are present and correct
ls installed-tools/bin/
riscv64-unknown-elf-addr2line  riscv64-unknown-elf-elfedit     riscv64-unknown-elf-gcc-ranlib     riscv64-unknown-elf-gprof     riscv64-unknown-elf-objdump  riscv64-unknown-elf-strip
riscv64-unknown-elf-ar         riscv64-unknown-elf-g++         riscv64-unknown-elf-gcov           riscv64-unknown-elf-ld        riscv64-unknown-elf-ranlib
riscv64-unknown-elf-as         riscv64-unknown-elf-gcc         riscv64-unknown-elf-gcov-dump      riscv64-unknown-elf-ld.bfd    riscv64-unknown-elf-readelf
riscv64-unknown-elf-c++        riscv64-unknown-elf-gcc-14.2.0  riscv64-unknown-elf-gcov-tool      riscv64-unknown-elf-lto-dump  riscv64-unknown-elf-run
riscv64-unknown-elf-c++filt    riscv64-unknown-elf-gcc-ar      riscv64-unknown-elf-gdb            riscv64-unknown-elf-nm        riscv64-unknown-elf-size
riscv64-unknown-elf-cpp        riscv64-unknown-elf-gcc-nm      riscv64-unknown-elf-gdb-add-index  riscv64-unknown-elf-objcopy   riscv64-unknown-elf-strings

# Note that no libsanitizer stuff is built
find installed-tools -name "*san*"
installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/plugin/include/asan.h
installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/plugin/include/sanitizer.def
installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/plugin/include/tsan.h
installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/plugin/include/tree-ssanames.h
installed-tools/lib/gcc/riscv64-unknown-elf/14.2.0/plugin/include/ubsan.h
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/18903785/build.log.zip)

TommyMurphyTM1234, 2025-02-21:
And for the Linux toolchain build...
```
# Clean up after previous bare-metal toolchain build
make distclean
rm -rf installed-tools

make linux 2>&1 | tee build.log

ls installed-tools/bin
riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc         riscv64-unknown-linux-gnu-gdb              riscv64-unknown-linux-gnu-gprof     riscv64-unknown-linux-gnu-ranlib
riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gcc-14.2.0  riscv64-unknown-linux-gnu-gdb-add-index    riscv64-unknown-linux-gnu-gprofng   riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-gcc-ar      riscv64-unknown-linux-gnu-gfortran         riscv64-unknown-linux-gnu-ld        riscv64-unknown-linux-gnu-run
riscv64-unknown-linux-gnu-c++        riscv64-unknown-linux-gnu-gcc-nm      riscv64-unknown-linux-gnu-gp-archive       riscv64-unknown-linux-gnu-ld.bfd    riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-gcc-ranlib  riscv64-unknown-linux-gnu-gp-collect-app   riscv64-unknown-linux-gnu-lto-dump  riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-cpp        riscv64-unknown-linux-gnu-gcov        riscv64-unknown-linux-gnu-gp-display-html  riscv64-unknown-linux-gnu-nm        riscv64-unknown-linux-gnu-strip
riscv64-unknown-linux-gnu-elfedit    riscv64-unknown-linux-gnu-gcov-dump   riscv64-unknown-linux-gnu-gp-display-src   riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-g++        riscv64-unknown-linux-gnu-gcov-tool   riscv64-unknown-linux-gnu-gp-display-text  riscv64-unknown-linux-gnu-objdump

find installed-tools -name "*san*"
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/plugin/include/asan.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/plugin/include/sanitizer.def
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/plugin/include/tsan.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/plugin/include/tree-ssanames.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/plugin/include/ubsan.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/include/sanitizer
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/include/sanitizer/tsan_interface.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/include/sanitizer/hwasan_interface.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/include/sanitizer/lsan_interface.h
installed-tools/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/include/sanitizer/asan_interface.h
installed-tools/sysroot/lib/liblsan.so.0
installed-tools/sysroot/lib/liblsan.so.0.0.0
installed-tools/sysroot/lib/libtsan.so.2.0.0
installed-tools/sysroot/lib/libsanitizer.spec
installed-tools/sysroot/lib/liblsan.la
installed-tools/sysroot/lib/liblsan_preinit.o
installed-tools/sysroot/lib/libubsan.so.1
installed-tools/sysroot/lib/libtsan.la
installed-tools/sysroot/lib/liblsan.a
installed-tools/sysroot/lib/libasan.so.8
installed-tools/sysroot/lib/liblsan.so
installed-tools/sysroot/lib/libasan.so
installed-tools/sysroot/lib/libubsan.la
installed-tools/sysroot/lib/libtsan_preinit.o
installed-tools/sysroot/lib/libasan.a
installed-tools/sysroot/lib/libasan.la
installed-tools/sysroot/lib/libasan.so.8.0.0
installed-tools/sysroot/lib/libubsan.a
installed-tools/sysroot/lib/libtsan.so
installed-tools/sysroot/lib/libubsan.so
installed-tools/sysroot/lib/libubsan.so.1.0.0
installed-tools/sysroot/lib/libtsan.a
installed-tools/sysroot/lib/libasan_preinit.o
installed-tools/sysroot/lib/libtsan.so.2
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan.so.0
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan.so.0.0.0
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan.so.2.0.0
installed-tools/riscv64-unknown-linux-gnu/lib/libsanitizer.spec
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan.la
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan_preinit.o
installed-tools/riscv64-unknown-linux-gnu/lib/libubsan.so.1
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan.la
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan.a
installed-tools/riscv64-unknown-linux-gnu/lib/libasan.so.8
installed-tools/riscv64-unknown-linux-gnu/lib/liblsan.so
installed-tools/riscv64-unknown-linux-gnu/lib/libasan.so
installed-tools/riscv64-unknown-linux-gnu/lib/libubsan.la
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan_preinit.o
installed-tools/riscv64-unknown-linux-gnu/lib/libasan.a
installed-tools/riscv64-unknown-linux-gnu/lib/libasan.la
installed-tools/riscv64-unknown-linux-gnu/lib/libasan.so.8.0.0
installed-tools/riscv64-unknown-linux-gnu/lib/libubsan.a
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan.so
installed-tools/riscv64-unknown-linux-gnu/lib/libubsan.so
installed-tools/riscv64-unknown-linux-gnu/lib/libubsan.so.1.0.0
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan.a
installed-tools/riscv64-unknown-linux-gnu/lib/libasan_preinit.o
installed-tools/riscv64-unknown-linux-gnu/lib/libtsan.so.2
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/18904748/build.log.zip)

TommyMurphyTM1234, 2025-02-25:
> I initially used the bare-metal toolchain, which is intended for systems without an operating system. However, the Linux toolchain is necessary for building the kernel and U-Boot.

If you need to build both into the same prefix directory then do this: 
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1666
cd issue-1666
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in
./configure --prefix=`pwd`/installed-tools --enable-libsanitizer --with-arch=rv64imafdc --with-abi=lp64d --with-cmodel=medany
make 2>&1 | tee build-newlib.log
make linux 2>&1 | tee build-linux.log
```
> Additionally, I successfully integrated AddressSanitizer (ASan) for both the kernel and U-Boot.

What do you mean? 
What exactly did you do?
I presume that you didn't manage to enable libsanitizer for the bare-metal Newlib toolchain?

Can the issue be closed?

TommyMurphyTM1234, 2025-02-28:
> Can the issue be closed?

I'm assuming so.

closed 2025-02-28 by TommyMurphyTM1234