riscv-collab/riscv-gnu-toolchain #1667: Compiler error in gcc

micsolutions, 2025-02-11:
Declaring an array with auto using constexpr will result in an internal compiler error when using `riscv32-unknown-elf-g++`:

```cpp
  // 1) No problems:
  constexpr std::array<float, 3> test1 = {1.0f, 2.0f, 3.0f};

  // 2) internal compiler error
  constexpr auto test2 = std::array<float, 3>{1.0f, 2.0f, 3.0f};
```

Here is the full error message:

```
internal compiler error: in reshape_init_r, at cp/decl.c:6043
   constexpr auto test2 = std::array<float, 3>{1.0f, 2.0f, 3.0f};
                  ^~~~~
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://gcc.gnu.org/bugs/> for instructions.
```

micsolutions, 2025-02-11:
Another possibly related bug:

```cpp
const auto x = std::array{1.0f, 2.0f, 3.0f};
const auto y = std::array{4.0f, 5.0f, 6.0f};

// error: the value of 'x' is not usable in a constant expression
static_assert(x.size() == y.size());

// Works fine by itself:
constexpr auto xSize = x.size();
constexpr auto xSize = x.size();

// error: the value of 'xSize' is not usable in a constant expression
static_assert(xSize == ySize);
```

cmuellner, 2025-02-11:
I tried the following:
```
```

TommyMurphyTM1234, 2025-02-11:
> Testing with CE shows:

Just to clarify, @cmuellner means Compiler Explorer:

- https://godbolt.org/

TommyMurphyTM1234, 2025-02-11:
> Regarding your second example, this might be an issue with your code. GCC trunk and 14.2.0 states:
> 
> ```
> <source>:17:16: error: conflicting declaration 'constexpr const auto xSize'
>    17 | constexpr auto xSize = x.size();
>       |                ^~~~~
> <source>:16:16: note: previous declaration as 'constexpr const unsigned int xSize'
>    16 | constexpr auto xSize = x.size();
>       |                ^~~~~
> <source>:20:24: error: 'ySize' was not declared in this scope; did you mean 'xSize'?
>    20 | static_assert(xSize == ySize);
>       |                        ^~~~~
>       |                        xSize
> Compiler returned: 1
> ```
> 
> I'm not C++ language lawyer enough to know if C++ allows multiple declarations like this (I would not expect that):
> 
> ```
> constexpr auto xSize = x.size();
> constexpr auto xSize = x.size();
> ```

I think that this is simply a typo and...

> Another possibly related bug:
> ```
> const auto x = std::array{1.0f, 2.0f, 3.0f};
> const auto y = std::array{4.0f, 5.0f, 6.0f};
> 
> // error: the value of 'x' is not usable in a constant expression
> static_assert(x.size() == y.size());
> 
> // Works fine by itself:
> constexpr auto xSize = x.size();
> constexpr auto xSize = x.size();
> 
> // error: the value of 'xSize' is not usable in a constant expression
> static_assert(xSize == ySize);
> ```

should actually be:

> ```
> // Works fine by itself:
> constexpr auto xSize = x.size();
> constexpr auto ySize = y.size();
> ```

fshstk, 2025-02-11:
Apologies, it turns out that I actually filed this issue in the wrong repository by mistake...

We are in fact working with a rather outdated fork of this repo and I must have had the wrong tab active when I clicked the new issue button.

Because of this, and since you were unable to reproduce the behaviour on your end, I am going to assume this is a downstream problem. Please feel free to close this issue if you agree.

> I think that this is simply a typo and...
>
>
> should actually be:

Correct. I re-typed the example rather than copy-pasting it. The code I fed into the compiler was okay.

*- Fabian*

micsolutions, 2025-02-11:
Apologies, it turns out that I actually filed this issue in the wrong repository by mistake...

We are in fact working with a rather outdated fork of this repo and I must have had the wrong tab active when I clicked the new issue button.

Because of this, and since you were unable to reproduce the behaviour on your end, I am going to assume this is a downstream problem and am closing the issue.

> I think that this is simply a typo and...
>
>
> should actually be:

Correct. I re-typed the example rather than copy-pasting it. The code I fed into the compiler was okay.

*- Fabian*

closed 2025-02-11 by micsolutions

TommyMurphyTM1234, 2025-02-11:
> We are in fact working with a rather outdated fork of this repo

There have been so many changes, enhancements and fixes to the RISC-V specific and general support in GCC and all related tools in the past few years that it really doesn't make sense to use old versions of the toolchain if at all possible. Much better to use a more recent version of the tools built from the main branch of this repo, the upstream sources, using the likes of [crosstool-NG](https://crosstool-ng.github.io/) if necessary, or even a binary distribution such as @ilg-ul's excellent [xPack version](https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/) or simply the RISC-V tools provided by your preferred OS distro's software repositories.