riscv-collab/riscv-gnu-toolchain #1669: git submodule update fails

nfraly, 2025-02-13:
This might be a re-opening of an old issue, or it may be new, not sure. Failing to grab submodules during make, and submodule update is failing as well. I have a local copy from last night that worked to use, but just wanted to bring this issue forward so it can get some attention (first time posting an issue to a big repo like this, sorry for any faux pas).

TommyMurphyTM1234, 2025-02-13:
See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1654#issuecomment-2591195261

This works fine for me so try not doing any recursive clone manually:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1669
cd issue-1669
./configure ...
make 
```

nfraly, 2025-02-13:
Disabling the shallow cloning appears to have worked. Thank you!!

jleidel, 2025-02-13:
Also confirming that this persists.  Following @TommyMurphyTM1234's instructions from above does work temporarily.

TommyMurphyTM1234, 2025-02-13:
Thanks for the updates. As I mentioned, @mickflemm explains the issue here 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1654#issuecomment-2591195261

but, unfortunately, at the moment I don't really know what specific changes are needed in order to prepare a PR that fixes the problem.

I presume that the CI builds will also fail next time that they run to create a release...

hpretl, 2025-02-13:
Here the sed's for the above WA:
```
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in
```

TommyMurphyTM1234, 2025-02-24:
> But I did not get any .gitmodule file when I cloned the github repository, should I just make the changes in the Makefile.in to overcome the failing of submodules?

No. You need to ensure that the `.gitmodules` file has been fetched by the clone operation and then edit that and the `Makefile.in` before running `configure`. This post outlines how to use `sed` to make the necessary changes to circumvent the shallow clone problem for now:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585

Note that files whose names start with a dot/period/'.' are often not visible by default to perhaps it's there but you're just not seeing it? Try `ls -a` in a terminal shell to see if it's actually there. If it's not then you'll have to try cloning afresh.

aWZHY0yQH81uOYvH, 2025-02-25:
The workaround (removing `shallow = true` and `--depth 1`) does not work for me. Even a standard `git submodule update --init` fails with
```
error: Server does not allow request for unadvertised object beb2cdbcda911764b2bed5e57921fe90493260bd
fatal: Fetched in submodule path 'binutils', but it did not contain beb2cdbcda911764b2bed5e57921fe90493260bd. Direct fetching of that commit failed.
```

Awab-Ahsan, 2025-02-25:
I am trying to setup RISCV GCC toolchain on redhat. I have succesffuly set up it in ubuntu along with spike and proxyt kernel. But now with redhat I am facing issues with respect to cloning of the git.
I have followed both traditional way and the ways mentioned above both are not working for me.
Traditional
`git clone https://github.com/riscv/riscv-gnu-toolchain
Cloning into 'riscv-gnu-toolchain'...
remote: Enumerating objects: 9537, done.
remote: Counting objects: 100% (428/428), done.
remote: Compressing objects: 100% (106/106), done.
remote: Total 9537 (delta 367), reused 322 (delta 322), pack-reused 9109 (from 3)
Receiving objects: 100% (9537/9537), 6.52 MiB | 1.10 MiB/s, done.
Resolving deltas: 100% (4747/4747), done.
../configure --prefix=$RISC_V --with-arch=rv32gv --with-abi=ilp32
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
make
cd /home/advancedresearch/Downloads/riscv-gnu-toolchain/build/.. && \
flock `git rev-parse --git-dir`/config git submodule init /home/advancedresearch/Downloads/riscv-gnu-toolchain/build/../gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /home/advancedresearch/Downloads/riscv-gnu-toolchain/build/../gcc/
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Cloning into '/home/advancedresearch/Downloads/riscv-gnu-toolchain/gcc'...
remote: Enumerating objects: 146242, done.        
remote: Counting objects: 100% (146242/146242), done.        
remote: Compressing objects: 100% (112586/112586), done.        
remote: Total 146242 (delta 37682), reused 90993 (delta 31629), pack-reused 0        
Receiving objects: 100% (146242/146242), 175.63 MiB | 808.00 KiB/s, done.
Resolving deltas: 100% (37682/37682), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: Server does not allow request for unadvertised object 04696df09633baf97cdbbdd6e9929b9d472161d3
fatal: Fetched in submodule path 'gcc', but it did not contain 04696df09633baf97cdbbdd6e9929b9d472161d3. Direct fetching of that commit failed.
make: *** [Makefile:348: /home/advancedresearch/Downloads/riscv-gnu-toolchain/build/../gcc/.git] Error 128
`

Now if i follow the way told above it still fails
[advancedresearch@advancedresearch Downloads]$ git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1669
Cloning into 'issue-1669'...
remote: Enumerating objects: 9537, done.
remote: Counting objects: 100% (427/427), done.
remote: Compressing objects: 100% (105/105), done.
remote: Total 9537 (delta 366), reused 322 (delta 322), pack-reused 9110 (from 3)
Receiving objects: 100% (9537/9537), 6.50 MiB | 1.09 MiB/s, done.
Resolving deltas: 100% (4747/4747), done.
[advancedresearch@advancedresearch Downloads]$ cd issue-1669/
[advancedresearch@advancedresearch issue-1669]$ ./configure --prefix=$RISC_V --with-arch=rv32gv --with-abi=ilp32
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
[... 9 lines cut ...]
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
[advancedresearch@advancedresearch issue-1669]$ make
cd /home/advancedresearch/Downloads/issue-1669 && \
flock `git rev-parse --git-dir`/config git submodule init /home/advancedresearch/Downloads/issue-1669/gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /home/advancedresearch/Downloads/issue-1669/gcc/
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Cloning into '/home/advancedresearch/Downloads/issue-1669/gcc'...
remote: Enumerating objects: 146242, done.        
remote: Counting objects: 100% (146242/146242), done.        
remote: Compressing objects: 100% (112586/112586), done.        
remote: Total 146242 (delta 37681), reused 90993 (delta 31629), pack-reused 0        
Receiving objects: 100% (146242/146242), 175.63 MiB | 499.00 KiB/s, done.
Resolving deltas: 100% (37681/37681), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: Server does not allow request for unadvertised object 04696df09633baf97cdbbdd6e9929b9d472161d3
fatal: Fetched in submodule path 'gcc', but it did not contain 04696df09633baf97cdbbdd6e9929b9d472161d3. Direct fetching of that commit failed.
make: *** [Makefile:348: /home/advancedresearch/Downloads/issue-1669/gcc/.git] Error 128
Please suggest me what to do

TommyMurphyTM1234, 2025-02-25:
> Please suggest me what to do

Please read my earlier post: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189

```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1669

cd issue-1669

sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

./configure --prefix=`pwd`/installed-tools ...

make
```

TVS2004Chaitanya, 2025-02-25:
Dear Sir,

I have a small doubt, I installed the toolchain with vector extension with the command:
./configure --prefix=/home/chaitanya/riscv_vector --enable-multilib --with-arch=rv64gcv --with-abi=lp64d
I am trying to do simple vector addition in C. This is the C code
#include <stdio.h>
#include <riscv_vector.h>

void vector_add(const int *a, const int *b, int *c, size_t n) {
    size_t vl;
    for (size_t i = 0; i < n; i += vl) {
        vl = vsetvl_e32m1(n - i); // Set vector length for remaining elements

        vint32m1_t vec_a = vle32_v_i32m1(&a[i], vl); // Load vector from a[]
        vint32m1_t vec_b = vle32_v_i32m1(&b[i], vl); // Load vector from b[]

        vint32m1_t vec_c = vadd_vv_i32m1(vec_a, vec_b, vl); // Perform vector addition

        vse32_v_i32m1(&c[i], vec_c, vl); // Store result to c[]
    }
}

int main() {
    const size_t N = 8; // Number of elements
    int a[N] = {1, 2, 3, 4, 5, 6, 7, 8};
    int b[N] = {8, 7, 6, 5, 4, 3, 2, 1};
    int c[N] = {0}; // Output array

    vector_add(a, b, c, N);

    // Print results
    printf("Result: ");
    for (size_t i = 0; i < N; i++) {
        printf("%d ", c[i]);
    }
    printf("\n");

    return 0;
}

I am getting the following error messages
***@***.***:~/Desktop/RISCV_PE_6thsem$ riscv64-unknown-elf-gcc -march=rv64gcv -mabi=lp64d vector.c
vector.c: In function 'vector_add':
vector.c:7:14: error: implicit declaration of function 'vsetvl_e32m1' [-Wimplicit-function-declaration]
    7 |         vl = vsetvl_e32m1(n - i); // Set vector length for remaining elements
      |              ^~~~~~~~~~~~
vector.c:9:28: error: implicit declaration of function 'vle32_v_i32m1' [-Wimplicit-function-declaration]
    9 |         vint32m1_t vec_a = vle32_v_i32m1(&a[i], vl); // Load vector from a[]
      |                            ^~~~~~~~~~~~~
vector.c:9:28: error: incompatible types when initializing type 'vint32m1_t' using type 'int'
vector.c:10:28: error: incompatible types when initializing type 'vint32m1_t' using type 'int'
   10 |         vint32m1_t vec_b = vle32_v_i32m1(&b[i], vl); // Load vector from b[]
      |                            ^~~~~~~~~~~~~
vector.c:12:28: error: implicit declaration of function 'vadd_vv_i32m1' [-Wimplicit-function-declaration]
   12 |         vint32m1_t vec_c = vadd_vv_i32m1(vec_a, vec_b, vl); // Perform vector addition
      |                            ^~~~~~~~~~~~~
vector.c:12:28: error: incompatible types when initializing type 'vint32m1_t' using type 'int'
vector.c:14:9: error: implicit declaration of function 'vse32_v_i32m1' [-Wimplicit-function-declaration]
   14 |         vse32_v_i32m1(&c[i], vec_c, vl); // Store result to c[]
      |         ^~~~~~~~~~~~~
vector.c: In function 'main':
vector.c:20:16: error: variable-sized object may not be initialized except with an empty initializer
   20 |     int a[N] = {1, 2, 3, 4, 5, 6, 7, 8};
      |                ^
vector.c:21:16: error: variable-sized object may not be initialized except with an empty initializer
   21 |     int b[N] = {8, 7, 6, 5, 4, 3, 2, 1};
      |                ^
vector.c:22:16: error: variable-sized object may not be initialized except with an empty initializer
   22 |     int c[N] = {0}; // Output array
      |                ^

My compiler is not able to recognize the vector instrinsics. Can you please tell me where I am going wrong.

ThankYou,
TVS Chaitanya.

  *   #1669 (comment)<https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189>

  *   #1669 (comment)<https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189>

Awab-Ahsan, 2025-02-25:
> > Now if i follow the way told above it still fails
> 
> You haven't done anything different in the second attempt that I can see so the results/failure are the same.
> 
> > Please suggest me what to do
> 
> Please read my earlier post:
> 
> * [git submodule update fails #1669 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189)
> 
> In summary - to workaround the "shallow clone" problem for now do this:
> 
> ```
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1669
> 
> cd issue-1669
> 
> # Work around for "shallow clone" problem thanks to @hpretl 
> # https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
> sed -i '/shallow = true/d' .gitmodules
> sed -i 's/--depth 1//g' Makefile.in
> 
> ./configure ...
> 
> make
> ```

gcc is started building by using this method, it is taking more time than usual but so far going great. Will update if it resolves the issue. Thanks alot

Awab-Ahsan, 2025-02-25:
> Dear Sir,
>
> I have a small doubt, I installed the toolchain with vector extension with the command:
> ./configure --prefix=/home/chaitanya/riscv_vector --enable-multilib --with-arch=rv64gcv [...]

check the intrinsic functions github repo they have changed the implementation have to place __riscv_ before each vector command also the syntax of some instructions got changed. Here is the link [https://github.com/riscv-non-isa/rvv-intrinsic-doc/blob/main/auto-generated/overloaded_intrinsic_funcs/02_vector_integer_arithmetic_intrinsics.adoc](url)

TommyMurphyTM1234, 2025-02-25:
@TVS2004Chaitanya - please do not post off topic additional issues about vector support in this thread. If necessary open a new issue. I have deleted the off topic posts.

TommyMurphyTM1234, 2025-02-25:
> The workaround (removing `shallow = true` and `--depth 1`) does not work for me. Even a standard `git submodule update --init` fails with ...

Please read my earlier post:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189

In summary - to workaround the "shallow clone" problem for now do this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1669

cd issue-1669

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

./configure ...

make
```

aWZHY0yQH81uOYvH, 2025-02-25:
> > The workaround (removing `shallow = true` and `--depth 1`) does not work for me. Even a standard `git submodule update --init` fails with ...
> 
> Please read my earlier post:
> 
> * [git submodule update fails #1669 (comment)](https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2678800189)

It's worth noting that it seems like you have to delete and re-clone the entire repo. Yesterday I was following these same steps but just deleted the broken submodules which did not allow the workaround to work.

TommyMurphyTM1234, 2025-02-26:
> It's worth noting that it seems like you have to delete and re-clone the entire repo.

Yes, that's why my instructions above show a clean/new clone of the repo rather than trying to use the workaround on the results of a previously failed clone. 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2682013720

TommyMurphyTM1234, 2025-04-30:
Fixed by this PR:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1702

closed 2025-04-30 by TommyMurphyTM1234