riscv-collab/riscv-gnu-toolchain #1670: Relaxation is breaking PIE code in some cases

ZhekaS, 2025-02-13:
Consider the simple code in `main.c`:
```c
volatile unsigned temp;
int main(void) {
    temp = 0x12345678;
    return 0;
}

```
and the following linker sctipt `link.ld`:

```ld
ENTRY(main)

SECTIONS {
  .text : {
    *(.text .text.*)
  }
  .rodata : ALIGN(16) {
    *(.srodata .srodata.*)
    *(.rodata .rodata.*)
  }

  .data ALIGN(16): {
    _data_start = .;
    *(.got .got.*)
    *(.sdata .sdata.*)
    *(.data .data.*)
  }
  .bss ALIGN(16): {
    _bss_start = .;
    *(.sbss .sbss.*)
    *(.bss .bss.*)
    _bss_end = .;
  }
}

```

When building with
> riscv64-unknown-elf-gcc -nostdlib -nostartfiles -O3 -fpie -static-pie -T link.ld main.c -o main.elf
and dumping:
> riscv64-unknown-elf-objdump -x -S main.elf 

The following disassembly is produced:
```
0000000000000000 <main>:
   0:   123457b7                lui     a5,0x12345
   4:   67878793                addi    a5,a5,1656 # 12345678 <_bss_end+0x12345664>
   8:   00f02823                sw      a5,16(zero) # 10 <_data_start>
   c:   4501                    li      a0,0
   e:   8082                    ret
```
As we can see the access to `temp` is relative to `zero` instead of `pc`, rendering this code position dependent.

However, if we "move" the section `.bss` section to a higher address by aligning to, say, 2048 in the linker script:
```
...
.bss ALIGN(2048): {
...
```
the result is PC-relative addressing of `temp` (as I would expect):
```
0000000000000000 <main>:
   0:   123457b7                lui     a5,0x12345
   4:   67878793                addi    a5,a5,1656 # 12345678 <_bss_end+0x12344e74>
   8:   00000717                auipc   a4,0x0
   c:   7ef72c23                sw      a5,2040(a4) # 800 <temp>
  10:   4501                    li      a0,0
  12:   8082                    ret
```
But! If we reduce the alignment, as in 
```
...
.bss ALIGN(2000): {
...
```
we see again zero-based addressing:
```
   0:   123457b7                lui     a5,0x12345
   4:   67878793                addi    a5,a5,1656 # 12345678 <_bss_end+0x12344ea4>
   8:   7cf02823                sw      a5,2000(zero) # 7d0 <temp>
   c:   4501                    li      a0,0
   e:   8082                    ret
```
and it appears that as long as the offset of `temp` from the address `0x0` is less than `2048`, it will produce zero-based access.

If we do the same exercise but with `-mno-relax`, all the accesses are PC-relative as I would expect.

For completeness here is the code with relocations as produced by 
> riscv64-unknown-elf-gcc -nostdlib -nostartfiles -O3 -fpie -static-pie -c main.c 

```
0000000000000000 <main>:
   0:   123457b7                lui     a5,0x12345
   4:   67878793                addi    a5,a5,1656 # 12345678 <main+0x12345678>
   8:   00000717                auipc   a4,0x0
                        8: R_RISCV_PCREL_HI20   .LANCHOR0
                        8: R_RISCV_RELAX        *ABS*
   c:   00f72023                sw      a5,0(a4) # 8 <main+0x8>
                        c: R_RISCV_PCREL_LO12_S .L0 
                        c: R_RISCV_RELAX        *ABS*
  10:   4501                    li      a0,0
  12:   8082                    ret
```

I am not an expert in how relaxation and relocations work, but this behavior seems off to me.

TommyMurphyTM1234, 2025-02-14:
If you suspect a code generation/processing issue with GCC, Binutils `as` and/or `ld` then you need to discuss or report this in the relevant upstream repo(s) where maintenance of RISC-V toolchain support is done these days - it's no longer done here.

TommyMurphyTM1234, 2025-02-18:
Any update @ZhekaS?
Did you raise the issues upstream?

ZhekaS, 2025-02-18:
> Any update [@ZhekaS](https://github.com/ZhekaS)? Did you raise the issues upstream?

Yes. Here it is:
https://sourceware.org/bugzilla/show_bug.cgi?id=32718

TommyMurphyTM1234, 2025-03-07:
> > Any update [@ZhekaS](https://github.com/ZhekaS)? Did you raise the issues upstream?
> 
> Yes. Here it is: https://sourceware.org/bugzilla/show_bug.cgi?id=32718

Great - thanks for the update. I'll close this issue so.

closed 2025-03-07 by TommyMurphyTM1234