riscv-collab/riscv-gnu-toolchain #1671: source-code link return 403

zhongchengyong, 2025-02-14:
After git clone, configure, when I execute the 'make' command, it occurs:
`Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Cloning into '/home/zhong/riscv-gnu-toolchain/gcc'...
fatal: unable to access 'https://gcc.gnu.org/git/gcc.git/': The requested URL returned error: 403
fatal: clone of 'https://gcc.gnu.org/git/gcc.git' into submodule path '/home/zhong/riscv-gnu-toolchain/gcc' failed
Failed to clone 'gcc'. Retry scheduled
Cloning into '/home/zhong/riscv-gnu-toolchain/gcc'...
fatal: unable to access 'https://gcc.gnu.org/git/gcc.git/': The requested URL returned error: 403
fatal: clone of 'https://gcc.gnu.org/git/gcc.git' into submodule path '/home/zhong/riscv-gnu-toolchain/gcc' failed
`
How to solve this issue?

TommyMurphyTM1234, 2025-02-14:
Seems like a (presumably transient) issue with the upstream GCC repo and not with this repo. Also, try fetching cloning the GCC repo separately and see what happens.

Note that there is a separate problem with fetching from some upstream repos that may require a workaround to disable shallow cloning for now. 

See here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669

A workaround using `sed` is described here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585

TommyMurphyTM1234, 2025-02-18:
I presume that this was a transient issue and/or related to the known issue with shallow cloning.

closed 2025-02-18 by TommyMurphyTM1234

zhongchengyong, 2025-02-18:
@TommyMurphyTM1234 Sorry for the late reply. For anyone who has same issue, maybe you can try replace the GNU link to github mirror repo, I have done this by the fork:https://github.com/zhongchengyong/riscv-gnu-toolchain