riscv-collab/riscv-gnu-toolchain #1672: Reproducing -Ox (O1, O2, etc.) Behavior with Specific Compiler Options

aaaaala, 2025-02-14:
I'm trying to reproduce the behavior of the -Ox optimization levels (such as -O1, -O2, etc.) using a collection of specific compiler options like `-falign-jumps`. However, even though I use the `-Q --help=optimizers `option to inspect the individual optimizations enabled by -Ox, I'm unable to generate the same output as when using -Ox directly.
Is there a way to correctly combine these individual options to match the -Ox behavior? Additionally, is there any other method to see exactly how the compiler unfolds these optimizations under -Ox?

TommyMurphyTM1234, 2025-02-14:
You should probably ask this question in the upstream GCC mailing lists or similar forum. That's where all GCC maintenance/enhancement development, including of RISC-V support, is done these days. Not in this repo.

- https://gcc.gnu.org/

cmuellner, 2025-02-14:
The individual optimizations are documented here: https://gcc.gnu.org/onlinedocs/gcc/Optimize-Options.html

HTH

aaaaala, 2025-02-15:
I compiled the CoreMark project using the individual optimization options listed under the -O1 tag in the documentation. Afterward, I compared the disassembly of the resulting executable with that of the original -O1 compilation. However, I found that the outputs were not the same.

TommyMurphyTM1234, 2025-02-15:
As I suggested earlier, you'll probably have to check the upstream GCC source code to see how it "translates" `-Ox` into the individual optimisation options and see if there is some difference to using these explicitly together. And, if that doesn't explain it then you'll probably need to ask about this in a GCC mailing list or similar forum.

TommyMurphyTM1234, 2025-02-15:
> As I suggested earlier, you'll probably need to check the upstream GCC source code to see how it "translates" `-Ox` into the individual optimisation option

This seems to be the relevant part of the GCC source code?

- https://github.com/gcc-mirror/gcc/blob/d0d5204afff226ffc5d329b8e4734dda081dd9b4/gcc/opts.cc#L572-L718

aaaaala, 2025-02-15:
That's right. I’ve seen that file, but compiling it in this way isn’t correct.

TommyMurphyTM1234, 2025-02-15:
> That's right. I’ve seen that file, but compiling it in this way isn’t correct.

You'll have to ask in the upstream GCC mailing lists or a similarly appropirate forum so.

aaaaala, 2025-02-15:
Sure, I appreciate your help. Thank you for your answer!

closed 2025-02-15 by aaaaala