riscv-collab/riscv-gnu-toolchain #1675: Compile from code C program into RISC-V Vector instruction assembly code program

Kient9, 2025-02-19:
I have a program which coded by C language and normally in can be compiled into assembly code with instructions like: add, mul,.etc. But now I want those instructions in assembly code must be in Risc-V Vector instruction formats like: vadd.v, vle64.v,.etc. I have tried to use riscv64-unknown-elf-gcc to convert it but it still become normal assembly code with instructions not belong to RISC-V Vector instruction formats. 
My command is: riscv64-unknown-elf-gcc -S -march=rv64gcv -mabi=lp64d -O3 -o my_program.s my_program.c.

How can i do it, please let me know!
Thank you!

TommyMurphyTM1234, 2025-02-19:
The `arch/abi` and optimisation options that you're using should cause vector instructions to be generated if possible and appropriate for the source code being compiled.

Please provide an example piece of source code that you expect to use vector instructions when compiled.

What toolchain are you using? Where did you get it or how did you build it? What version of GCC is it based on?

Kient9, 2025-02-19:
Thank you for replying me!
- The toolchain is riscv-gnu-toolchain that i built from the official source: https://github.com/riscv-collab/riscv-gnu-toolchain#prerequisites
 Here is my image about the toolchain that I have built:
[image]

- the version of GCC in my laptop is gcc (Ubuntu 9.4.0-1ubuntu1~20.04.2) 9.4.0
- My source code in C language is attached as the image: 

[image]

TommyMurphyTM1234, 2025-02-19:
Please don't use screenshots and expect people to transcribe your source code manually. Please provide the example source code in a form that can be easily copied and pasted.

You also haven't clarified what version of GCC the RISC-V toolchain that you're using is based on. Or exactly how you configured and built it.

cmuellner, 2025-02-19:
If you want to see specific vector instructions, then you can use assembler-files, inline-assembly or vector intrinsics.

If you want auto-vectorization (let the compiler vectorize a scalar implementation), you need a program compatible with the SLP vectorizer and the RISC-V backend.
If you are interested in that, then I suggest looking into the GCC regression test suite, which has tons of examples for auto-vectorization in RISC-V (https://github.com/gcc-mirror/gcc/tree/master/gcc/testsuite/gcc.target/riscv/rvv/autovec).

TommyMurphyTM1234, 2025-02-19:
> If you want auto-vectorization (let the compiler vectorize a scalar implementation), you need a program compatible with the SLP vectorizer and the RISC-V backend.

This may also be of relevance? 

- https://gcc.gnu.org/projects/tree-ssa/vectorization.html

TommyMurphyTM1234, 2025-02-19:
This worked for me:

Using the first test program from here (slighty modified to fix the function declaration):

- https://gcc.gnu.org/projects/tree-ssa/vectorization.html#vectorizab

```
/* test.c */
int a[256], b[256], c[256];
void foo (void) {
  int i;

  for (i=0; i<256; i++){
    a[i] = b[i] + c[i];
  }
}
```

```
# Latest CI toolchain build:
# https://github.com/riscv-collab/riscv-gnu-toolchain/releases
# https://github.com/riscv-collab/riscv-gnu-toolchain/releases/tag/2025.01.20
wget https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2025.01.20/riscv64-elf-ubuntu-24.04-gcc-nightly-2025.01.20-nightly.tar.xz
tar xvf riscv64-elf-ubuntu-24.04-gcc-nightly-2025.01.20-nightly.tar.xz

cd riscv/bin
cp .../test.c .

./riscv64-unknown-linux-gnu-gcc -c -march=rv64gcv -O3 test.c
./riscv64-unknown-linux-gnu-objdump -d test.o

test.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <foo>:
   0:   00000737                lui     a4,0x0
   4:   00070713                mv      a4,a4
   8:   000005b7                lui     a1,0x0
   c:   40070513                addi    a0,a4,1024 # 400 <c>
  10:   80058593                addi    a1,a1,-2048 # fffffffffffff800 <.LANCHOR1+0xffffffffffffe800>
  14:   10000693                li      a3,256

0000000000000018 <.L2>:
  18:   0d06f7d7                vsetvli a5,a3,e32,m1,ta,ma
  1c:   0205e087                vle32.v v1,(a1)
  20:   02056107                vle32.v v2,(a0)
  24:   00279613                slli    a2,a5,0x2
  28:   8e9d                    sub     a3,a3,a5
  2a:   95b2                    add     a1,a1,a2
  2c:   9532                    add     a0,a0,a2
  2e:   021100d7                vadd.vv v1,v1,v2
  32:   020760a7                vse32.v v1,(a4)
  36:   9732                    add     a4,a4,a2
  38:   f2e5                    bnez    a3,18 <.L2>
  3a:   8082                    ret
```

Kient9, 2025-02-20:
> Please don't use screenshots and expect people to transcribe your source code manually. Please provide the example source code in a form that can be easily copied and pasted.
> 
> You also haven't clarified what version of GCC the RISC-V toolchain that you're using is based on. Or exactly how you configured and built it. The first screenshot doesn't clarify anything.
> 
> If you mean that your RISC-V toolchain is using GCC 9.4 then that's almost certainly far too old. The current repo uses GCC 14.2 by default.

- I did as these steps for configuring it:
cd riscv-gnu-toolchain
mkdir build && cd build
 ./configure --prefix=/path/riscv-gnu-toolchain/build --with-arch=rv32gc --with-abi=ilp32d
make

- The version of my GCC Risc-V toolchain is 14.2.0, I have checked it.
My C program is:
`
#include <stdlib.h>
#include <stdio.h>
#define R1 2 // number of rows in Matrix-1
#define C1 2 // number of columns in Matrix-1
#define R2 2 // number of rows in Matrix-2
#define C2 2 // number of columns in Matrix-2
 
void multiplyMatrix(int m1[][C1], int m2[][C2], int (*result)[C2])
{
    
 
    //printf("Resultant Matrix is:\n");
 
    for (int i = 0; i < R1; i++) {
        for (int j = 0; j < C2; j++) {
            result[i][j] = 0;
            for (int k = 0; k < C2; k++) {
                result[i][j] += m1[i][k] * m2[k][j];
            }
 
           // printf("%d\t", result[i][j]);
        }
 
        //printf("\n");
    }
}
 
int main()
{
    int m1[R1][C1] = { { 1, 1 }, { 2, 2 } };
 
    int m2[R2][C2] = { { 1, 1 }, { 2, 2 } };
    int result[R1][C2];
 
    multiplyMatrix(m1, m2, result);

    return 0;
}`
After I compiled it by this command: riscv32-unknown-elf-gcc -S -march=rv32imv -mabi=ilp32 -O3 -o main.s main.c
and the assembly code I received:
`	.file	"main.c"
	.option nopic
	.attribute arch, "rv32i2p1_m2p0_f2p2_d2p2_v1p0_zicsr2p0_zve32f1p0_zve32x1p0_zve64d1p0_zve64f1p0_zve64x1p0_zvl128b1p0_zvl32b1p0_zvl64b1p0"
	.attribute unaligned_access, 0
	.attribute stack_align, 16
	.text
	.align	2
	.globl	multiplyMatrix
	.type	multiplyMatrix, @function
multiplyMatrix:
	sw	zero,0(a2)
	lw	a5,0(a1)
	lw	a4,0(a0)
	mul	a4,a4,a5
	sw	a4,0(a2)
	lw	a3,8(a1)
	lw	a5,4(a0)
	sw	zero,4(a2)
	mul	a5,a5,a3
	add	a5,a5,a4
	sw	a5,0(a2)
	lw	a5,4(a1)
	lw	a4,0(a0)
	mul	a4,a4,a5
	sw	a4,4(a2)
	lw	a3,12(a1)
	lw	a5,4(a0)
	sw	zero,8(a2)
	mul	a5,a5,a3
	add	a5,a5,a4
	sw	a5,4(a2)
	lw	a5,0(a1)
	lw	a4,8(a0)
	mul	a4,a4,a5
	sw	a4,8(a2)
	lw	a3,8(a1)
	lw	a5,12(a0)
	sw	zero,12(a2)
	mul	a5,a5,a3
	add	a5,a5,a4
	sw	a5,8(a2)
	lw	a5,4(a1)
	lw	a4,8(a0)
	mul	a4,a4,a5
	sw	a4,12(a2)
	lw	a5,12(a0)
	lw	a3,12(a1)
	mul	a5,a5,a3
	add	a5,a5,a4
	sw	a5,12(a2)
	ret
	.size	multiplyMatrix, .-multiplyMatrix
	.section	.text.startup,"ax",@progbits
	.align	2
	.globl	main
	.type	main, @function
main:
	li	a0,0
	ret
	.size	main, .-main
	.ident	"GCC: (g04696df09-dirty) 14.2.0"
	.section	.note.GNU-stack,"",@progbits`

Totally no RVV instruction format, do i need any conditions for the C program to be compatible with the tool for generating RVV instruction format ?
Thank you!

Kient9, 2025-02-20:
> Try this:
> 
> ```
> #define R1 2 // number of rows in Matrix-1
> #define C1 2 // number of columns in Matrix-1
> #define R2 2 // number of rows in Matrix-2
> #define C2 2 // number of columns in Matrix-2
> 
> int m1[R1][C1];
> int m2[R2][C2];
> int result[R1][C2];
> 
> void multiplyMatrix()
> {
>     for (int i = 0; i < R1; i++) {
>         for (int j = 0; j < C2; j++) {
>             result[i][j] = 0;
>             for (int k = 0; k < C2; k++) {
>                 result[i][j] += m1[i][k] * m2[k][j];
>             }
>         }
>     }
> }
> ```
> 
> This vectorizes fine with `-Ofast -march=rv64gcv`.

Thank you, it finally worked.

closed 2025-02-20 by TommyMurphyTM1234