riscv-collab/riscv-gnu-toolchain #1682: Need help with vector extensions

Nash1818, 2025-02-26:
Following #1669 , whenever I try to use --with-arch=rv64gcv for the build (newlib) and use riscv_vector.h in my C/ c++ code and try to compile it (with unknown-elf) it shows implicit function error for vsetvl (for 64 and 32 bit both)... anyone who can guide me as to how to enable vector intrinsics properly and try out vector instructions through my C code? I want to learn more about it and contribute to RISC-V compiler development... Thanks a lot in advance....

TommyMurphyTM1234, 2025-02-26:
You need to provide a reproducible test case and clear instructions on how to replicate your issue.
Have you read the relevant documentation?

- https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Vector-Intrinsics.html

TommyMurphyTM1234, 2025-02-28:
Any update @Nash1818?

TommyMurphyTM1234, 2025-03-03:
Closing due to lack of update/reproducible test case.

closed 2025-03-03 by TommyMurphyTM1234