riscv-collab/riscv-gnu-toolchain #1683: Code built with address sanitizer crashes when running in qemu-user

yuv418, 2025-03-03:
Hello, we're trying to cross compile some RISCV code and run it in qemu-user. For testing, it is helpful to have address sanitizer enabled. We used the following:

`./configure --prefix=$RV64PREFIX --with-arch=rv64gc --with-abi=lp64 --enable-libsanitizer`

Here's a test C program:

```c
#include <stdio.h>
#include <stdlib.h>

int main() {
	int* q = malloc(32);
	printf("%p\n", q);
}
```

Now, we compile test.c and run as follows:

```
$RV64PREFIX/bin/riscv64-unknown-linux-gnu-gcc  -fsanitize=address test.c
qemu-riscv64 -L  $RV64PREFIX/sysroot/ ./a.out
```

Which is then met with the following:
```
AddressSanitizer:DEADLYSIGNAL
=================================================================
==2608716==ERROR: AddressSanitizer: SEGV on unknown address 0x0015556500f8 (pc 0x0040008d0938 bp 0x0040008007b0 sp 0x0040007fff60 T-1)
==2608716==The signal is caused by a READ memory access.
AddressSanitizer: CHECK failed: asan_suppressions.cpp:47 "((suppression_ctx)) != (0)" (0x0, 0x0) (tid=2608716)
AddressSanitizer: CHECK failed: asan_suppressions.cpp:47 "((suppression_ctx)) != (0)" (0x0, 0x0) (tid=2608716)
Trace/breakpoint trap
```

Unfortunately, it's been hard to debug this from time constraints. My first thought is to find out whether or not this is something caused by a misconfiguration or an upstream issue where ASan straight up doesn't support running in qemu-user on RISCV. 

For what it's worth, I also built with `./configure --prefix=$RV64PREFIX --enable-llvm --enable-linux --enable-libsanitizer`, which from my understanding builds for `rv64gc/lp64d`. Compiling with ASan in both Clang and GCC in this configuration yields the same error as above.

As a stopgap, we've built [Valgrind for RISCV](https://github.com/petrpavlu/valgrind-riscv64), but even that's from my understanding still a WIP and there have been some bugs. 

Either way, any pointers in the right direction would be appreciated.

TommyMurphyTM1234, 2025-03-03:
You'll probably need to report/discuss this in the upstream QEMU project.

- https://www.qemu.org/contribute/report-a-bug/

TommyMurphyTM1234, 2025-03-07:
Any update @yuv418.
Did you raise the issue upstream?
I can't see anything here:

- https://gitlab.com/qemu-project/qemu/-/issues/?sort=created_date&state=opened&search=riscv&search=sanitizer&first_page_size=20

TommyMurphyTM1234, 2025-03-10:
Closing due to lack of update.

closed 2025-03-10 by TommyMurphyTM1234