riscv-collab/riscv-gnu-toolchain #1687: error when i make linux.

jelasin, 2025-03-07:
```
./configure --prefix=/opt/riscv
make linux
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... /usr/bin/ftp
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
rm -rf stamps/build-glibc-linux-rv64gc-lp64d build-glibc-linux-rv64gc-lp64d
mkdir build-glibc-linux-rv64gc-lp64d
cd build-glibc-linux-rv64gc-lp64d && \
	CC="riscv64-unknown-linux-gnu-gcc " \
	CXX="this-is-not-the-compiler-youre-looking-for" \
	CFLAGS="   -mcmodel=medlow -O2 " \
	CXXFLAGS="   -mcmodel=medlow -O2 " \
	ASFLAGS="  -mcmodel=medlow " \
	/home/pwn/riscv/GNU/riscv-gnu-toolchain/glibc/configure \
	--host=riscv64-unknown-linux-gnu \
	--prefix=/usr \
	--disable-werror \
	--enable-shared \
	--enable-obsolete-rpc \
	--with-headers=/home/pwn/riscv/GNU/riscv-gnu-toolchain/linux-headers/include \
	--disable-multilib \
	--enable-kernel=3.0.0 \
	 \
	--libdir=/usr/lib libc_cv_slibdir=/lib libc_cv_rtlddir=/lib
checking build system type... x86_64-pc-linux-gnu
checking host system type... riscv64-unknown-linux-gnu
checking for riscv64-unknown-linux-gnu-gcc... riscv64-unknown-linux-gnu-gcc 
checking for suffix of object files... configure: error: in '/home/pwn/riscv/GNU/riscv-gnu-toolchain/build-glibc-linux-rv64gc-lp64d':
configure: error: cannot compute suffix of object files: cannot compile
See 'config.log' for more details
make: *** [Makefile:444: stamps/build-glibc-linux-rv64gc-lp64d] Error 1
```
config.log
```
This file contains any messages produced by compilers while
running configure, to aid debugging if configure makes a mistake.

It was created by riscv-toolchain configure 1.0, which was
generated by GNU Autoconf 2.71.  Invocation command line was

  $ ./configure --prefix=/opt/riscv

## --------- ##
## Platform. ##
## --------- ##

hostname = jesin
uname -m = x86_64
uname -r = 6.8.0-52-generic
[... 285 lines cut ...]
with_musl_src='$(srcdir)/musl'
with_newlib_src='$(srcdir)/newlib'
with_pk_src='$(srcdir)/pk'
with_qemu_src='$(srcdir)/qemu'
with_spike_src='$(srcdir)/spike'
with_system_zlib='--with-system-zlib'
with_uclibc_src='$(srcdir)/uclibc-ng'

## ----------- ##
## confdefs.h. ##
## ----------- ##

/* confdefs.h */
#define PACKAGE_NAME "riscv-toolchain"
#define PACKAGE_TARNAME "riscv-toolchain"
#define PACKAGE_VERSION "1.0"
#define PACKAGE_STRING "riscv-toolchain 1.0"
#define PACKAGE_BUGREPORT ""
#define PACKAGE_URL ""
#define HAVE_LIBGMP 1
#define HAVE_LIBMPFR 1
#define HAVE_LIBMPC 1

configure: exit 0
```

TommyMurphyTM1234, 2025-03-07:
Try doing a `make distclean` (and maybe also `sudo rm -rf /opt/riscv` before doing the build again.

If that doesn't help then try doing a build completely from scratch:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1687
cd issue-1687

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

./configure ...
make linux 2>&1 | tee build.log
```
If that still fails then zip and upload the `build.log` and maybe also clarify what's on your `PATH` as something there may be contributing to the problem.

There are many other issues logged that mention the "cannot compute suffix" error in case any of them are of any use:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue%20cannot%20compute%20suffix

closed 2025-03-07 by jelasin

TommyMurphyTM1234, 2025-03-07:
I see that the issue has since been closed but just for completeness here is my successful build log for a fresh/clean build using the instructions in my previous post:

- [build.log.zip](https://github.com/user-attachments/files/19126862/build.log.zip)