riscv-collab/riscv-gnu-toolchain #1688: 'riscv64-unknown-elf-gdb --versi…' terminated by signal SIGSEGV (Address boundary error)

xsw632, 2025-03-08:
[image]this is ubuntu22.04,i have followed the instructions ,but i can not even find any other one has this SIGSEGV.before i use this sudo ln -s /lib/x86_64-linux-gnu/libtinfo.so.6 /lib/x86_64-linux-gnu/libtinfow.so.6 to avoid another problem.i am wondering if this is due to dynamic library

TommyMurphyTM1234, 2025-03-08:
You'll need to explain how you built your toolchain - the `git clone`, `configure` and `make` commands that you used, and capture, zip and upload the build log. 

You'll also need to explain how exactly you're invoking GDB here in order to reproduce the issue.

Text logs are better than screenshots as not everybody can read the latter so easily and they don't facilitate searching/processing.

I don't know why you're creating a symbolic link for that native x86_64 shared library which is not part of the `riscv-gnu-toolchain` bundle.

Do you actually need to build your own toolchain? If not then a binary distribution such as from Embecosm or xPack might be a better option...

- https://embecosm.com/downloads/tool-chain-downloads/
- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/

TommyMurphyTM1234, 2025-03-08:
I cannot reproduce the issue as follows:
```
# New Ubuntu 22.04 LTS WSL2 virtual machine

# Update all software
sudo apt update
sudo apt upgrade

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1688
cd issue-1688

# Install riscv-gnu-toolchain prerequisites
sudo .github/setup-apt.sh

# Temporary workaround for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

# Configure and make
./configure --prefix=`pwd`/installed-tools
make 2>&1 | tee build.log

# Run GDB
./installed-tools/bin/riscv64-unknown-elf-gdb --version
GNU gdb (GDB) 15.2
Copyright (C) 2024 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

ldd installed-tools/bin/riscv64-unknown-elf-gdb
        linux-vdso.so.1 (0x00007ffd165ea000)
        libpython3.10.so.1.0 => /lib/x86_64-linux-gnu/libpython3.10.so.1.0 (0x00007f9492caa000)
        libexpat.so.1 => /lib/x86_64-linux-gnu/libexpat.so.1 (0x00007f9492c79000)
        libmpfr.so.6 => /lib/x86_64-linux-gnu/libmpfr.so.6 (0x00007f94929ca000)
        libgmp.so.10 => /lib/x86_64-linux-gnu/libgmp.so.10 (0x00007f9492948000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f949271c000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9492633000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9492613000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f94923ea000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f94923ce000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f9493c4d000)

```
- [build.log.zip](https://github.com/user-attachments/files/19143411/build.log.zip)

xsw632, 2025-03-08:
thank you for your answers.I find it is because of my miniconda.when i type python --version ,it shows 3.12.so i create a new pyenv 2.17.12.and that really works.thank you for your answers again

TommyMurphyTM1234, 2025-03-08:
I don't really understand but does this mean that you have figured out your problem and this issue can be closed?

xsw632, 2025-03-08:
yes thank you very much