riscv-collab/riscv-gnu-toolchain #1689: Python on RISCV

amithmath, 2025-03-11:
Can we call Python code from C/C++ on riscv32-elf? For example:

#include <python.h>

#include "interface.h"

struct API api;   /* global var */

int main(int argc, char *argv[])
{
    int rc;

    // Start Python interpreter and initialize "api" in interface.py using 
    // old style "Embedding Python in Another Application":
    // https://docs.python.org/2/extending/embedding.html#embedding-python-in-another-application
    PyObject *pName, *pModule, *py_results;
    PyObject *fill_api;
#define PYVERIFY(exp) if ((exp) == 0) { fprintf(stderr, "%s[%d]: ", __FILE__, __LINE__); PyErr_Print(); exit(1); }

    Py_SetProgramName(argv[0]);  /* optional but recommended */
    Py_Initialize();
    PyRun_SimpleString(
            "import sys;"
            "sys.path.insert(0, '.')" );

    PYVERIFY( pName = PyString_FromString("interface") )
    PYVERIFY( pModule = PyImport_Import(pName) )
    Py_DECREF(pName);
    PYVERIFY( fill_api = PyObject_GetAttrString(pModule, "fill_api") )

    // "k" = [unsigned long],
    // see https://docs.python.org/2/c-api/arg.html#c.Py_BuildValue
    PYVERIFY( py_results = PyObject_CallFunction(fill_api, "k", &api) )
    assert(py_results == Py_None);

    // Call Python function from C using cffi.
    printf("sum: %f\n", api.add_numbers(12.3, 45.6));

    // More complex example.
    char buffer[20];
    char * result = api.dump_buffer(buffer, sizeof buffer);
    assert(result != 0);
    printf("buffer: %s\n", result);

    // Let Python perform garbage collection on result now.
    rc = api.release_object(result);
    assert(rc == 0);

    // Close Python interpreter.
    Py_Finalize();

    return 0;
}

amithmath, 2025-03-11:
Well, the issue is, is there any possibility to call python code from c using riscv-gnu-tool-unknown-elf executable?

TommyMurphyTM1234, 2025-03-11:
Probably, if you have a suitable RISC-V Linux target system. But, again, nothing to do with `riscv-gnu-toolchain` per se.

amithmath, 2025-03-11:
But I see in risc-v installation directory   I see python.h isn't this header file used for Py_Initialize(),  Py_Finalize(), etc.  to call python code?

closed 2025-03-11 by TommyMurphyTM1234