riscv-collab/riscv-gnu-toolchain #1692: [Compilation error] Unable to determine floating-point ABI

davidcb98, 2025-03-20:
Good morning,

I am having difficulties compiling the toolchain. First I got the error:

> In file included from ../../.././gcc/libgcc/libgcov-interface.c:26:
> ../../.././gcc/libgcc/libgcov.h:49:10: fatal error: sys/mman.h: No such file or directory
>    49 | #include <sys/mman.h>
>       |          ^~~~~~~~~~~~
> compilation terminated.
> make[2]: *** [Makefile:928: _gcov_execlp.o] Error 1
> make[2]: *** [Makefile:928: _gcov_dump.o] Error 1
> make[2]: Leaving directory '/mnt/home/soft/riscv/originals/riscv-gnu-toolchain/build-gcc-linux-stage1/riscv64-unknown-linux-gnu/libgcc'
> make[1]: *** [Makefile:14207: all-target-libgcc] Error 2
> make[1]: Leaving directory '/mnt/home/soft/riscv/originals/riscv-gnu-toolchain/build-gcc-linux-stage1'
> make: *** [Makefile:474: stamps/build-gcc-linux-stage1] Error 2

I solved it by adding the line `#undef HAVE_SYS_MMAN_H` in the file `gcc/libgcc/libgcov.h` (I followed what they said in comment 16 of [https://gcc.gnu.org/bugzilla/show_bug.cgi?id=100289](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=100289).)

Now I have a problem with ABI:

> checking for maximum supported _FORTIFY_SOURCE level... 2
> checking whether the assembler requires one version per symbol... configure: error: Unable to determine floating-point ABI
> no
> checking for __builtin_trap with no external dependencies... make: *** [Makefile:444: stamps/build-glibc-linux-rv64gc-lp64d] Error 1
> make: *** Waiting for unfinished jobs....
> yes
> checking whether the C++ compiler supports thread_local... no
> running configure fragment for sysdeps/unix/sysv/linux/riscv
> checking for egrep -e... /usr/bin/grep -E
> configure: error: Unable to determine floating-point ABI
> make: *** [Makefile:421: stamps/build-glibc-linux-headers] Error 1

To compile I have done:

```
git clone https://github.com/riscv/riscv-gnu-toolchain --recursive
cd riscv-gnu-toolchain
vi gcc/libgcc/libgcov.h --> Add #undef HAVE_SYS_MMAN_H
./configure --prefix=/mnt/home/soft/riscv/programs/x86_64/2025.01.20 --with-arch=rv64gc --with-abi=lp64d
make linux -j30
```

Thank you very much for your time,
David

TommyMurphyTM1234, 2025-03-20:
What Linux distro and version are you using? 

Please do a clean `clone`,  `configure` and `make ... 2>&1 | tee build.log` (without any `-j30` parallelism) and zip/upload the build log. 

Please also do this without modifying any files such as `libgcov.h`. This should not be necessary if you are using a recent Linux distro version.

BTW why `-j30` anyway? Do you actually have 30 cores/threads on your host? Maybe `-j$(nproc)` would be more appropriate (depending on your host)? But please do the build log capture without any `-j` option at all because with it interleaved logging is difficult to analyse.

davidcb98, 2025-03-20:
Good morning.

It will take a while for the log to be ready.  I'll upload it in a few hours

I put 30 cores because I'm working on a supercomputer. 
The operating system is OpenSuse Leap 15.4. The compiler I use is gcc/13.2.0. 

Precisely because I'm on a supercomputer, I couldn't install the dependencies with apt-get (zypper in the case of Suse), so I hope they are all already installed, as they seem to be typical. 

Thank you,
David

davidcb98, 2025-03-20:
Here is the log

[compile_david_j1.log](https://github.com/user-attachments/files/19367367/compile_david_j1.log)

and the commands

```
git clone https://github.com/riscv/riscv-gnu-toolchain --recursive
module load gcc/13.2.0 
./configure --prefix=/mnt/home/soft/riscv/programs/x86_64/2025.01.20_new --with-arch=rv64gc --with-abi=lp64d
export PATH=/mnt/home/soft/riscv/programs/x86_64/2025.01.20_new/bin:${PATH}
mkdir /mnt/home/soft/riscv/programs/x86_64/2025.01.20_new/
make linux 2>&1 | tee compile_david_j1.log
```

TommyMurphyTM1234, 2025-03-21:
This worked for me:

1. Install OpenSUSE Leap 15.4 in a VirtualBox VM.
2. Apply any available package updates.
3. Install GCC 14.2.
4. Clone the `riscv-gnu-toolchain` repo.
5. Apply the "shallow clone" workaround for now: https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
6. I asked ChatGPT to create an OpenSUSE equivalent of the [`setup-apt.sh` script](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh) in order to install the prerequisites - and even then I had to fiddle a bit to get everything needed installed.
7. Configured and built the toolchain and it worked fine.
8. Build log (there were a couple of false starts with the GDB build due to missing `libexpat-devel`): [build.log.zip](https://github.com/user-attachments/files/19392518/build.log.zip)
```
./installed-tools/riscv64-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=./riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1692/installed-tools/libexec/gcc/riscv64-unknown-linux-gnu/14.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/user/issue-1692/gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/home/user/issue-1692/installed-tools --with-sysroot=/home/user/issue-1692/installed-tools/sysroot --with-pkgversion= --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././gcc --disable-default-pie --disable-multilib --with-abi=lp64d --with-arch=rv64gc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 14.2.0 () 
```

TommyMurphyTM1234, 2025-03-22:
Just to recap - this worked for me:
```
# Install OpenSUSE 15.4 (Desktop with XFCE) on VirtualBox VM
# Boot into it...

# Install updates
sudo zypper update

# Install GCC 13 and make it the default GCC
sudo zypper install gcc13 gcc13-c++
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-13 100
sudo update-alternatives --install /usr/bin/g++ g++ /usr/bin/g++-13 100
sudo update-alternatives --config gcc

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib64/gcc/x86_64-suse-linux/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-suse-linux
Configured with: ../configure CFLAGS=' -fmessage-length=0 -grecord-gcc-switches -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' CXXFLAGS=' -fmessage-length=0 -grecord-gcc-switches -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' XCFLAGS=' -fmessage-length=0 -grecord-gcc-switches -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' TCFLAGS=' -fmessage-length=0 -grecord-gcc-switches -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' GDCFLAGS=' -fmessage-length=0 -grecord-gcc-switches -O2 -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g' --prefix=/usr --infodir=/usr/share/info --mandir=/usr/share/man --libdir=/usr/lib64 --libexecdir=/usr/lib64 --enable-languages=c,c++,objc,fortran,obj-c++,ada,go,d,m2 --enable-offload-targets=nvptx-none, --enable-offload-defaulted --without-cuda-driver --enable-checking=release --disable-werror --with-gxx-include-dir=/usr/include/c++/13 --with-libstdcxx-zoneinfo=/usr/share/zoneinfo --enable-ssp --disable-libssp --disable-libvtv --enable-cet=auto --disable-libcc1 --disable-plugin --with-bugurl=https://bugs.opensuse.org/ --with-pkgversion='SUSE Linux' --with-slibdir=/lib64 --with-system-zlib --enable-libstdcxx-allocator=new --disable-libstdcxx-pch --enable-libphobos --enable-version-specific-runtime-libs --with-gcc-major-version-only --enable-linker-build-id --enable-linux-futex --enable-gnu-indirect-function --program-suffix=-13 --without-system-libunwind --enable-multilib --with-arch-32=x86-64 --with-tune=generic --enable-link-serialization --build=x86_64-suse-linux --host=x86_64-suse-linux
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 13.2.1 20230912 [revision b96e66fd4ef3e36983969fb8cdd1956f551a074b] (SUSE Linux) 

# Used ChatGPT and some tinkering to convert setup-apt.sh (https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh) to this:
sudo zypper install -y autoconf automake curl python3 python3-pip python3-tomli \
    mpc-devel mpfr-devel gmp-devel gawk gcc gcc-c++ make bison flex texinfo \
    gperf libtool patchutils bc zlib-devel libexpat-devel git ninja cmake \
    glib2-devel expect dtc python3-pyelftools libslirp-devel

# Clone riscv-gnu-toolchain
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Workaround for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

# Configure and build toolchain (note that arch/abi is rv64gc/lp64 by default)
./configure --prefix=`pwd`/installed-tools
make linux 2>&1 | tee build.log
```
- [build.log.zip](https://github.com/user-attachments/files/19405156/build.log.zip)

TommyMurphyTM1234, 2025-03-24:
Any update @davidcb98?
As shown above I cannot reproduce any failure when building the toolchain on OpenSUSE 15.4.
What happens if you follow the same flow as I did?

davidcb98, 2025-03-24:
Good morning,

sorry, I haven't had time to look at it these days. I see that I'm missing some zypper dependencies. I keep getting the <sys/mman.h> bug and maybe that's why, but it seems strange to me because the compiler that doesn't find it is the one installed with the toolchain. 

As I work in a cluster, it's not a good idea to install things with zypper install so maybe I have to install some dependencies by hand. 

Can't you make a docker image of the toolchain and update it with new releases? You can put it in DockerHub. Then there would be no problem to install it.

David

TommyMurphyTM1234, 2025-03-24:
> I keep getting the <sys/mman.h> bug and maybe that's why

Sorry, I have no idea why. As I said, the flow that I followed above just worked. 

> Can't you make a docker image of the toolchain and update it with new releases?

Who is "you"? If you mean me then this isn't something that I'm prepared to or have the time to do, sorry. 

The `riscv-gnu-toolchain` repo uses GitHub actions to do CI builds of the toolchain but only on Ubuntu right now. 

- https://github.com/riscv-collab/riscv-gnu-toolchain/actions

Why do you have to build the tools on the cluster rather than building on another separate OpenSUSE host and copying the binary toolchain over?

@ilg-ul's xPack project provides binary builds of the bare-metal toolchain built in such a way (e.g. old GLIBC dependency, static linking, and building/bundling all dynamic dependencies) that it is as self-contained as possible and portable across all sorts of GNU/Linux distros. 

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/

Perhaps you could adapt the build framework/scripts to do the same for the Linux toolchain? 

- https://xpack-dev-tools.github.io/riscv-none-elf-gcc-xpack/docs/developer/

Another option might be to use crosstool-NG to build the toolchain?

- https://crosstool-ng.github.io/

TommyMurphyTM1234, 2025-03-25:
Since the issue can't be reproduced I am closing the issue.
If you identify a reproducible failure case then please post details of it and reopen the issue.

TommyMurphyTM1234, 2025-03-25:
> I keep getting the <sys/mman.h> bug

This old closed issue mentions a missing `<sys/mman.h>` error (not sure what you mean by "bug"?) but I'm not sure if it's the same thing...

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/565

closed 2025-03-25 by TommyMurphyTM1234

davidcb98, 2025-03-25:
Hello @TommyMurphyTM1234 . I was referring in general to the repository administrators. I have opened a pull request with suggesting a Dockerfile. 

Updating a Dockerfile with repository releases is trivial. The only thing that takes a bit of time is generating the docker image and uploading it to DockerHub. But hey, I think it's worth the effort.

Let me know what you think,
David