riscv-collab/riscv-gnu-toolchain #1693: Unable to build riscv64im/riscv32im toolchain

Sh0g0-1758, 2025-03-20:
Hello! 

I am using the following script to build the toolchain:

```
#!/bin/bash
set -e

HOST=$1
PREFIX=$PWD/.dist/$HOST

# Extract bit size from host name
if [[ $HOST =~ riscv32im ]]; then
  ARCH="rv32im"
  ABI="ilp32"
elif [[ $HOST =~ riscv64im ]]; then
  ARCH="rv64im"
  ABI="lp64"
else
  echo "Unknown host: $HOST"
  exit 1
fi

pushd riscv-gnu-toolchain

echo "building toolchain for host: $HOST, arch: $ARCH, abi: $ABI"
./configure --prefix=$PREFIX --with-cmodel=medany --disable-gdb --with-arch=$ARCH --with-abi=$ABI
make -j$(nproc 2>/dev/null || sysctl -n hw.logicalcpu)

popd

pushd .dist

tar cvJf $HOST.tar.xz $HOST

popd
```

but the build fails with an error message saying that sprintf should not be used and an unkown compile option. Am I missing any flag that I should pass ? Refer the [logs](https://github.com/maksimryndin/rv64im-toolchain/actions/runs/13977724681/job/39135500316) for more information.

TommyMurphyTM1234, 2025-03-20:
Please try the build instructions in the README and see if they work before trying to create a proprietary script for the build. I.e.:
```
# Clone riscv-gnu-toolchain 
git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

# Configure and make
./configure ...
make ...
```

TommyMurphyTM1234, 2025-03-21:
This worked fine for me:
```
#
# Distro details
#

cat /etc/*-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=24.04
DISTRIB_CODENAME=noble
DISTRIB_DESCRIPTION="Ubuntu 24.04.2 LTS"
PRETTY_NAME="Ubuntu 24.04.2 LTS"
NAME="Ubuntu"
VERSION_ID="24.04"
VERSION="24.04.2 LTS (Noble Numbat)"
VERSION_CODENAME=noble
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=noble
LOGO=ubuntu-logo

#
# GCC details
#

gcc -v
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-linux-gnu/13/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 13.3.0-6ubuntu2~24.04' --with-bugurl=file:///usr/share/doc/gcc-13/README.Bugs --enable-languages=c,ada,c++,go,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-13 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/libexec --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-libstdcxx-backtrace --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-13-fG75Ri/gcc-13-13.3.0/debian/tmp-gcn/usr --enable-offload-defaulted --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 13.3.0 (Ubuntu 13.3.0-6ubuntu2~24.04)

#
# Clone riscv-gnu-toolchain 
#

git clone https://github.com/riscv-collab/riscv-gnu-toolchain
cd riscv-gnu-toolchain

# Install prerequisites
sudo .github/setup-apt.sh

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

#
# Build rv32im/ilp32 toolchain
#

./configure --prefix=`pwd`/installed-tools --with-cmodel=medany --disable-gdb --with-arch=rv32im --with-abi=ilp32
make 2>&1 | tee build-rv32im-ilp32.log

#
# Build rv64im/lp64 toolchain
#

make distclean && rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools --with-cmodel=medany --disable-gdb --with-arch=rv64im --with-abi=ilp64
make 2>&1 | tee build-rv64im-ilp64.log
```
- [build-rv32im-ilp32.log.zip](https://github.com/user-attachments/files/19387561/build-rv32im-ilp32.log.zip)
- 

BTW - in case it helps at all you can build a single multilib toolchain for both `rv32im/ilp32` and `rv64im/lp64` - e.g.:
```
./configure ... --with-arch=rv64im --with-arch=lp64 --with-multilib-generator="rv32im-ilp32--"
```
```

Sh0g0-1758, 2025-03-21:
Hey! Thanks for the help. I applied the shallow clone workaround you mentioned and it builds fine for me now. Though I am facing issues getting the toolchain build for macos on osx/x64. I am mounting a new disk image like so but facing an error. Is there some standard procedure for it too ? : 

```
    MOUNT_POINT="/Volumes/RISCVToolchain"
    WORK_DIR="$MOUNT_POINT/workdir"

    hdiutil create -size 8g -fs "APFS Case-sensitive" -volname RISCVToolchain ~/RISCVToolchain.dmg -quiet
    hdiutil attach ~/RISCVToolchain.dmg -quiet
    mkdir -p "$WORK_DIR"
    cd "$WORK_DIR"

```

TommyMurphyTM1234, 2025-03-21:
> Though I am facing issues getting the toolchain build for macos on osx/x64.

Also, I have no idea about macOS build issues and no access to hardware allowing me to exercise such a build.

Do you actually need to build the toolchain from scratch or could you use a binary distribution such as...

-

Sh0g0-1758, 2025-03-21:
Thanks. I will check it out.

closed 2025-03-21 by Sh0g0-1758