riscv-collab/riscv-gnu-toolchain #1697: Makefile error

shubham-1020, 2025-03-26:
i am getting the error in the building of the toolchain
Makefile:51: *** commands commence before first target.  Stop.

TommyMurphyTM1234, 2025-03-26:
Same as I said here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1696

> You'll need to explain exactly how you are configuring and building the toolchain. And capture, zip and upload the full build log. E.g.:
>```
> # Clone repo
> git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1696
> cd issue-1696
>
> # Install prerequisites
> sudo .github/setup-apt.sh
> 
> # Work around for "shallow clone" problem thanks to @hpretl 
> # https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
> sed -i '/shallow = true/d' .gitmodules
> sed -i 's/--depth 1//g' Makefile.in
> 
> # Configure and make
> ./configure ...
> make 2>&1 | tee build.log
>```

TommyMurphyTM1234, 2025-03-26:
Maybe this is your problem? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1598

The same error is mentioned in this issue: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1577

shubham-1020, 2025-03-26:
they both are different on the previous post it was from the Windows subsystem and the newer issue from the MacOS M1
trieed multiple attemps to install it but failed 
can you guide me for this ?

closed 2025-03-27 by shubham-1020

TommyMurphyTM1234, 2025-03-27:
It would be useful if you clarified what the resolution was here. It might help others in the future.