riscv-collab/riscv-gnu-toolchain #1698: getting an error building the build-newlib

shubham-1020, 2025-03-27:
checking where to find the target windmc... pre-installed checking whether to enable maintainer-specific portions of Makefiles... no configure: creating ./config.status config.status: creating Makefile make -C build-newlib make [1]: Entering directory
'/home/harig/riscv/riscv-gnu-toolchain/build-newlib'
make[2]: Entering directory '/home/harig/riscv/riscv-gnu-toolchain/build-newlib* mkdir -p
•/etc
Configuring in ./etc configure: creating cache ./config.cache
checking whether to enable maintainer-specific portions of Makefiles... no checking for a BSD-compatible install... /bin/install -c checking whether build environment is sane... yes checking for a thread-safe mkdir -p... /bin/mkdir -p checking for gawk...
/usr/bin/gawk
checking whether make sets $(MAKE)... yes checking whether make supports nested variables... yes checking whether make supports nested variables... (cached) yes configure: updating cache ./config.cache
checking that generated files are newer than configure... done configure: creating ./config.status config.status: creating Makefile make [3]: Entering directory
'/home/harig/riscv/riscv-gnu-toolchain/build-newlib/etc'
make [3]: Nothing to be done for 'all'.
make [3]: Leaving directory '/home/harig/riscv/riscv-gnu-toolchain/build-newlib/etc
Checking multilib configuration for newlib... mkdir -p -- riscv64-unknown-elf/newlib
Configuring in riscv64-unknown-elf/newlib configure: creating cache ./config. cache checking build system type... x86_64-pc-linux-gnu checking host system type... riscv64-unknown-elf checking for a BSD-compatible install... /bin/install -c checking whether build environment is sane... yes
checking for riscv64-unknown-elf-strip... riscv64-unknown-elf-strip checking for a thread-safe mkdir -p... /bin/mkdir -p checking for gawk... /usr/bin/gawk checking whether make sets $(MAKE)... yes checking whether make supports nested variables... yes checking for style of include used by make... GNU
checking for riscv64-unknown-elf-gcc... riscv64-unknown-elf-gcc -B/home/harig/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/-isystem /home/harig/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib/ta rg-include -isystem /home/harig/riscv/riscv-gnu-toolchain/newlib/newlib/libc/include -B/home/harig/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/libgloss/riscv64 -L/home/harig/riscv/riscv-gnu-toolchain/build-newlib/riscv6
4-unknown-elf/libgloss/libnosys -L/home/harig/riscv/riscv-gnu-toolchain/newlib/libgloss/riscv64
checking whether we are cross compiling... yes
checking for suffix of object files... configure: error: in */home/harig/riscv/riscv-gnu-toolchain/build-newlib/riscv64-unknown-elf/newlib*: configure: error: cannot compute suffix of object files: cannot compile
See "config.log' for more details
make [2]: *** [Makefile: 8448: configure-target-newlib] Error 1 make [2]: Leaving directory '/home/harig/riscv/riscv-gnu-toolchain/build-newlib' make [1]:
*** [Makefile: 879: all] Error 2
make [1]: Leaving directory '/home/harig/riscv/riscv-gnu-toolchain/build-newlib' make: *** [Makefile:685: stamps/build-newlib] Error 2

getting this error after i run the build-newlib

TommyMurphyTM1234, 2025-03-27:
Once again you haven't provided essential information needed to reproduce your issue...

- Host OS
- The exact commands and options that you are using to clone, configure and make the toolchain
- Build log
- Etc. 

You should also make sure that you are doing a clean build by first doing `make distclean && rm -rf <prefix-dir>` in case a previous (failed?) build interferes with the current build. 

For what it's worth this error: 
> ```
> configure: error: cannot compute suffix of object files: cannot compile
> ```
appears in several existing issues so you could be hitting an issue that others have previously encountered and got around:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue%20%20cannot%20compute%20suffix

shubham-1020, 2025-03-27:
**running the WSL (running the ubutnutu )**
 **commands -** 
      => $ git clone https://github.com/riscv/riscv-gnu-toolchain
      => install the dependecies form the repo
       => ./configure --prefix=/opt/riscv
       make linux
       => ./configure --prefix=/opt/riscv --enable-multilib
       => ./configure --enable-qemu-system --prefix=/opt/riscv
            make build-sim SIM=qemu
       => $ sudo apt-get install python3-pyelftools
_this runs fine upto but when i run the **Testing GCC** ----_
=> ./configure --prefix=$RISCV --disable-linux --with-arch=rv64ima # or --with-arch=rv32ima
make newlib
make report-newlib SIM=gdb 

### i got the error here .

### my folder structure looks like  ----
home/hari/riscv/riscv-gnu-toolchain

is this is the error because of the incorrrect path ?
currently i had the screenshotsthat upto this command i guess it ran  fine 
=> ./configure --enable-qemu-system --prefix=/opt/riscv
            make build-sim SIM=qemu

### after this command i got something like this in my system ------

Installing /home/harig/riscv/riscv-gnu-toolchain/build-qemu/pc-bios/descriptors/60-edk2-arm.json to /opt/riscv/share/qemu/firmware Installing /home/harig/riscv/riscv-gnu-toolchain/build-qemu/pc-bios/descriptors/60-edk2-i386.json to /opt/riscv/share/qemu/firmware Installing /home/harig/riscv/riscv-gnu-toolchain/build-qemu/pc-bios/descriptors/60-edk2-x86_64.json to /opt/riscv/share/qemu/firmware
Installing /home/harig/riscv/riscv-gnu-toolchain/build-qemu/pc-bios/descriptors/60-edk2-loongarch64.json to /opt/riscv/share/qemu/firmware
Installing /home/harig/riscv/riscv-gnu-toolchain/build-qemu/pc-bios/descriptors/60-edk2-riscv64.json to /opt/riscv/share/qemu/firmware
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/ar to /opt/riscv/share/qemu/keymaps Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/bepo to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/cz to /opt/riscv/share/qemu/keymaps Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/da to /opt/riscv/share/qemu/keymaps Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/de to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/de-ch to /opt/riscv/share/qemu/keymaps Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/en-gb to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/en-us
to/opt/riscv/share/gemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/es to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/et to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fi to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fo to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fr to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fr-be to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fr-ca
to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/fr-ch
to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/hr to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/hu to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/is to /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/it to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/ja to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/lt to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/lv to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/mk to Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/nl
/opt/riscv/share/qemu/keymaps
to
/opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/no
Installing /home/harig/riscv/riscv-gnu-toolchain/qmu/pc-bios/keymaps/pl to
/opt/riscv/share/qemu/keymaps
• /opt/riscv/share/qemu/keymaps
Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/pt to Installing /home/harig/riscv/riscv-gnu-toolchain/qemu/pc-bios/keymaps/pt-br
/opt/riscv/share/qemu/keymaps
[... 22 lines cut ...]
harig@SAC1:~/riscv/riscv-gnu-toolchain$ ls build-qemu
Makefile
compile commands. json
libmigration.a.p
Makefile.mtest
config-host.h
hmp-commands-info.h hmp-commands.h
nigratio
gemu-img
gemu -img-cmds.h
Makefile.ninja
Makefile. preregs
config-host.mak config-meson.cross
libgemu-riscv64-linux-user.a.p
gemu-io
regression spike test
newlib gemu scripts
stamps uclibc-ng
gemu-system-riscv32.p
qemu-system-riscv64
qemu-system-riscv64.p
gemu-version.h
riscv32-softmmu-config-target.h riscv32-softmmu-gdbstub-xml.c
riscv64-linux-user-config-target.h riscv64-linux-user-gdbstub-xml.c
system target tcg

TommyMurphyTM1234, 2025-03-27:
If you're doing different builds of the toolchain with different configurations then you should do `make distclean && rm -rf <prefix-dir>` before each `configure` and `make` to ensure that one build doesn't interfere with the next one and trigger the `cannot compute suffix` error. Copy the built/installed toolchain before deleting it if you need to retain it. 

Please read the issues that I linked to that deal with the `cannot compute suffix` error. In most or all of them doing `make distclean && rm -rf <prefix-dir>` before a new configuration/build addresses the problem

closed 2025-03-28 by shubham-1020

TommyMurphyTM1234, 2025-03-29:
What was the resolution to your build issues?