riscv-collab/riscv-gnu-toolchain #1700: Building fails due to binutils cloning / branch issue

emeb, 2025-04-06:
I'm trying to build newlib/multilib on Raspbian 64 with the following commands:
```
git clone https://github.com/riscv/riscv-gnu-toolchain
cd riscv-gnu-toolchain/
./configure --prefix=/opt/riscv-gcc/4-6-2025 --enable-multilib
make
```

It begins but after a few minutes I get the following result:
```
Cloning into '/home/ericb/Engineering/github/riscv-gnu-toolchain/binutils'...
remote: Enumerating objects: 40846, done.        
remote: Counting objects: 100% (40846/40846), done.        
remote: Compressing objects: 100% (31849/31849), done.        
remote: Total 40846 (delta 10609), reused 24462 (delta 7607), pack-reused 0        
Receiving objects: 100% (40846/40846), 78.21 MiB | 3.31 MiB/s, done.
Resolving deltas: 100% (10609/10609), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: Fetched in submodule path 'binutils', but it did not contain beb2cdbcda911764b2bed5e57921fe90493260bd. Direct fetching of that commit failed.
make: *** [Makefile:348: /home/ericb/Engineering/github/riscv-gnu-toolchain/binutils/.git] Error 128
```
I've tried several times with the same error. Any thoughts what's going on?

TommyMurphyTM1234, 2025-04-06:
Known issue: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669

Workaround here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585

You'll have to do a clean clone for that to work: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2682013720

closed 2025-04-06 by TommyMurphyTM1234

emeb, 2025-04-06:
Thanks, that appears to have resolved the problem.