riscv-collab/riscv-gnu-toolchain #1701: The build fails with the followiing error, "ERROR: unknown command "../.././gcc/gcc/config/riscv/arch-canonicalize" --with-abi=lp64d is not supported for ISA "

ckavalipati, 2025-04-07:
The failure log is as jus before this error is as below,

checking for struct tms... yes
checking for clock_t... yes
checking for F_SETLKW... yes
checking for O_CLOEXEC... yes
checking for fcntl.h... (cached) yes
checking whether O_NONBLOCK is declared... yes
checking for AF_UNIX... yes
checking for AF_INET6... yes
checking for _LK_LOCK... no
checking if mkdir takes one argument... no
ERROR: unknown command "../.././gcc/gcc/config/riscv/arch-canonicalize"
--with-abi=lp64d is not supported for ISA 
make[1]: *** [Makefile:4628: configure-gcc] Error 1
make[1]: Leaving directory '/proj/fw_rvl/devel/pkgs/riscv-gnu-toolchain_new/build-gcc-linux-stage1'
make: *** [Makefile:472: stamps/build-gcc-linux-stage1] Error 2

The configure command line is, 
./configure --prefix /proj/fw_rvl/tools/riscv/riscv_gnu/2025.01.20 --enable-multilib --with-multilib-generator="rv32imc-ilp32--a*zicsr*zifencei"

This problem is observed with latest top of trunk code and as well as, release 2025.01.20 tag. 

Looks like build process is expecting something from the the system. All the prerequisites are installed.

TommyMurphyTM1234, 2025-04-07:
> ```
> ERROR: unknown command "../.././gcc/gcc/config/riscv/arch-canonicalize"
> ```
Does the `gcc/gcc/config/riscv/arch-canonicalize` script exist in your cloned source base?
If not then perhaps there was some error at the time of cloning?

TommyMurphyTM1234, 2025-04-07:
This worked fine for me on Ubuntu 24.04.2 LTS:
```
# Clone repo
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1701
cd issue-1701

# Install prerequisites
sudo .github/setup-apt.sh

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

# This "shallow clone" problem may be fixed soon by this PR:
# https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1702

# Configure
# No --with-multilib-generator because it's not applicable to Linux toolchain build
./configure --prefix=`pwd`/installed-tools --enable-multilib

# Build Linux toolchain
make linux 2>&1 | tee build.log
```
- [build.log.zip](https://github.com/user-attachments/files/19631648/build.log.zip)

ckavalipati, 2025-04-07:
> > ```
> > ERROR: unknown command "../.././gcc/gcc/config/riscv/arch-canonicalize"
> > ```
> 
> Does the `gcc/gcc/config/riscv/arch-canonicalize` script exist in your cloned source base? If not then perhaps there was some error at the time of cloning?

Yes, the script gcc/gcc/config/riscv/arch-canonicalize exits in my clones gcc source base

TommyMurphyTM1234, 2025-04-07:
> Yes, the script gcc/gcc/config/riscv/arch-canonicalize exits in my clones gcc source base

I can't reproduce your problem and you haven't provided the information that I mentioned earlier - details of exactly how you're cloning, configuring and building the toolchain, your full build log, and maybe also details of your host platform.

TommyMurphyTM1234, 2025-04-10:
Any update @ckavalipati?
If you follow my build instructions above what happens? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1701#issuecomment-2783097535

TommyMurphyTM1234, 2025-04-12:
Please detail the exact commands that you are using to clone, configure and build the toolchain when this happens.
Also provide details of the host distro that you are using.
I can't reproduce your problem so far on Ubuntu 24.04.2 LTS.
If the script in question exists but the build process can't find it then maybe there's some odd issue with permissions or something like that.

TommyMurphyTM1234, 2025-04-16:
Any update @ckavalipati?

closed 2025-04-17 by TommyMurphyTM1234

ckavalipati, 2025-04-17:
> Any update [@ckavalipati](https://github.com/ckavalipati)?

My apologies for delayed response. The steps that I have followed are similar to what you have suggested (reproduced below). 

The distribution I am using on this server is CentOS 7 with latest updates. This is a production server with custom tools. All the development tools, such as GCC are updated and if anything is missing, we can try to add them. 

I have attached build logs and errors, have you found any clues from them? 

What does the build system dependent on? Does it require any features form the system or the shell? 

Cloning steps:

git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1701
cd issue-1701

# Install prerequisites
sudo .github/setup-apt.sh

# Work around for "shallow clone" problem thanks to @hpretl 
# https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
sed -i '/shallow = true/d' .gitmodules
sed -i 's/--depth 1//g' Makefile.in

# This "shallow clone" problem may be fixed soon by this PR:
# https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1702

# Configure
# No --with-multilib-generator because it's not applicable to Linux toolchain build
./configure --prefix=`pwd`/installed-tools --enable-multilib

make linux

TommyMurphyTM1234, 2025-04-17:
> The distribution I am using on this server is CentOS 7 with latest updates.

I did a build on CentOS 8 a while ago in case that's of any use? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1503#issuecomment-2212554558

> This is a production server with custom tools.

I don't know what this means.

> What does the build system dependent on?

The packages mentioned here: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/.github/setup-apt.sh

I used ChatGPT to "translate" this into a draft script for use on CentOS in case that's of any use?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1664#issuecomment-2709674585

> I have attached build logs and errors, have you found any clues from them?

No, that's why I'm asking for clear instructions on how to reproduce the issue.

> Is it possible for me to clone all the dependent repos by running a script?

Yes, you can do a recursive clone if you work around the shallow clone issue - e.g. using `--depth 1`.

ckavalipati, 2025-04-17:
I run the steps as suggetsed by you. Looks build process has dependency. Is there a way can we increase verbosity of the build logs?

TommyMurphyTM1234, 2025-04-17:
> Is there a way can we increase verbosity of the build logs?

Not that I know of.

I'm trying to install a CentOS 7 VM to see what happens if I try the build locally.

ckavalipati, 2025-04-17:
@TommyMurphyTM1234 , I meant to say that I cannot change my Linux distribution to Ubuntu. I believe I have all prerequisites.

TommyMurphyTM1234, 2025-04-17:
You still haven't answered this question:

> What does this mean?
> 
> > The distribution I am using on this server is CentOS 7 with latest updates. This is a production server with custom tools. All the development tools, such as GCC are updated and if anything is missing, we can try to add them.

ckavalipati, 2025-04-18:
Looks like the issue is in the ~/gcc/gcc/config.gcc where architecture is determined, 

	PYTHON=`which python || which python3 || which python2`
	if test "x${PYTHON}" != x; then
		with_arch=`${PYTHON} ${srcdir}/config/riscv/arch-canonicalize -misa-spec=${with_isa_spec} ${with_arch}`
	fi

In my system, older python version came with the distribution is getting selected.  If I change the order of the search, as below, the gcc configuration successfully completed.

         PYTHON=`which python3 || which python || which python2`

TommyMurphyTM1234, 2025-04-18:
Maybe that's why the `riscv-gnu-toolchain` prerequisites specify Python3?

- https://github.com/riscv-collab/riscv-gnu-toolchain/blob/fcc7a8cf7c7b47121783e3b5c7d87aee7317632f/.github/setup-apt.sh#L6
- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#prerequisites

And why the ChatGPT "translation" of `setup-apt.sh` suggests creating a link from Python to Python3?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1664#issuecomment-2709674585

TommyMurphyTM1234, 2025-04-18:
If you think that there's an issue with `gcc/gcc/config.gcc` then you should report it in the GCC bugbase.

- https://gcc.gnu.org/bugs/
- https://github.com/gcc-mirror/gcc/blob/dd982198656d914a4958bf86356a4c996c728b9d/gcc/config.gcc#L4775-L4778

ckavalipati, 2025-04-18:
This issue should be reported to the GCC. 

Looking at the riscv-gnu-toolchain Makefile, I am not seeing any way to pass optional parameters. 

For example, if I want to pass additional parameters when configuring gdb, how can I do that? 

I am showing the make rule that is used from the top level Makefile. 

stamps/build-gdb-newlib: $(GDB_SRCDIR) $(GDB_SRC_GIT) $(PREPARATION_STAMP)
	rm -rf $@ $(notdir $@)
	mkdir $(notdir $@)
# CC_FOR_TARGET is required for the ld testsuite.
	cd $(notdir $@) && CC_FOR_TARGET=$(NEWLIB_CC_FOR_TARGET) $</configure \
		--target=$(NEWLIB_TUPLE) \
		$(CONFIGURE_HOST) \
		--prefix=$(INSTALL_DIR) \
		 \
		--disable-werror \
		$(GDB_TARGET_FLAGS) \
		--enable-gdb \
		--disable-gas \
		--disable-binutils \
		--disable-ld \
		--disable-gold \
		--disable-gprof
	$(MAKE) -C $(notdir $@)
	$(MAKE) -C $(notdir $@) install
	mkdir -p $(dir $@) && touch $@