riscv-collab/riscv-gnu-toolchain #1703: unable to build gccgo

Sh0g0-1758, 2025-04-09:
Hey! I am running the following command to configure and build gccgo but I don't see a binary for it in the bin folder. 

```
./configure --prefix=/home/shogo/.dist/rv64gccgo --with-cmodel=medany --disable-gdb --with-arch=rv64ima --with-abi=lp64 --with-languages=c,c++,go
```

Am I missing something ?

TommyMurphyTM1234, 2025-04-09:
Anything useful here? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues?q=is%3Aissue%20gccgo%20

I'm not sure what the status is of Go(lang) support for RISC-V. You might want to check in the upstream GCC project where such enhancement and maintenance work is done.

- https://gcc.gnu.org/lists.html

TommyMurphyTM1234, 2025-04-10:
I only noticed this now...

> Hey! I am running the following command to configure and build gccgo but I don't see a binary for it in the bin folder.
> 
> ```
> ./configure \
>   --prefix=/home/shogo/.dist/rv64gccgo \
>   --with-cmodel=medany \
>   --disable-gdb \
>   --with-arch=rv64ima \
>   --with-abi=lp64 \
>   --enable-languages=c,c++,go
> ```
> 
> Am I missing something ?

`--enable-languages` is not a valid option for the `riscv-gnu-toolchain` [top-level `configure` script](https://github.com/riscv-collab/riscv-gnu-toolchain/blob/master/configure):
```
./configure \
  --prefix=/home/shogo/.dist/rv64gccgo \
  --with-cmodel=medany \
  --disable-gdb \
  --with-arch=rv64ima \
  --with-abi=lp64 \
  --enable-languages=c,c++,go
configure: WARNING: unrecognized options: --enable-languages
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/bash
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed
configure: WARNING: unrecognized options: --enable-languages
```
Did you mean `--with-languages`?
This works for me:
```
./configure --prefix=`pwd`/installed-tools --disable-gdb --with-languages=c,c++,fortran,go
make linux

...

ls installed-tools/bin/
riscv64-unknown-linux-gnu-addr2line  riscv64-unknown-linux-gnu-gcc         riscv64-unknown-linux-gnu-gcov-tool        riscv64-unknown-linux-gnu-gprofng   riscv64-unknown-linux-gnu-readelf
riscv64-unknown-linux-gnu-ar         riscv64-unknown-linux-gnu-gcc-14.2.0  riscv64-unknown-linux-gnu-gfortran         riscv64-unknown-linux-gnu-ld        riscv64-unknown-linux-gnu-size
riscv64-unknown-linux-gnu-as         riscv64-unknown-linux-gnu-gcc-ar      riscv64-unknown-linux-gnu-gp-archive       riscv64-unknown-linux-gnu-ld.bfd    riscv64-unknown-linux-gnu-strings
riscv64-unknown-linux-gnu-c++        riscv64-unknown-linux-gnu-gcc-nm      riscv64-unknown-linux-gnu-gp-collect-app   riscv64-unknown-linux-gnu-lto-dump  riscv64-unknown-linux-gnu-strip
riscv64-unknown-linux-gnu-c++filt    riscv64-unknown-linux-gnu-gcc-ranlib  riscv64-unknown-linux-gnu-gp-display-html  riscv64-unknown-linux-gnu-nm
riscv64-unknown-linux-gnu-cpp        riscv64-unknown-linux-gnu-gccgo       riscv64-unknown-linux-gnu-gp-display-src   riscv64-unknown-linux-gnu-objcopy
riscv64-unknown-linux-gnu-elfedit    riscv64-unknown-linux-gnu-gcov        riscv64-unknown-linux-gnu-gp-display-text  riscv64-unknown-linux-gnu-objdump
riscv64-unknown-linux-gnu-g++        riscv64-unknown-linux-gnu-gcov-dump   riscv64-unknown-linux-gnu-gprof            riscv64-unknown-linux-gnu-ranlib

```
The bare-metal toolchain build (with `--with-languages=c,c++,go`) fails:
```
checking for gawk... (cached) /usr/bin/gawk
checking for --enable-version-specific-runtime-libs... no
configure: error: "could not determine GOOS from riscv64-unknown-elf"
make[2]: *** [Makefile:14731: configure-target-libgo] Error 1
make[2]: Leaving directory '/home/user/issue-1693/build-gcc-newlib-stage2'
make[1]: *** [Makefile:1055: all] Error 2
make[1]: Leaving directory '/home/user/issue-1693/build-gcc-newlib-stage2'
make: *** [Makefile:758: stamps/build-gcc-newlib-stage2] Error 2
```
but I think that Go[lang] requires an OS and so only builds with the Linux toolchain?

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/459#issuecomment-508488833

TommyMurphyTM1234, 2025-04-10:
Just for completeness, this also built successfully for me:
```
./configure \
  --prefix=`pwd`/installed-tools \
  --with-cmodel=medany \
  --disable-gdb \
  --with-arch=rv64ima \
  --with-abi=lp64 \
  --enable-languages=c,c++,fortran,go

make linux 2>&1 | tee build.log
```

TommyMurphyTM1234, 2025-04-11:
Did that help at all @Sh0g0-1758?

Sh0g0-1758, 2025-04-11:
Hey! Thanks a lot. I am able to build gccgo now. We have to give gccgo some target to build otherwise GOOS complains.

closed 2025-04-11 by Sh0g0-1758

TommyMurphyTM1234, 2025-04-11:
> We have to give gccgo some target to build otherwise GOOS complains.

I don't understand what this means.