riscv-collab/riscv-gnu-toolchain #1704: riscv32-unknown-linux-gnu-ld: internal error: aborting at riscv-gnu-toolchain/binutils/ld/ldlang.c:11034 in cmdline_check_object_only_section

petaleleo, 2025-04-10:
Hello! I am try compile linux for 32-bit RISC-V Linux for QEMU:

build risc-gnu-toolchain:

```
./configure --prefix=/opt/riscv --with-arch=rv32gc --with-abi=ilp32d --enable-linux
 make -j $(nproc)

```

```
riscv32-unknown-linux-gnu-gcc -v
Using built-in specs.
COLLECT_GCC=riscv32-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-linux-gnu/14.2.0/lto-wrapper
Target: riscv32-unknown-linux-gnu
Configured with: /home/aleck/src/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-pkgversion=g04696df0963 --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././gcc --disable-default-pie --disable-multilib --with-abi=ilp32d --with-arch=rv32gc_zicsr --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-O2    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2    -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 14.2.0 (g04696df0963)
```

linux

```
git checkout v5.4
make ARCH=riscv CROSS_COMPILE=riscv32-unknown-linux-gnu- defconfig
make ARCH=riscv CROSS_COMPILE=riscv32-unknown-linux-gnu- KCFLAGS="-march=rv32gc_zicsr"
```

I've got an error:
```
  CALL    scripts/checksyscalls.sh
<stdin>:830:2: warning: #warning syscall fstat64 not implemented [-Wcpp]
<stdin>:1127:2: warning: #warning syscall fstatat64 not implemented [-Wcpp]
<stdin>:1511:2: warning: #warning syscall clone3 not implemented [-Wcpp]
  CALL    scripts/atomic/check-atomics.sh
  CHK     include/generated/compile.h
  LD      arch/riscv/kernel/vdso/vdso-syms.o
riscv32-unknown-linux-gnu-ld: internal error: aborting at /home/aleck/src/riscv-gnu-toolchain/binutils/ld/ldlang.c:11034 in cmdline_check_object_only_section
riscv32-unknown-linux-gnu-ld: please report this bug
make[3]: *** [arch/riscv/kernel/vdso/Makefile:46: arch/riscv/kernel/vdso/vdso-syms.o] Error 1
make[2]: *** [scripts/Makefile.build:509: arch/riscv/kernel/vdso] Error 2
make[1]: *** [scripts/Makefile.build:509: arch/riscv/kernel] Error 2
make: *** [Makefile:1652: arch/riscv] Error 2
```

TommyMurphyTM1234, 2025-04-10:
You should log this upstream in the GCC bugbase since that's where ongoing maintenance/development (including RISC-V target specific work) is done these days and not in this repo or any GCC fork.

- https://gcc.gnu.org/bugs/

TommyMurphyTM1234, 2025-04-10:
> You should log this upstream in the GCC bugbase since that's where ongoing maintenance/development (including RISC-V target specific work) is done these days and not in this repo or any GCC fork.
> 
> * https://gcc.gnu.org/bugs/

Correction - the Binutils bugbase since this error arises in the linker and not the compiler.

- https://sourceware.org/binutils/docs/binutils/Bug-Reporting.html

petaleleo, 2025-04-11:
I did some trick
edit file scripts/dtc/dtc-lexer.lex.c
search yylloc and correct line code to:
extern YYLTYPE yylloc;

TommyMurphyTM1234, 2025-04-12:
> I did some trick edit file scripts/dtc/dtc-lexer.lex.c search yylloc and correct line code to: extern YYLTYPE yylloc;

I tried that but then get a different error:
```
  UPD     include/generated/utsrelease.h
  CC      scripts/mod/empty.o
cc1: error: ABI requires '-march=rv64'
make[1]: *** [scripts/Makefile.build:266: scripts/mod/empty.o] Error 1
make: *** [Makefile:1111: prepare0] Error 2
```
If you want feedback then you need to provide full details on how to reproduce your issue.

petaleleo, 2025-04-14:
It is not easy to reproduce my path to this error. 
I need some time for repeat my scenario.
Thank you very much for you attention for my case!

petaleleo, 2025-04-17:
So, i tried to reproduce my path of error but i did'n can . My path finished with success.
My target was to compile RISCV 32bit  linux with busybox on QEMU.
I got instruction from here: https://risc-v-getting-started-guide.readthedocs.io/en/latest/linux-qemu.html

riscv-gnu-toolchain:
sudo ./configure --prefix=/opt/riscv --with-arch=rv32gc_zicsr --with-abi=ilp32d --enable-linux
sudo make -j $(nproc) linux

$ riscv32-unknown-linux-gnu-gcc --version
riscv32-unknown-linux-gnu-gcc (g04696df09) 14.2.0

export PATH=/opt/riscv32/bin/:$PATH
export ARCH=riscv
export CROSS_COMPILE=riscv32-unknown-linux-gnu-

Linux:
git checkout master
make clean
make rv32_defconfig
make -j $(nproc)

 busybox-1.37.0:
make menuconfig
-- Settings -> Build options -> Build static binary (no shared libs)
make -j $(nproc)

image busybox32.bin:
dd if=/dev/zero of=busybox32.bin bs=1M count=64
mkfs.ext4 busybox32.bin
mkdir -p mnt32
sudo mount busybox32.bin mnt32
sudo mkdir -p bin etc dev lib proc sbin tmp usr usr/bin usr/lib usr/sbin
sudo cp busybox-1.37.0/busybox mnt32/bin
sudo ln -s ../bin/busybox bin/sh
sudo ln -s ../bin/busybox sbin/init
sudo umount mnt32

qemu:
qemu-system-riscv32   -nographic   -machine virt   -kernel linux/arch/riscv/boot/Image   -append "console=ttyS0 root=/dev/vda rw"   -drive file=busybox32.bin,format=raw,id=hd0,if=none   -device virtio-blk-device,drive=hd0

TommyMurphyTM1234, 2025-04-30:
Closing since this cannot be reproduced.

closed 2025-04-30 by TommyMurphyTM1234