riscv-collab/riscv-gnu-toolchain #1705: Support for Benchmark data

RRArinc, 2025-04-16:
Hi,
I got an opportunity to build RISCV toolchain versions supporting GCC 13.2 and and GCC 14.2 and compile few sample c programs.

Can anyone here support me what additional benefit of using GCC 14.2 version over GCC 13.2 version.
I learnt from chat GPT it is more aggressive register access and allocation.

Kindly let me know if anyone have data and share here on this topic for more understanding.

Thanks

TommyMurphyTM1234, 2025-04-16:
I would suggest looking at the release notes for the different versions of the various components used for each toolchain version to see what changes were made - GCC, Binutils, Glibc/Newlib, GDB, etc.

Liaoshihua, 2025-04-16:
You can see the details in https://gcc.gnu.org/gcc-14/changes.html

TommyMurphyTM1234, 2025-04-16:
> You can see the details in https://gcc.gnu.org/gcc-14/changes.html

Yes, but changes to other tools such as the Binutils assembler and linker and other components such as supporting libraries (e.g. Glibc/Newlib) may also have a material impact on the linked executable programs ultimately generated. So the release notes/change logs for those should also be checked in case there are any relevant changes.

TommyMurphyTM1234, 2025-04-21:
Closing the issue as it's not really a `riscv-gnu-toolchain` issue but a general question about GCC and related tools.

closed 2025-04-21 by TommyMurphyTM1234