riscv-collab/riscv-gnu-toolchain #1708: Unable to build for Linux

AtharvaPhadnis, 2025-04-21:
I am trying to build the toolchain for Linux (kernel version : 5.15.0-131-generic). I am on x86_64, Intel(R) Xeon(R) CPU E5-2660 v3 @ 2.60GHz

It seems like the build process is not able to fetch the `gcc` submodule, or is fetching incorrect commits for the `binutils` submodule. I see the latest commit has caused the build to fail in your Github actions workflow as well (same error) : https://github.com/riscv-collab/riscv-gnu-toolchain/actions/runs/14570139348/job/40865783638

I am attaching the output of `make linux` below. Prior to this I ran `./configure --prefix=/users/phadnis/riscv`

```
cd /users/phadnis/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /users/phadnis/riscv-gnu-toolchain/gcc/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /users/phadnis/riscv-gnu-toolchain/gcc/
cd /users/phadnis/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /users/phadnis/riscv-gnu-toolchain/glibc/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /users/phadnis/riscv-gnu-toolchain/glibc/
cd /users/phadnis/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /users/phadnis/riscv-gnu-toolchain/binutils/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /users/phadnis/riscv-gnu-toolchain/binutils/
mkdir -p /users/phadnis/riscv/.test || \
        (echo "Sorry, you don't have permission to write to" \
         "'/users/phadnis/riscv'. Please make sure that the location is " \
         "writable or use --prefix to specify another destination.'" \
         && exit 1)
mkdir -p /users/phadnis/riscv/sysroot/usr/
cd /users/phadnis/riscv-gnu-toolchain && \
flock `git rev-parse --git-dir`/config git submodule init /users/phadnis/riscv-gnu-toolchain/gdb/ && \
flock `git rev-parse --git-dir`/config git submodule update --progress --depth 1 /users/phadnis/riscv-gnu-toolchain/gdb/
cp -a /users/phadnis/riscv-gnu-toolchain/linux-headers/include /users/phadnis/riscv/sysroot/usr/
rm -r /users/phadnis/riscv/.test
mkdir -p stamps/ && touch stamps/check-write-permission
Submodule 'gcc' (https://gcc.gnu.org/git/gcc.git) registered for path 'gcc'
Submodule 'glibc' (https://sourceware.org/git/glibc.git) registered for path 'glibc'
Cloning into '/users/phadnis/riscv-gnu-toolchain/gcc'...
mkdir -p stamps/ && touch stamps/build-linux-headers
Submodule 'binutils' (https://sourceware.org/git/binutils-gdb.git) registered for path 'binutils'
Cloning into '/users/phadnis/riscv-gnu-toolchain/glibc'...
Submodule 'gdb' (https://sourceware.org/git/binutils-gdb.git) registered for path 'gdb'
Cloning into '/users/phadnis/riscv-gnu-toolchain/binutils'...
Cloning into '/users/phadnis/riscv-gnu-toolchain/gdb'...
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: clone of 'https://sourceware.org/git/glibc.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/glibc' failed
Failed to clone 'glibc'. Retry scheduled
Cloning into '/users/phadnis/riscv-gnu-toolchain/glibc'...
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: clone of 'https://sourceware.org/git/binutils-gdb.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/gdb' failed
Failed to clone 'gdb'. Retry scheduled
Cloning into '/users/phadnis/riscv-gnu-toolchain/gdb'...
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: clone of 'https://gcc.gnu.org/git/gcc.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/gcc' failed
Failed to clone 'gcc'. Retry scheduled
Cloning into '/users/phadnis/riscv-gnu-toolchain/gcc'...
fatal: unable to access 'https://sourceware.org/git/glibc.git/': Empty reply from server
fatal: clone of 'https://sourceware.org/git/glibc.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/glibc' failed
Failed to clone 'glibc' a second time, aborting
make: *** [Makefile:348: /users/phadnis/riscv-gnu-toolchain/glibc/.git] Error 1
make: *** Waiting for unfinished jobs....
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: clone of 'https://gcc.gnu.org/git/gcc.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/gcc' failed
Failed to clone 'gcc' a second time, aborting
make: *** [Makefile:348: /users/phadnis/riscv-gnu-toolchain/gcc/.git] Error 1
remote: Enumerating objects: 40871, done.
remote: Counting objects: 100% (40871/40871), done.
remote: Compressing objects: 100% (31219/31219), done.
remote: Enumerating objects: 40871, done.        | 2.35 MiB/s
remote: Counting objects: 100% (40871/40871), done.        /s
remote: Compressing objects: 100% (31219/31219), done.
remote: Total 40871 (delta 10645), reused 24678 (delta 8259), pack-reused 0
Receiving objects: 100% (40871/40871), 78.17 MiB | 3.10 MiB/s, done.
Resolving deltas: 100% (10645/10645), done. MiB | 3.92 MiB/s
remote: Total 40871 (delta 10648), reused 24678 (delta 8259), pack-reused 0
fatal: clone of 'https://gcc.gnu.org/git/gcc.git' into submodule path '/users/phadnis/riscv-gnu-toolchain/gcc' failed
Failed to clone 'gcc' a second time, aborting
make: *** [Makefile:348: /users/phadnis/riscv-gnu-toolchain/gcc/.git] Error 1
remote: Enumerating objects: 40871, done.
remote: Counting objects: 100% (40871/40871), done.
remote: Compressing objects: 100% (31219/31219), done.
remote: Enumerating objects: 40871, done.        | 2.35 MiB/s
remote: Counting objects: 100% (40871/40871), done.        /s
remote: Compressing objects: 100% (31219/31219), done.
remote: Total 40871 (delta 10645), reused 24678 (delta 8259), pack-reused 0
Receiving objects: 100% (40871/40871), 78.17 MiB | 3.10 MiB/s, done.
Resolving deltas: 100% (10645/10645), done. MiB | 3.92 MiB/s
remote: Total 40871 (delta 10648), reused 24678 (delta 8259), pack-reused 0
Receiving objects: 100% (40871/40871), 78.15 MiB | 2.64 MiB/s, done.
Resolving deltas: 100% (10648/10648), done.
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0
error: RPC failed; curl 16 Error in the HTTP2 framing layer
fatal: the remote end hung up unexpectedly
fatal: Fetched in submodule path 'binutils', but it did not contain 815d9a14cbbb3b81843f7566222c87fb22e7255d. Direct fetching of that commit failed.
fatal:
make: *** [Makefile:348: /users/phadnis/riscv-gnu-toolchain/binutils/.git] Error 128
remote: Enumerating objects: 24003, done.
remote: Counting objects: 100% (24003/24003), done.
remote: Compressing objects: 100% (9550/9550), done.
remote: Total 12594 (delta 11396), reused 3766 (delta 2973), pack-reused 0
Receiving objects: 100% (12594/12594), 4.76 MiB | 718.00 KiB/s, done.
Resolving deltas: 100% (11396/11396), completed with 8361 local objects.
From https://sourceware.org/git/binutils-gdb
 * branch              23c84db5b3cb4e8a0d555c76e1a0ab56dc8355f3 -> FETCH_HEAD
Submodule path 'gdb': checked out '23c84db5b3cb4e8a0d555c76e1a0ab56dc8355f3'
Submodule path 'gdb': checked out '23c84db5b3cb4e8a0d555c76e1a0ab56dc8355f3'
```

If this is not due to some issues in the upstream of the submodules, could someone please let me know if I am doing something incorrect on my end?

TommyMurphyTM1234, 2025-04-21:
Known issue:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669

Please use the workaround here for now: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2657214585
  - https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1669#issuecomment-2682013720

closed 2025-04-21 by TommyMurphyTM1234

AtharvaPhadnis, 2025-04-21:
Thank you!
Should have checked the recent issues first, my bad...