riscv-collab/riscv-gnu-toolchain #1709: add llvm  libc++ support

calvin2021y, 2025-04-28:
thanks for this great projects, is there any plan to add add llvm  libc++ support?

cmuellner, 2025-04-28:
I'm not aware that anyone is working on this.
But PRs are welcome!

TommyMurphyTM1234, 2025-04-30:
> any plan to add add llvm libc++ support?

What exactly do you mean?
When you build the LLVM toolchain doesn't it build C++ libraries?

- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#llvm--clang

> To build LLVM with C and C++ support the configure flag `--enable-llvm` can be used.

calvin2021y, 2025-05-01:
Thanks for the tips.

`--enable-llvm` build with newlib, no `compiler-rt` because `sysroot` not work like linux_glib target.

`--enable-llvm` can not build with `musl`.

TommyMurphyTM1234, 2025-05-01:
> `--enable-llvm` build with newlib, no `compiler-rt` because `sysroot` not work like linux_glib target.

Sorry, I don't really understand what this means. 
Maybe you can explain in more detail?

calvin2021y, 2025-05-01:
if you try build with this,  will fail with sysroot problem.

```diff
 stamps/build-llvm-newlib: $(LLVM_SRCDIR) $(LLVM_SRC_GIT) $(BINUTILS_SRCDIR) $(BINUTILS_SRC_GIT) \
@@ -1218,7 +1176,10 @@ stamps/build-llvm-newlib: $(LLVM_SRCDIR) $(LLVM_SRC_GIT) $(BINUTILS_SRCDIR) $(BI
            -DCMAKE_BUILD_TYPE=Release \
            -DLLVM_TARGETS_TO_BUILD="RISCV" \
            -DLLVM_ENABLE_PROJECTS="clang;lld" \
+           -DLLVM_ENABLE_RUNTIMES="compiler-rt" \
            -DLLVM_DEFAULT_TARGET_TRIPLE="$(NEWLIB_TUPLE)" \
+           -DDEFAULT_SYSROOT="$(INSTALL_DIR)/$(NEWLIB_TUPLE)" \
+           -DLLVM_RUNTIME_TARGETS=$(call make_tuple,$(XLEN),elf) \
            -DLLVM_INSTALL_TOOLCHAIN_ONLY=On \
            -DLLVM_BINUTILS_INCDIR=$(BINUTILS_SRCDIR)/include \
            -DLLVM_PARALLEL_LINK_JOBS=4
```

TommyMurphyTM1234, 2025-05-01:
Sorry, I still don't understand. I think that you may need to provide a clearer and more detailed description of the issue(s?) including detailed instructions for reproduction and maybe relevant logs and/or error messages.

index-0, 2025-05-08:
If you want to use `libclang_rt.builtins` instead of `libgcc` for baremetal use, you can edit the following:

```diff
diff --git a/Makefile.in b/Makefile.in
index b597a40..31e9486 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -1218,10 +1218,20 @@ stamps/build-llvm-newlib: $(LLVM_SRCDIR) $(LLVM_SRC_GIT) $(BINUTILS_SRCDIR) $(BI
            -DCMAKE_BUILD_TYPE=Release \
            -DLLVM_TARGETS_TO_BUILD="RISCV" \
            -DLLVM_ENABLE_PROJECTS="clang;lld" \
+           -DLLVM_ENABLE_RUNTIMES="compiler-rt" \
            -DLLVM_DEFAULT_TARGET_TRIPLE="$(NEWLIB_TUPLE)" \
+           -DDEFAULT_SYSROOT="$(INSTALL_DIR)/$(NEWLIB_TUPLE)" \
            -DLLVM_INSTALL_TOOLCHAIN_ONLY=On \
            -DLLVM_BINUTILS_INCDIR=$(BINUTILS_SRCDIR)/include \
-           -DLLVM_PARALLEL_LINK_JOBS=4
+           -DLLVM_PARALLEL_LINK_JOBS=4 \
+           -DCOMPILER_RT_BUILD_BUILTINS=ON \
+           -DCOMPILER_RT_BUILD_LIBFUZZER=OFF \
+           -DCOMPILER_RT_BUILD_MEMPROF=OFF \
+           -DCOMPILER_RT_BUILD_PROFILE=OFF \
+           -DCOMPILER_RT_BUILD_SANITIZERS=OFF \
+           -DCOMPILER_RT_BUILD_XRAY=OFF \
+           -DCOMPILER_RT_BAREMETAL_BUILD=ON \
+           -DCOMPILER_RT_DEFAULT_TARGET_ONLY=ON
        $(MAKE) -C $(notdir $@)
        $(MAKE) -C $(notdir $@) install
        cp $(notdir $@)/lib/LLVMgold.$(SHARED_LIB_SUFFIX) $(INSTALL_DIR)/lib
```

As for `compiler_rt` as a whole, I don't think it's possible to build for a baremetal target, as it depends heavily on OS features (threads, files, allocators).

cmuellner, 2025-05-08:
As @index-0 mentioned, libc++ does not support baremetal toolchains (just like glibc).
And for the Linux toolchain, libc++ is included.

If there is interest in a build configuration with `libclang_rt.builtins` (as mentioned by @index-0), we could add that (somebody needs to step up and create a PR).

@calvin2021y, is your question answered?

calvin2021y, 2025-05-08:
Thanks, very clear.

libclang_rt.builtins is what I need.

cmuellner, 2025-05-08:
Ok, then I will close this.

And as mentioned, feel free to create a PR that adds a new build target for l`ibclang_rt.builtins`.