riscv-collab/riscv-gnu-toolchain #1715: cannot build, reported `/home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: unrecognised options "--64"`

GEJXD, 2025-05-08:
```text
./configure --prefix=/home/JXD/riscv --with-arch=rv64gcv_zicsr --with-abi=lp64d --enable-qemu-system --enable-llvm --enable-linux

checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking for grep that handles long lines and -e... /usr/bin/grep
checking for fgrep... /usr/bin/grep -F
checking for grep that handles long lines and -e... (cached) /usr/bin/grep
checking for bash... /bin/sh
checking for __gmpz_init in -lgmp... yes
checking for mpfr_init in -lmpfr... yes
checking for mpc_init2 in -lmpc... yes
checking for curl... /usr/bin/curl
checking for wget... /usr/bin/wget
checking for ftp... no
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/wrapper/awk/awk
config.status: creating scripts/wrapper/sed/sed

make linux 2>&1 | tee build.log

/home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: 未识别的选项 "--64"
make[2]: *** [Makefile:2995：build/genmodes.o] 错误 1
make[2]: 离开目录“/home/JXD/Project/opencamp/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc”
make[1]: *** [Makefile:4673：all-gcc] 错误 2
make[1]: 离开目录“/home/JXD/Project/opencamp/riscv-gnu-toolchain/build-gcc-linux-stage1”
make: *** [Makefile:472：stamps/build-gcc-linux-stage1] 错误 2

for full detail please see build.zip
```

[build.zip](https://github.com/user-attachments/files/20098182/build.zip)

TommyMurphyTM1234, 2025-05-08:
> sed -i '/shallow = true/d' .gitmodules
> sed -i 's/--depth 1//g' Makefile.in

This is not the cause of the problem but this workaround is no longer needed since the following PR was recently merged:

- https://github.com/riscv-collab/riscv-gnu-toolchain/pull/1702

This error seems strange as it complains about `--64` but that option is not passed in the command line just before the error:

> ```
> g++ -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
>	-o build/genmodes.o ../.././gcc/gcc/genmodes.cc
> /home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: 未识别的选项 "--64"
> ```

`未识别的选项` seems to translate as `unrecognized option`.

Are you sure that your environment doesn't contain some variable (e.g. `ASFLAGS` or something similar perhaps) that is causing `--64` to be passed here? What is the output of the `env` command on your host?

I've kicked off a build myself to see if I can reproduce this but I expect not to be able to as I have built the tools many times recently without any issues.

TommyMurphyTM1234, 2025-05-08:
Actually, looking at this again:
```
/home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: 未识别的选项 "--64"
```
That path to `as` might suggest that the build process is picking up the RISC-V Linux `as` instead of the hosts's native x86_64 `as`:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... riscv64-unknown-linux-gnu
```
So perhaps your system has a copy of the RISC-V Linux toolchain installed (and aliased to `gcc`, `as` etc, without `riscv64-unknown-linux-gnu-` prefix) and this us getting picked up and causing your build to fail?

What do the following return?
```
which gcc
which as

gcc -v
as -v
```

TommyMurphyTM1234, 2025-05-08:
Similarly, I'm not sure that this is correct and maybe suggests a problem with your environment:
```
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking target system type... riscv64-unknown-linux-gnu

...

checking for riscv64-unknown-linux-gnu-cc... no
checking for riscv64-unknown-linux-gnu-gcc... no
checking for riscv64-unknown-linux-gnu-c++... no
checking for riscv64-unknown-linux-gnu-g++... no
checking for riscv64-unknown-linux-gnu-cxx... no
checking for riscv64-unknown-linux-gnu-gxx... no
checking for riscv64-unknown-linux-gnu-gcc... no
checking for riscv64-unknown-linux-gnu-gfortran... no
checking for riscv64-unknown-linux-gnu-gccgo... no
checking for riscv64-unknown-linux-gnu-gdc... no
checking for riscv64-unknown-linux-gnu-gm2... no
checking for ar... /home/JXD/riscv/riscv64-unknown-linux-gnu/bin/ar
checking for as... /home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as
```

GEJXD, 2025-05-08:
> Actually, looking at this again:
> 
> ```
> /home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: 未识别的选项 "--64"
> ```
> 
> That path to `as` might suggest that the build process is picking up the RISC-V Linux `as` instead of the hosts's native x86_64 `as`:
> 
> ```
> checking build system type... x86_64-pc-linux-gnu
> checking host system type... x86_64-pc-linux-gnu
> checking target system type... riscv64-unknown-linux-gnu
> ```
> 
> So perhaps your system has a copy of the RISC-V Linux toolchain installed (and aliased to `gcc`, `as` etc, without `riscv64-unknown-linux-gnu-` prefix) and this us getting picked up and causing your build to fail?
> 
> What do the following return?
> 
> ```
> which gcc
> which as
> 
> gcc --version
> as --version
> ```

Thanks to reply, the outputs are:
```
/usr/bin/gcc
/usr/bin/as
gcc (GCC) 15.1.1 20250425
Copyright © 2025 Free Software Foundation, Inc.
本程序是自由软件；请参看源代码的版权声明。本软件没有任何担保；
包括没有适销性和某一专用目的下的适用性担保。
GNU 汇编器 (GNU Binutils) 2.44.0
Copyright (C) 2025 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or later.
This program has absolutely no warranty.
This assembler was configured for a target of `x86_64-pc-linux-gnu'.
```
and the `env` output are:
```
SHELL=/usr/bin/zsh
HYPRLAND_CMD=Hyprland
XDG_SESSION_PATH=/org/freedesktop/DisplayManager/Session1
HYDE_STATE_HOME=/home/JXD/.local/state/hyde
GTK_IM_MODULE=fcitx
XDG_BACKEND=wayland
QT_WAYLAND_DISABLE_WINDOWDECORATION=1
XDG_DATA_HOME=/home/JXD/.local/share
XDG_CONFIG_HOME=/home/JXD/.config
XMODIFIERS=@im=fcitx
LIBVA_DRIVER_NAME=nvidia
DESKTOP_SESSION=hyprland
ELECTRON_OZONE_PLATFORM_HINT=auto
HL_INITIAL_WORKSPACE_TOKEN=56c41459-f5d9-40af-abb5-a3382d85ffc0
SCREENRC=/home/JXD/.config/screen/screenrc
XDG_SEAT=seat0
PWD=/home/JXD/Project/opencamp
XDG_VIDEOS_DIR=/home/JXD/Videos
LOGNAME=JXD
XDG_SESSION_DESKTOP=Hyprland
QT_QPA_PLATFORMTHEME=qt6ct
XDG_SESSION_TYPE=wayland
HYDE_CONFIG_HOME=/home/JXD/.config/hyde
_=/usr/bin/env
XDG_PICTURES_DIR=/home/JXD/Pictures
MOTD_SHOWN=pam
HOME=/home/JXD
[... 41 lines cut ...]
HYDE_DATA_HOME=/home/JXD/.local/share/hyde
OLDPWD=/home/JXD/Project/opencamp/riscv-gnu-toolchain
KITTY_WINDOW_ID=1
TERM=xterm-kitty
COLORTERM=truecolor
KITTY_PID=20300
KITTY_PUBLIC_KEY=1:Kyv<C^d&KWYr*bf6j;&F5IpqCw>pG~`at?rQWEqf
TERMINFO=/usr/lib/kitty/terminfo
KITTY_INSTALLATION_DIR=/usr/lib/kitty
P9K_SSH=0
_P9K_SSH_TTY=/dev/pts/3
ZSH=/home/JXD/.oh-my-zsh
PAGER=less
LESS=-R
LSCOLORS=Gxfxcxdxbxegedabagacad
LS_COLORS=rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.7z=01;31:*.ace=01;31:*.alz=01;31:*.apk=01;31:*.arc=01;31:*.arj=01;31:*.bz=01;31:*.bz2=01;31:*.cab=01;31:*.cpio=01;31:*.crate=01;31:*.deb=01;31:*.drpm=01;31:*.dwm=01;31:*.dz=01;31:*.ear=01;31:*.egg=01;31:*.esd=01;31:*.gz=01;31:*.jar=01;31:*.lha=01;31:*.lrz=01;31:*.lz=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.lzo=01;31:*.pyz=01;31:*.rar=01;31:*.rpm=01;31:*.rz=01;31:*.sar=01;31:*.swm=01;31:*.t7z=01;31:*.tar=01;31:*.taz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tgz=01;31:*.tlz=01;31:*.txz=01;31:*.tz=01;31:*.tzo=01;31:*.tzst=01;31:*.udeb=01;31:*.war=01;31:*.whl=01;31:*.wim=01;31:*.xz=01;31:*.z=01;31:*.zip=01;31:*.zoo=01;31:*.zst=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.jxl=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:
NVM_DIR=/home/JXD/.nvm
NVM_CD_FLAGS=-q
NVM_BIN=/home/JXD/.nvm/versions/node/v20.18.2/bin
NVM_INC=/home/JXD/.nvm/versions/node/v20.18.2/include/node
CRYPTOGRAPHY_OPENSSL_NO_LEGACY=true
P9K_TTY=old
_P9K_TTY=/dev/pts/3
```
But strangely, I found `riscv64-linux-gnu-as` and other RISC-V tools in the `/usr/bin` directory, but I have no memory of ever installing the related toolchain myself。

TommyMurphyTM1234, 2025-05-08:
> But strangely, I found `riscv64-linux-gnu-as` and other RISC-V tools in the `/usr/bin` directory, but I have no memory of ever installing the related toolchain myself。

Sounds like that's your problem so - the build process is finding and using a RISC-V assembler when it should be using a native x86_64 one.

GEJXD, 2025-05-08:
> My build log differs from yours at the same stage of the build:
> 
> * My log:
> 
> ```
> make[2]: Entering directory '/home/user/Downloads/issue-1715/build-gcc-linux-stage1/gcc'
> TARGET_CPU_DEFAULT="" \
> HEADERS="auto-host.h ansidecl.h" DEFINES="" \
> /bin/bash ../.././gcc/gcc/mkconfig.sh config.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="options.h insn-constants.h config/elfos.h config/gnu-user.h config/linux.h config/glibc-stdint.h config/riscv/riscv.h config/riscv/linux.h config/initfini-array.h defaults.h" DEFINES="LIBC_GLIBC=1 LIBC_UCLIBC=2 LIBC_BIONIC=3 LIBC_MUSL=4 DEFAULT_LIBC=LIBC_GLIBC ANDROID_DEFAULT=0 HEAP_TRAMPOLINES_INIT=0 TARGET_DEFAULT_ASYNC_UNWIND_TABLES=1 TARGET_DEFAULT_ISA_SPEC=ISA_SPEC_CLASS_20191213 TARGET_RISCV_ATTRIBUTE=1 TARGET_RISCV_DEFAULT_ARCH=rv64imafdcv_zicsr_zifencei_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b TARGET_RISCV_DEFAULT_ABI=lp64d" \                                                                                                                                                                                           /bin/bash ../.././gcc/gcc/mkconfig.sh tm.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="config/riscv/riscv-protos.h config/linux-protos.h tm-preds.h" DEFINES="" \
> /bin/bash ../.././gcc/gcc/mkconfig.sh tm_p.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="auto-host.h ansidecl.h" DEFINES="" \
> /bin/bash ../.././gcc/gcc/mkconfig.sh bconfig.h
> g++ -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \                                                                                                          -o build/genmodes.o ../.././gcc/gcc/genmodes.cc                                                                                                                                     g++ -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \                                                                                                          -o build/errors.o ../.././gcc/gcc/errors.cc                                                                                                                                         g++   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -static-libstdc++ -static-libgcc  -o build/genmodes \                                                                                                                                                                              build/genmodes.o build/errors.o ../build-x86_64-pc-linux-gnu/libiberty/libiberty.a
> build/genmodes -h > tmp-modes.h
> /bin/bash ../.././gcc/gcc/../move-if-change tmp-modes.h insn-modes.h
> echo timestamp > s-modes-h
> build/genmodes -i > tmp-modes-inline.h
> /bin/bash ../.././gcc/gcc/../move-if-change tmp-modes-inline.h \
>   insn-modes-inline.h
> echo timestamp > s-modes-inline-h
> ```
> 
> * Your log:
> 
> ```
> make[2]: 进入目录“/home/JXD/Project/opencamp/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc”
> TARGET_CPU_DEFAULT="" \
> HEADERS="auto-host.h ansidecl.h" DEFINES="" \
> /bin/sh ../.././gcc/gcc/mkconfig.sh config.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="options.h insn-constants.h config/elfos.h config/gnu-user.h config/linux.h config/glibc-stdint.h config/riscv/riscv.h config/riscv/linux.h config/initfini-array.h defaults.h" DEFINES="LIBC_GLIBC=1 LIBC_UCLIBC=2 LIBC_BIONIC=3 LIBC_MUSL=4 DEFAULT_LIBC=LIBC_GLIBC ANDROID_DEFAULT=0 HEAP_TRAMPOLINES_INIT=0 TARGET_DEFAULT_ASYNC_UNWIND_TABLES=1 TARGET_DEFAULT_ISA_SPEC=ISA_SPEC_CLASS_20191213 TARGET_RISCV_ATTRIBUTE=1 TARGET_RISCV_DEFAULT_ARCH=rv64imafdcv_zicsr_zifencei_zve32f_zve32x_zve64d_zve64f_zve64x_zvl128b_zvl32b_zvl64b TARGET_RISCV_DEFAULT_ABI=lp64d" \
> /bin/sh ../.././gcc/gcc/mkconfig.sh tm.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="config/riscv/riscv-protos.h config/linux-protos.h tm-preds.h" DEFINES="" \
> /bin/sh ../.././gcc/gcc/mkconfig.sh tm_p.h
> TARGET_CPU_DEFAULT="" \
> HEADERS="auto-host.h ansidecl.h" DEFINES="" \
> /bin/sh ../.././gcc/gcc/mkconfig.sh bconfig.h
> g++ -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings   -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
> 	-o build/genmodes.o ../.././gcc/gcc/genmodes.cc
> /home/JXD/riscv/riscv64-unknown-linux-gnu/bin/as: 未识别的选项 "--64"
> make[2]: *** [Makefile:2995：build/genmodes.o] 错误 1
> make[2]: 离开目录“/home/JXD/Project/opencamp/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc”
> make[1]: *** [Makefile:4673：all-gcc] 错误 2
> make[1]: 离开目录“/home/JXD/Project/opencamp/riscv-gnu-toolchain/build-gcc-linux-stage1”
> make: *** [Makefile:472：stamps/build-gcc-linux-stage1] 错误 2
> ```

Thanks, i will check it later

closed 2025-05-08 by GEJXD

GEJXD, 2025-05-16:
The `checking target system type... riscv64-unknown-linux-gnu` in `build.log` seems to be correct, I downloaded the log from other issues and it shows the same.

The problem seems to be that after compiling out `riscv64-unknow-linux-as`, the native `gcc` calls this as the assembler instead of the native `as`. The option `--64` is only available on i386-related architectures and is automatically passed to `as` as a default parameter on 64-based machines.

Oddly enough, I see that your log shows the shell used as `/bin/bash`, but mine shows `/bin/sh`. I try to run the build command directly using `sh` and it builds successfully.