riscv-collab/riscv-gnu-toolchain #1716: Do we support building 'compiler-rt' when "--enable-llvm" is enabled?

BaoshanPang, 2025-05-13:
It seems by default compiler-rt is not built, is there a option to enable it? Thanks.

TommyMurphyTM1234, 2025-05-13:
Do you mean for the bare-metal or the Linux toolchain build? 

Perhaps this is relevant? 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1709#issuecomment-2862843154

BaoshanPang, 2025-05-13:
Thanks for the pointer.
Yes, I mean for the ''bare-metal". I am able to build libclang_rt.builtins using the suggested change. but I am actually looking for a way to build 'profile'. I get this error after setting ON to COMPILER_RT_BUILD_PROFILE:
```

/home/bpang/myworks/github/riscv-gnu-toolchain/llvm/compiler-rt/lib/profile/GCDAProfiling.c:24:10: fatal error: 'errno.h' file not found
   24 | #include <errno.h>
      |          ^~~~~~~~~

```

I have been able to build 'profile' for Linux toolchain with ABI=lp32d, but can't find a way to build it for ABI=ilp32. I need to use a 'soft-float' version of it.

TommyMurphyTM1234, 2025-05-13:
> Yes, I mean for the ''bare-metal".

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1709#issuecomment-2862843154

> As for `compiler_rt` as a whole, I don't think it's possible to build for a baremetal target, as it depends heavily on OS features (threads, files, allocators).

You should probably try building what you're looking for from the upstream LLVM/Clang project repos and if it doesn't work there then log the issue/discussion upstream.