riscv-collab/riscv-gnu-toolchain #1729: How to use -fsanitize=undefined in riscv-gcc

aaaaala, 2025-06-04:
**Problem Description:**

I encountered an issue while trying to use the `-fsanitize=undefined` flag with RISC-V GCC. The compiler cannot find the `-lubsan` library during the linking stage. Here are the terminal records:

``` bash
riscv64-unknown-linux-gnu-gcc -O2 -fsanitize=undefined -lubsan -o build/output src/*
/opt/riscv-gnu-toolchain/lib/gcc/riscv64-unknown-linux-gnu/14.2.0/../../../../riscv64-unknown-linux-gnu/bin/ld: cannot find -lubsan
collect2: error: ld returned 1 exit status.
```
- RISC-V GCC Version: 14.2.0
- Operating System: Ubuntu 22.04

However, there will be no such issue when compiling with the local x86 gcc. 

So, how can I resolve this issue and successfully use -fsanitize=undefined with RISC-V GCC

aaaaala, 2025-06-04:
Thank you, I have solved this problem now.

TommyMurphyTM1234, 2025-06-04:
> Thank you, I have solved this problem now.

How? It may help others to know. Thanks.

aaaaala, 2025-06-04:
The way I used `--enable-libsanitizer` when compiling riscv-gcc was inappropriate, this caused the linker to miss the library(lubsan).

It is mentioned in this web address, https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1666