riscv-collab/riscv-gnu-toolchain #1734: Building toolchain with 64 bits architecture failing.

FloDarPie, 2025-06-11:
I'm following the tutorial for Clang+LLVM, on my 
- *Linux pro 6.1.0-37-amd64 SMP PREEMPT_DYNAMIC Debian 6.1.140-1 (2025-05-22) x86_64 GNU/Linux*

Instruction : 
```
# Build rv64gc toolchain with LLVM
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
make -j$(nproc) all build-sim SIM=qemu
```

Use case :
```
RISCV=$HOME/cross_compil/riscv64-clang+llvm
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64d
make -j4 all 
```
A **g++** error is raised. I don't think this is normal. I understand that the ABI lp64d is in fact not 

[configure_log.txt](https://github.com/user-attachments/files/20694503/configure_log.txt)
[install_log.txt](https://github.com/user-attachments/files/20694504/install_log.txt)

### More details

If i do :
```
RISCV=$HOME/cross_compil/riscv32-clang+llvm
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv32gc --with-abi=lp32d
make -j4 all 
```

It's well installed.

If i do (modify the ABI norm) :
```
RISCV=$HOME/cross_compil/riscv64-clang+llvm
./configure --prefix=$RISCV --enable-llvm --enable-linux --with-arch=rv64gc --with-abi=lp64
make -j4 all 
```

I crash with
```
g++ -no-pie   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H -no-pie -static-libstdc++ -static-libgcc  -o lto1 \
	lto/lto-lang.o lto/lto.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o lto/lto-common.o simple-diagnostic-path.o lazy-diagnostic-path.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -lmpc -lmpfr -lgmp -rdynamic  -lz  libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
g++ -no-pie   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H -no-pie -static-libstdc++ -static-libgcc  -o lto-dump \
	lto/lto-lang.o lto/lto-object.o attribs.o lto/lto-partition.o lto/lto-symtab.o lto/lto-dump.o lto/lto-common.o simple-diagnostic-path.o lazy-diagnostic-path.o libbackend.a main.o libcommon-target.a libcommon.a ../libcpp/libcpp.a ../libdecnumber/libdecnumber.a  -lmpc -lmpfr -lgmp -rdynamic  -lz  libcommon.a ../libcpp/libcpp.a   ../libbacktrace/.libs/libbacktrace.a ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a 
collect2: fatal error: ld terminated with signal 9 [Processus arrêté]
compilation terminated.
make[2]: *** [../.././gcc/gcc/lto/Make-lang.in:102 : lto-dump] Erreur 1
make[2]: *** Attente des tâches non terminées....
rm gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod fsf-funding.pod gpl.pod cpp.pod gcov.pod lto-dump.pod
make[2] : on quitte le répertoire « /home/florian/Documents/apprentissage/compilation/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc »
make[1]: *** [Makefile:4693 : all-gcc] Erreur 2
make[1] : on quitte le répertoire « /home/florian/Documents/apprentissage/compilation/riscv-gnu-toolchain/build-gcc-linux-stage1 »
make: *** [Makefile:471 : stamps/build-gcc-linux-stage1] Erreur 2
```

PS:
It's my first issue, i'm open to add more information or perhaps i've misused the *./configure* file. Sorry for the disturbance.

cmuellner, 2025-06-11:
Does this also happen with `make all` instead of `make -j4 all`?
Also, what does `journalctl` say about the segfault?

My guess is that you run out of memory, and the kernel kills the linker process (see https://www.kernel.org/doc/html/next/admin-guide/mm/concepts.html#oom-killer).

TommyMurphyTM1234, 2025-06-11:
> Does this also happen with `make all` instead of `make -j4 all`?

The first step when a problem occurs with a parallel build it to see what happens with a non-parallel build. And the parallel build log is almost impossible to read because output from different jobs is interleaved so a log from a non-parallel build is needed for that reason too.

FloDarPie, 2025-06-13:
I have just finished compiling. Everything is working well. Thank you for your quick responses.

closed 2025-06-13 by FloDarPie