riscv-collab/riscv-gnu-toolchain #1737: I get the "X is not supported for ISA" error *too late* on compilation

Slicudis, 2025-07-06:
Every time I want to compile the toolchain, I get this "--with-abi=ilp32f is not supported for ISA" error after like 5 minutes of waiting for compilation, which is quite annoying. Shouldn't this error be detected just after using the "./configure" command? Why waste my time like that?

I'm not a dev, so please... could any programmer update it to detect it detect on start? Many people, including me, would appreciate it a lot.

TommyMurphyTM1234, 2025-07-06:
You need to explain more clearly what you're trying to do, what exactly the problem is, and how to reproduce it. 

> Every time I want to compile the toolchain, I get this "--with-abi=ilp32f is not supported for ISA" error after like 5 minutes of waiting for compilation

Every time? If you get the same error every time then why try the same thing without changing anything?

> Many people,

What people?

Slicudis, 2025-07-06:
> You need to explain more clearly what you're trying to do, what exactly the problem is, and how to reproduce it.
> 
> How exactly are you configuring and building your toolchain? What is the build log?
> 
> > Every time I want to compile the toolchain, I get this "--with-abi=ilp32f is not supported for ISA" error after like 5 minutes of waiting for compilation
> 
> Every time? If you get the same error every time then why try the same thing without changing anything?
> 
> > Many people,
> 
> What people?
> 
> > Why waste my time like that?
> 
> Why waste ours by logging such a poor issue/bug report? 😐

``
checking if mkdir takes one argument... no
--with-abi=lp64d is not supported for ISA
make[1]: *** [Makefile:4651: configure-gcc] Error 1
make[1]: Leaving directory '/mnt/e/heavy_programs/riscv-gnu-toolchain/build-gcc-newlib-stage1'
make: *** [Makefile:646: stamps/build-gcc-newlib-stage1] Error 2
root@SantiagoL:/mnt/e/heavy_programs/riscv-gnu-toolchain#
root@SantiagoL:/mnt/e/heavy_programs/riscv-gnu-toolchain#
root@SantiagoL:/mnt/e/heavy_programs/riscv-gnu-toolchain# riscv64-unknown-elf--elf
^[[Ariscv64-unknown-elf--elf: command not found
root@SantiagoL:/mnt/e/heavy_programs/riscv-gnu-toolchain# riscv64-unknown-elf
riscv64-unknown-elf: command not found
``

With people I mean friends. There's really nothing else to explain as I've tried with different configurations, like

``
./configure --with-arch=rv32imac_zifencei_zicond_zicbom_zicboz_zicbop --with-abi=ilp32
``

``
./configure --with-arch=rv32imac_zifencei_zicond_zicbom_zicboz_zicbop --with-abi=ilp32
``

``
./configure --enable-multilib
``

*nothing* worked. It just threw the same error of "--with-abi=ilp32f is not supported for ISA"

I read posts of other people for hours and found *no* answers. I have to test with different configurations and execute "make newlib" over and over and over again, waiting for 5-10 minutes on each attempt just to get that error. It's frustrating, ya know? I'm sorry if I'm just mad.

TommyMurphyTM1234, 2025-07-06:
You still haven't clearly explained the steps involved in reproducing your issue (e.g. the `git clone`, `configure` and `make` commands involved) or attached a full build log showing the issue.

Your original issue mentions this: 

> I get this "--with-abi=ilp32f is not supported for ISA" error

but the partial log output that you posted subsequently mentions this: 

> `--with-abi=lp64d is not supported for ISA`

Sounds like you may be using an incompatible `arch` and `abi`. But I'm guessing because you have not provided sufficient information.

TommyMurphyTM1234, 2025-07-06:
> 3\. I've tried with different configurations, like:
> 
> ...
> 
> `./configure --enable-multilib`

This worked fine for me on Ubuntu 24.04 LTS.
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1737
cd issue-1737
./configure --prefix=`pwd`/installed-tools --enable-multilib
make 2>&1 | tee build.log
```
- `build.log.zip`: [build.log.zip](https://github.com/user-attachments/files/21089607/build.log.zip)

TommyMurphyTM1234, 2025-07-06:
> 3\. I've tried with different configurations, like:
> 
> `./configure --with-arch=rv32imac_zifencei_zicond_zicbom_zicboz_zicbop --with-abi=ilp32`

And this:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1737
cd issue-1737
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imac_zifencei_zicond_zicbom_zicboz_zicbop --with-abi=ilp32
make 2>&1 | tee build.log
```
- Build log (zipped): 

For some reason I can't upload the build log:
```
<!-- Failed to upload "build2.log.zip" -->
```

TommyMurphyTM1234, 2025-07-07:
Can you provide full details of how to reproduce your problem?
E.g.

1. Details of the host OS and host tools
2. Your `git clone` command
3. Your `configure` command
4. Your `make` command
5. Your build log

Does the same issue occur on a clean host OS virtual or physical installation?

TommyMurphyTM1234, 2025-07-09:
Any update @Slicudis?

TommyMurphyTM1234, 2025-07-11:
Closing due to lack of update.
If full/clear instructions for reproducing this issue become available then please post them and reopen the issue.