riscv-collab/riscv-gnu-toolchain #1740: Enabling RISC-V Big-Endian Support in GCC Toolchain

Utatsu00, 2025-07-15:
I'd like to use the -mbig-endian option, which is part of the RISC-V GCC toolchain.

The GCC manual states that the -mbig-endian option targets riscv32be or riscv64be.

When I attempt to compile a RISC-V program in big-endian mode using the -mbig-endian option, do I need any special configurations? Or can I simply build the toolchain as usual and then just include the -mbig-endian option during compilation?

Utatsu00, 2025-07-16:
When I build riscv-gcc-toolchain, I set the --target option in configure to riscv32be-*-*, but when I actually run make, the setting is not reflected and riscv32-*-* is set as the target.
Is this a problem, or can the compiler executable riscv32-unknown-elf-gcc also be compiled with the option -mbig-endian to compile with the big-endian I'm assuming?

TommyMurphyTM1234, 2025-07-16:
> When I build riscv-gcc-toolchain

You'll need to state more clearly exactly how you're building your toolchain - e.g. your `git clone`, `configure`, `make` commands and maybe capture, zip and upload your build log.

TommyMurphyTM1234, 2025-07-17:
For what it's worth, if I build the toolchain in its default configuration (i.e. LE and default `arch/abi` of `rv64gc/lp64d` and with `--enable-multilib`) and compile a simple test source file to object it has the following in the disassembly:
```
test.o:     file format elf64-littleriscv
```
If I compile it with `-mbig-endian` it has:
```
test-be.o:     file format elf64-bigriscv
```
I don't see any difference in the actual code generated for a simple example but I'm not sure if that's significant for the trivial test code that I used.

However If I try to compile **and link to ELF** with `-mbig-endian` it fails:
```
./riscv64-unknown-elf-gcc test.c -mbig-endian
/home/user/Downloads/issue-1742/installed-tools/lib/gcc/riscv64-unknown-elf/15.1.0/../../../../riscv64-unknown-elf/bin/ld: /home/user/Downloads/issue-1742/installed-tools/lib/gcc/riscv64-unknown-elf/15.1.0/../../../../riscv64-unknown-elf/lib/crt0.o: ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf64-bigriscv'
...
```
which doesn't really surprise me since the toolchain (even with `--enable-multilib`) doesn't provide support for BE libs etc.

So while the toolchain itself can seemingly compile in BE mode, by default full support for BE (e.g. multilibs etc.) doesn't seem to be available automatically.

I would be inclined to ask the likes of the Codethink if they have any comments on how to build a BE capable toolchain (although their focus seems to be the Linux rather than bare metal toolchain in case that matters)...

TommyMurphyTM1234, 2025-08-14:
Any update @Utatsu00?
Does compiling with `-mbig-endian` do what you need?
If not then what specific problems do you have?

If changes/enhancements are required to support BE targets in the toolchin then this issue might need to be logged in the relevant upstream projects - e.g. GCC, Binutils, Newlib etc. - where ongoing development is done (not in this repo).

pz9115, 2025-08-15:
FYI,

An initial specification for RISC-V Big Endian ABI is under discussing in psABI group
https://github.com/riscv-non-isa/riscv-elf-psabi-doc/pull/470