riscv-collab/riscv-gnu-toolchain #1742: Failed to recompile tool chain.

buloruabutata, 2025-07-16:
I deleted the original tool chain, and then recompiled the riscv tool chain, but in the end I only installed tools such as gdb, but not gcc. It only took a few minutes to recompile.

```
make clean
./configure --prefix=$HOME/opt/riscv --enable-multilib
make -j16

ls ./riscv64-unknown-elf-*
./riscv64-unknown-elf-gdb  ./riscv64-unknown-elf-gdb-add-index  ./riscv64-unknown-elf-gstack  ./riscv64-unknown-elf-run
```

buloruabutata, 2025-07-16:
[build.log](https://github.com/user-attachments/files/21252511/build.log)，I also tried to make distclean.

buloruabutata, 2025-07-16:
I found the following, but I don't know how to solve it.
```
checking for cc... (cached) riscv64-unknown-elf-gcc
checking for gcc... (cached) riscv64-unknown-elf-gcc
```

TommyMurphyTM1234, 2025-07-16:
> and then recompiled the riscv tool chain

In my experience incremental builds, if that's what you're trying to do here, don't work (very well) with `riscv-gnu-toolchain` and it's more reliable to do full rebuilds.

See here for example: 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196#issuecomment-1438566964

maybe also using `ccache` to speed up builds. 

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1196#issuecomment-1566717860

TommyMurphyTM1234, 2025-07-16:
BTW 

> `make -j16`
> build.log

If you're attaching a build log for analysis it should be one for a build without parallel jobs as parallel build logs are impossible to analyse due to interleaving of log output from different jobs.

TommyMurphyTM1234, 2025-07-16:
> In order to add custom instructions

This is new information not included in your original issue.
You really need to clarify exactly what steps you are taking to build, modify, rebuild etc. your toolchain. 
E.g.:

1. Your `git clone` command
2. Your `configure` command(s?)
3. Your `make` command
4. Your non-parallel build logs (success and failure case if applicable)
5. Details of the changes that you are making to the toolchain sources

And this:

> After deleting the previously built toolchain and doing `make distclean` what does `which riscv64-unknown-elf-gcc` return?

cmuellner, 2025-07-16:
> In order to add custom instructions, I modified `include/opcode/riscv-opc.h` and `opcodes/riscv-opc.c` in binutils.

These changes likely broke your build.

We have the following dependencies for newlib: GCC2 and GDB
These have the following dependencies (the "->" means "depends on")
* GCC2 -> newlib -> GCC1 -> binutils
* GDB: -

Therefore, if Binutils fails to build successfully, the build process for GCC1, newlib, and GCC2 won't even begin.
However, GDB will happily build.

My recommendation:
* Don't edit files in or below the riscv-gnu-toolchain directory (unless you want to contribute to this project)
* Instead, use an external clone of Binutils for development
* During development, build/test your Binutils sources outside of the riscv-gnu-toolchain repo
* After your feature is implemented and tested, build riscv-gnu-toolchain using `./configure [...] --with-binutils-src=path/to/binutils-gdb`

To build/test Binutils, you could use something like this:
```
./configure --target=riscv64-unknown-linux-gnu --prefix=$(pwd)/install
make -j$(nproc)
make -j$(nproc) install
make check-binutils
make check-gas
make check-ld
```

buloruabutata, 2025-07-18:
After using single-task compilation, config.log in RISCV-GNU-Toolchain/Build-Newlib/RISCV64-Unknown-ELF/Newlib found the following information.
Error: internal: bad RISC-V opcode (unknown operand type `i')

I mistakenly passed in I in the args field of riscv_opcode, and `i' was not a sign of immediate number. After it was modified to Xtu*@*, it was compiled successfully, with the first * representing the length of immediate number and the second representing the lowest bit of immediate number. For example, zimm[4:0] is Xtu5@15.

closed 2025-07-18 by buloruabutata