riscv-collab/riscv-gnu-toolchain #1750: Macro errors during toolchain build

MahimaDesai, 2025-07-31:
Multiple errors during building toolchain even when using a particular tag as well as recent master

steps used for build are:
git clone --branch  2025.07.03 --recursive https://github.com/riscv/riscv-gnu-toolchain.git "path_to_folder"
./configure --prefix=$RISCV_DIR/install --with-arch=rv32imc --with-abi=ilp32 --disable-linux --disable-gdb
 make -j$(nproc) newlib

The below errors occur multiple times and with any configuration used here  --with-arch=rv32imc or  --with-arch=rv32imcfzfinx

../.././gcc/gcc/c-family/c-cppbuiltin.cc:1674:7: error: ‘ENABLE_DECIMAL_FLOAT’ was not declared in this scope; did you mean ‘MODE_DECIMAL_FLOAT’?
 1674 |   if (ENABLE_DECIMAL_FLOAT && ENABLE_DECIMAL_BID_FORMAT)
      |       ^~~~~~~~~~~~~~~~~~~~
      |       MODE_DECIMAL_FLOAT
make[3]: *** [Makefile:1210: c/c-fold.o] Error 1
../.././gcc/gcc/attribs.h: In function ‘tree_node* lookup_attribute(const char*, const char*, tree)’:
../.././gcc/gcc/attribs.h:249:7: error: ‘CHECKING_P’ was not declared in this scope
  249 |   if (CHECKING_P && attr_name[0] != '_')
      |       ^~~~~~~~~~

./.././gcc/gcc/alloc-pool.h:316:7: error: ‘GATHER_STATISTICS’ was not declared in this scope; did you mean ‘GCC_STATISTICS’?
  316 |   if (GATHER_STATISTICS && !after_memory_report)
      |       ^~~~~~~~~~~~~~~~~
      |       GCC_STATISTICS
make[3]: *** [Makefile:1210: c-family/c-lex.o] Error 1
../.././gcc/gcc/attribs.h: In function ‘tree_node* lookup_attribute(const char*, const char*, tree)’:
../.././gcc/gcc/attribs.h:249:7: error: ‘CHECKING_P’ was not declared in this scope
  249 |   if (CHECKING_P && attr_name[0] != '_')

Please suggest what would be the cause of this issue.

TommyMurphyTM1234, 2025-07-31:
> `git clone --branch 2025.07.03`

What is this?
There's no such branch in this repo:

- https://github.com/riscv-collab/riscv-gnu-toolchain/branches

What happens if you build from the `master` branch?
If it still fails then please do a non parallel build and capture/upload the build log for analysis.

MahimaDesai, 2025-07-31:
[build.log](https://github.com/user-attachments/files/21532740/build.log)

Please refer this build log with respect to the build using master branch and also non parallel build.

git clone --branch 2025.07.03 ---- this here was referring to a particular tag.

(Please also suggest if the current toolchain will support --with-arch=rv32imcfzfinx)

TommyMurphyTM1234, 2025-07-31:
> Multiple errors during building toolchain even when using a particular tag as well **as recent master**
>
> steps used for build are: git clone --branch 2025.07.03 --recursive [...]

Building from the top of tree on Ubuntu 24.04.2 LTS worked fine for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1750
cd issue-1750
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imc --with-abi=ilp32 --disable-linux --disable-gdb
make newlib 2>&1 | tee build.log
```
- [build.log.zip](https://github.com/user-attachments/files/21537783/build.log.zip)

TommyMurphyTM1234, 2025-07-31:
I also tried this:
```
make distclean 
rm -rf installed-tools
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imcf_zfinx --with-abi=ilp32 --disable-linux --disable-gdb
make newlib 2>&1 | tee build.log
```
but it failed as follows:
```
xgcc: error: '-march=rv32imfc_zicsr_zfinx': z*inx conflicts with floating-point extensions
xgcc: error: '-march=rv32imfc_zicsr_zmmul_zfinx_zca_zcf': z*inx conflicts with floating-point extensions
xgcc: error: '-march=rv32imfc_zicsr_zfinx': z*inx conflicts with floating-point extensions
make[2]: *** [../.././gcc/gcc/c/Make-lang.in:152: s-selftest-c] Error 1
rm gfdl.pod gcc.pod gcov-dump.pod gcov-tool.pod fsf-funding.pod gpl.pod cpp.pod gcov.pod lto-dump.pod
make[2]: Leaving directory '/home/user/Downloads/issue-1750/build-gcc-newlib-stage1/gcc'
make[1]: *** [Makefile:4709: all-gcc] Error 2
make[1]: Leaving directory '/home/user/Downloads/issue-1750/build-gcc-newlib-stage1'
make: *** [Makefile:648: stamps/build-gcc-newlib-stage1] Error 2
```

pz9115, 2025-08-01:
Yes, `Zfinx` extension is only using for the no FPRs(floating-point registers) environment, so it conflict with `F` or `D` extension which rely on FPRs. Just using `--with-arch=rv32imcf` or `--with-arch=rv32imczfinx` to avoid this issue.

TommyMurphyTM1234, 2025-08-01:
I also tried doing a recursive clone of the tag mentioned in the original post but it failed:
```
git clone --branch 2025.07.03 --recursive https://github.com/riscv/riscv-gnu-toolchain.git issue-1750
...
Submodule path 'binutils': checked out '815d9a14cbbb3b81843f7566222c87fb22e7255d'
error: Server does not allow request for unadvertised object 935a51f3c66ece357ce0d18f3aa3627a13cef7d5
fatal: Fetched in submodule path 'dejagnu', but it did not contain 935a51f3c66ece357ce0d18f3aa3627a13cef7d5. Direct fetching of that commit failed.
```
See here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1748

If I don't do a recursive clone but let the `Makefile` clone what it needs then the build completes successfully:
```
git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git issue-1750
cd issue-1750
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imc_zfinx --with-abi=ilp32 --disable-linux --disable-gdb
make newlib 2>&1 | tee build.log
```
- [build.log.zip](https://github.com/user-attachments/files/21544209/build.log.zip)

To summarise - I cannot reproduce your issue.
When I get a chance later I'll have a look at your log to see if there's anything obvious.

TommyMurphyTM1234, 2025-08-01:
> When I get a chance later I'll have a look at your log to see if there's anything obvious.

Your log contains this:
```
/bin/bash ../.././gcc/gcc/../move-if-change tmp-omp-device-properties.h \
  omp-device-properties.h
echo timestamp > s-omp-device-properties-h
g++  -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H -fno-PIE -I. -Ic -I../.././gcc/gcc -I../.././gcc/gcc/c -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o c/c-lang.o -MT c/c-lang.o -MMD -MP -MF c/.deps/c-lang.TPo ../.././gcc/gcc/c/c-lang.cc
In file included from ../.././gcc/gcc/hash-table.h:246,
                 from ../.././gcc/gcc/coretypes.h:509,
                 from ../.././gcc/gcc/c/c-lang.cc:23:
../.././gcc/gcc/statistics.h:25:2: error: #error GATHER_STATISTICS must be defined
   25 | #error GATHER_STATISTICS must be defined
      |  ^~~~~
...
```
Mine has this:
```
/bin/bash ../.././gcc/gcc/../move-if-change tmp-omp-device-properties.h \
  omp-device-properties.h
echo timestamp > s-omp-device-properties-h
g++  -fno-PIE -c  -DIN_GCC_FRONTEND -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H -fno-PIE -I. -Ic -I../.././gcc/gcc -I../.././gcc/gcc/c -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o c/c-lang.o -MT c/c-lang.o -MMD -MP -MF c/.deps/c-lang.TPo ../.././gcc/gcc/c/c-lang.cc
g++  -fno-PIE -c  -DIN_GCC_FRONTEND -DIN_GCC_FRONTEND -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H -fno-PIE -I. -Ic-family -I../.././gcc/gcc -I../.././gcc/gcc/c-family -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include -I../.././gcc/gcc/../libcody  -I../.././gcc/gcc/../libdecnumber -I../.././gcc/gcc/../libdecnumber/dpd -I../libdecnumber -I../.././gcc/gcc/../libbacktrace   -o c-family/stub-objc.o -MT c-family/stub-objc.o -MMD -MP -MF c-family/.deps/stub-objc.TPo ../.././gcc/gcc/c-family/stub-objc.cc
```
Perhaps the issue is down to your build OS and/or GCC version?

MahimaDesai, 2025-08-01:
Yes, even i compared the build logs and found that this is the point after which the build fails.
My current environment is WSL2 Ubuntu 22.04.
Since the RISC-V GNU toolchain builds its own GCC, is there any particular gcc version or tag which could work.

TommyMurphyTM1234, 2025-08-01:
WSL2 running Ubuntu 22.04.5 LTS with all available updates also worked for me:
```
cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.5 LTS"

gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

cd ~
git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git issue-1750
cd issue-1750
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imc_zfinx --with-abi=ilp32 --disable-linux --disable-gdb
make newlib 2>&1 | tee build.log
```
- [build.log.zip](https://github.com/user-attachments/files/21550434/build.log.zip)

MahimaDesai, 2025-08-04:
I deleted the complete folder from my Ubuntu and cloned again and tried, it works.
But i have a question, now the toolchain is configured for "imc" extension, if i want to try "imcf" what steps can i follow ?
since the errors in build occurred last time when i tried to change and build for another extension.

TommyMurphyTM1234, 2025-08-04:
> I deleted the complete folder from my Ubuntu and cloned again and tried, it works. 

Great! Glad you got it sorted. I presume that something odd in your Ubuntu installation cased problems earlier.

> But i have a question, now the toolchain is configured for "imc" extension, if i want to try "imcf" what steps can i follow ? 

Any toolchain, regardless of its default `arch/abi`, can generate code for any other `arch/abi` once the relevant `-march=... -mabi...` options are passed at compile time. E.g. a toolchain configured for `rv32imc/ilp32` can compile code for `rv32imcf/ilp32f` if, when compiling the code, you pass `-march=rv32imcf -mabi=ilp32f`. However if your program also needs startup code and libraries linked then you will probably get link time errors. In that case if you want a single multilib (bare metal) toolchain that can compile and link and provide multilibs for `rv32imc/ilp32` and `rv32imcf/ilp32f` then you can do something like `configure ... --with-arch=rv32imc --with-abi=ilp32 --with-multilib-generator="rv32imcf-ilp32f--"`. This configures the toolchain for a default `arch/abi` of `rv32imc/ilp32` but also builds startup code/multilibs for `rv32imcf/ilp32f`.

- https://github.com/riscv-collab/riscv-gnu-toolchain?tab=readme-ov-file#build-with-customized-multi-lib-configure

> since the errors in build occurred last time when i tried to change and build for another extension.

I don't undestand what exactly you mean - please clarify what exactly you did and what errors occurred.

Some errors occurred earlier with `rv32imcf_zfinx` but that's expected because the `Z*inx` extensions cannot coexist with the `F` or `D` extensions as mentioned here:

- https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1750#issuecomment-3141958884

MahimaDesai, 2025-08-06:
I have now started to work on using this multilib generator command and getting the toolchain built
i am also trying to see if the issue can be reproduced
i will provide an update on the same by tomorrow and then you can close the issue

TommyMurphyTM1234, 2025-08-06:
For what it's worth, this worked for me:
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1750
cd issue-1750
./configure --prefix=`pwd`/installed-tools --with-arch=rv32imc --with-abi=ilp32 --with-multilib-generator="rv32imcf-ilp32f--;rv32imczfinx-ilp32--" --disable-gdb --disable-linux
make 2>&1 | tee build.log

# Note that the first line ".;" refers to the toolchain's default arch/abi - in this case rv32imc/ilp32
./bin/riscv32-unknown-elf-gcc -print-multi-lib
.;
rv32imfc_zicsr/ilp32f;@march=rv32imfc_zicsr@mabi=ilp32f
rv32imc_zicsr_zfinx/ilp32;@march=rv32imc_zicsr_zfinx@mabi=ilp32

./bin/riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=./bin/riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/home/user/issue-1750/installed-tools/libexec/gcc/riscv32-unknown-elf/15.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/user/issue-1750/gcc/configure --target=riscv32-unknown-elf --prefix=/home/user/issue-1750/installed-tools --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g1b306039a --with-system-zlib --enable-tls --with-newlib --with-sysroot=/home/user/issue-1750/installed-tools/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --enable-multilib --with-multilib-generator='rv32imcf-ilp32f--;rv32imczfinx-ilp32--' --with-abi=ilp32 --with-arch=rv32imc --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 15.1.0 (g1b306039a)
```
- [build.log.zip](https://github.com/user-attachments/files/21617673/build.log.zip)

MahimaDesai, 2025-08-08:
I tried to reproduce the issue and these are the steps that i did with a fresh ubuntu setup without any previous projects or files

cloned the repo and built with same config as above
git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git "$GCC_DIR"
./configure --prefix=$GCC_DIR/install \
        --with-arch=rv32imc_zicsr_zifencei \
        --with-abi=ilp32 \
        --with-multilib-generator="rv32imc_zicsr_zifencei-ilp32--;rv32imcf_zicsr_zifencei-ilp32f--;rv32imczfinx_zicsr_zifencei-ilp32--" \
        --disable-gdb --disable-linux
make newlib 2>&1 | tee build.log

Added the path to environment variables (bashrc)
This worked.

cleaned the setup by deleting the folder
rm -rf $(GCC_DIR)
repeated the above steps again --- Fails with the macro errors

MahimaDesai, 2025-08-08:
The toolchain is installed in a particular directory. To use the compiler to run programs in different projects, i add the path to environment variables.
Before the clean and build, i also remove the path added in bashrc.

the working build log is the first build
[working_build.log](https://github.com/user-attachments/files/21684624/working_build.log)

Failed log 
[build.log](https://github.com/user-attachments/files/21684654/build.log)

TommyMurphyTM1234, 2025-08-13:
Why are you doing this at all?
It seems pointless.

> ```
> echo "export PATH=$GCC_DIR/install/bin:\$PATH" >> ~/.bashrc
> source ~/.bashrc
> ```

> ```
> sed -i "\|$GCC_DIR/install/bin|d" ~/.bashrc
> source ~/.bashrc
> ```

> ```
> echo "export PATH=$GCC_DIR/install/bin:\$PATH" >> ~/.bashrc
> source ~/.bashrc
> ```

The middle commands are probably not going to clear the toolchain from your `PATH` so maybe that's the root of your problem?
If you don't mess with the `PATH` and `.bashrc` do you still get the problem?

> (Probably to clear the set environment variables and apply them)

I don't know what this means - please clarify.

TommyMurphyTM1234, 2025-08-13:
> The middle commands are probably not going to clear the toolchain from your `PATH` 

Just to confirm - that is the case. Your sequence of commands above will leave `.../riscv-gnu-toolchain/install/bin` on your `PATH` unless you open a new terminal session after changing `.bashrc`.

MahimaDesai, 2025-08-13:
I was trying to temporarily remove and then re-add the toolchain path in order to refresh the environment variables without restarting the terminal. 
My thought was that by removing the $GCC_DIR/install/bin entry from ~/.bashrc and then adding it back, I could ensure that there were no duplicate entries or incorrect paths causing the issue. 

But I understand your point — simply editing .bashrc might not fully clear the current session’s environment, so the problem might still persist even after doing this.

Now i was able to make this work only by restarting the WSL session after removing the environment variables and run the build again and it works.

MahimaDesai, 2025-08-13:
If you refer my steps above 
i don't do "export PATH=$GCC_DIR/install/bin:$PATH" before build but i set this after build 
and if i clean the setup it removes the environment variables from bashrc but doesn't actually clean them and instead i had restart the wsl and build again

MahimaDesai, 2025-08-13:
in a new terminal session 

cd ~
## GCC_DIR=$PWD/riscv-gnu-toolchain ## Referring to the folder only

git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git "$GCC_DIR"
cd "$GCC_DIR"
./configure --prefix=$GCC_DIR/install \
  --with-arch=rv32imc_zicsr_zifencei \
  --with-abi=ilp32 \
  --with-multilib-generator="rv32imc_zicsr_zifencei-ilp32--;rv32imcf_zicsr_zifencei-ilp32f--;rv32imczfinx_zicsr_zifencei-ilp32--" \
  --disable-gdb --disable-linux
make newlib -j$(nproc) 2>&1 | tee build.log

export PATH=$GCC_DIR/install/bin:$PATH

TommyMurphyTM1234, 2025-08-13:
I still don't understand why you're messing with the `PATH` and/or `.bashrc` at all. As I asked before, if you **don't** do this then does the build always work? If so then there's no valid issue here with the `riscv-gnu-toolchain` repo and build process. If the build only fails if you (unnecessarily?) mess with the environment then that's outside the control/scope of this repo.

TommyMurphyTM1234, 2025-08-14:
> ```
> export GCC_DIR=$PWD/riscv-gnu-toolchain
> 
> git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git "$GCC_DIR"
> cd "$GCC_DIR"
> ./configure --prefix=$GCC_DIR/install \
>   --with-arch=rv32imc_zicsr_zifencei \
>   --with-abi=ilp32 \
>   --with-multilib-generator="rv32imc_zicsr_zifencei-ilp32--;rv32imcf_zicsr_zifencei-ilp32f--;rv32imczfinx_zicsr_zifencei-ilp32--" \
>   --disable-gdb --disable-linux
> make newlib -j$(nproc) 2>&1 | tee build.log
> 
> echo "export PATH=$GCC_DIR/install/bin:\$PATH" >> ~/.bashrc
> source ~/.bashrc
> ```
> 
> Cleaning the setup
> 
> ```
> rm -rf "$GCC_DIR"
> sed -i "\|$GCC_DIR/install/bin|d" ~/.bashrc
> source ~/.bashrc
> ```
> 
> ```
> export GCC_DIR=$PWD/riscv-gnu-toolchain
> 
> git clone --branch 2025.07.03 https://github.com/riscv/riscv-gnu-toolchain.git "$GCC_DIR"
> cd "$GCC_DIR"
> ./configure --prefix=$GCC_DIR/install \
>   --with-arch=rv32imc_zicsr_zifencei \
>   --with-abi=ilp32 \
>   --with-multilib-generator="rv32imc_zicsr_zifencei-ilp32--;rv32imcf_zicsr_zifencei-ilp32f--;rv32imczfinx_zicsr_zifencei-ilp32--" \
>   --disable-gdb --disable-linux
> make newlib -j$(nproc) 2>&1 | tee build_rebuild.log
> 
> echo "export PATH=$GCC_DIR/install/bin:\$PATH" >> ~/.bashrc
> source ~/.bashrc
> ```

I cannot reproduce your issue using the command sequences above. 
The second build completes fine just as the first one did.