riscv-collab/riscv-gnu-toolchain #1759: Incorrect march names when building toolchain with `--with-multilib-generator` option

por3bski, 2025-08-26:
Building toolchain with following custom multilib configuration results in unexpected march names:

`./configure --prefix=/opt/riscv --with-multilib-generator="rv32imac-ilp32--;rv64imac-lp64--"`

```
riscv64-unknown-elf-gcc --print-multi-lib
.;
rv32imac_zaamo_zalrsc/ilp32;@march=rv32imac_zaamo_zalrsc@mabi=ilp32
rv64imac_zaamo_zalrsc/lp64;@march=rv64imac_zaamo_zalrsc@mabi=lp64
```

Each march has `_zaamo_zalrsc` suffix.

When compiling with default multilib configuration, names are as I would expect:

`./configure --prefix=/opt/riscv --enable-multilib`

```
riscv64-unknown-elf-gcc --print-multi-lib
.;
rv32i/ilp32;@march=rv32i@mabi=ilp32
rv32im/ilp32;@march=rv32im@mabi=ilp32
rv32iac/ilp32;@march=rv32iac@mabi=ilp32
rv32imac/ilp32;@march=rv32imac@mabi=ilp32
rv32imafc/ilp32f;@march=rv32imafc@mabi=ilp32f
rv64imac/lp64;@march=rv64imac@mabi=lp64
rv64imafdc/lp64d;@march=rv64imafdc@mabi=lp64d 
```

Are these differences correct behavior?

pz9115, 2025-08-27:
Hi @por3bski, I think the `--enable-multilib` was missing adding imply relationship for `a` extension, I will  update this with a new pr, thanks for you report it.

pz9115, 2025-08-27:
> > The current  option is more secure, since old version gcc can not recognize the  extension, so just ignore this difference.`--enable-multilib``zaamo&zalrsc`
> 
> Just to clarify, the above means that the following is true?
> 
> > Are these differences correct behavior?

Yes, I think, actually now `a` equals to `a+zaamo+zalrsc`.

TommyMurphyTM1234, 2025-08-27:
Thanks @pz9115 - is there any link to information about this change?

pz9115, 2025-08-28:
> Thanks [@pz9115](https://github.com/pz9115) - is there any link to information about this change?

Here are some refer links——

Spec: https://github.com/riscv/riscv-zaamo-zalrsc

ISA manual PR: https://github.com/riscv/riscv-isa-manual/pull/1304