riscv-collab/riscv-gnu-toolchain #1761: Build Fails (/bin/as: unrecognized option '--64')

www-asics-ws, 2025-08-27:
I've tried the --enable-multilib and also --with-arch=rv32gc --with-abi=ilp32d, each time the build fails with exactly the smae message.

Linux cpu11 6.8.0-65-generic #68~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue Jul 15 18:06:34 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

Ubuntu 22.04

HEADERS="options.h insn-constants.h config/elfos.h config/gnu-user.h config/linux.h config/glibc-stdint.h config/riscv/riscv.h config/riscv/linux.h config/initfini-array.h defaults.h" DEFINES="LIBC_GLIBC=1 LIBC_UCLIBC=2 LIBC_BIONIC=3 LIBC_MUSL=4 DEFAULT_LIBC=LIBC_GLIBC ANDROID_DEFAULT=0 HEAP_TRAMPOLINES_INIT=0 TARGET_DEFAULT_ASYNC_UNWIND_TABLES=1 TARGET_DEFAULT_ISA_SPEC=ISA_SPEC_CLASS_20191213 TARGET_RISCV_ATTRIBUTE=1 TARGET_RISCV_DEFAULT_ARCH=rv32imafdc_zicsr_zifencei_zaamo_zalrsc TARGET_RISCV_DEFAULT_ABI=ilp32d TARGET_DEFAULT_CMODEL=CM_MEDLOW" \
/bin/sh ../.././gcc/gcc/mkconfig.sh tm.h
TARGET_CPU_DEFAULT="" \
HEADERS="config/riscv/riscv-protos.h config/linux-protos.h tm-preds.h" DEFINES="" \
/bin/sh ../.././gcc/gcc/mkconfig.sh tm_p.h
TARGET_CPU_DEFAULT="" \
HEADERS="auto-host.h ansidecl.h" DEFINES="" \
/bin/sh ../.././gcc/gcc/mkconfig.sh bconfig.h
g++ -c   -g -O2   -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-error=narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -DHAVE_CONFIG_H  -DGENERATOR_FILE -I. -Ibuild -I../.././gcc/gcc -I../.././gcc/gcc/build -I../.././gcc/gcc/../include  -I../.././gcc/gcc/../libcpp/include  \
        -o build/genmodes.o ../.././gcc/gcc/genmodes.cc
/tools/risc_v_gcc/riscv32-unknown-linux-gnu/bin/as: unrecognized option '--64'
make[2]: *** [Makefile:3192: build/genmodes.o] Error 1
make[2]: Leaving directory '/share/risc-v-gcc/riscv-gnu-toolchain/build-gcc-linux-stage1/gcc'
make[1]: *** [Makefile:4693: all-gcc] Error 2
make[1]: Leaving directory '/share/risc-v-gcc/riscv-gnu-toolchain/build-gcc-linux-stage1'
make: *** [Makefile:469: stamps/build-gcc-linux-stage1] Error 2

TommyMurphyTM1234, 2025-08-27:
Please provide the full set of commands needed to reproduce this and capture, compress and attach the build log for the failed build. 

If possible please try it from scratch using the following commands on a clean install of Ubuntu 22.04 (physical machine or virtual machine or container etc.).
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1761
cd issue-1761
./configure --prefix=`pwd`/installed-tools --enable-multilib
make linux 2>&1 | tee build.log
```

www-asics-ws, 2025-08-27:
All was from scratch.
1. git clone
2. config
3. make

after error removed everything and repeated ALL 3 steps ...

TommyMurphyTM1234, 2025-08-27:
> Attached is the build log.

Where?

www-asics-ws, 2025-08-27:
> TommyMurphyTM1234 left a comment (riscv-collab/riscv-gnu-
> toolchain#1761)

It was attached to the email. Here it is again

www-asics-ws, 2025-08-27:
[build.log.gz](https://github.com/user-attachments/files/22006246/build.log.gz)

Apparently attaching to email doesn't work. Here it is ...

www-asics-ws, 2025-08-27:
Thank you very much for looking into it, Tommy. I installed the prebuilt version from Aug. 8. That should do the trick for me !

Best Regards,
rudi

closed 2025-08-27 by www-asics-ws

TommyMurphyTM1234, 2025-08-27:
> My build on Ubuntu 22.04.5 LTS with all available updates installed is still ongoing but gets past that point of the build without error:

Just for completeness here is my final (successful) build log: [build.log.zip](https://github.com/user-attachments/files/22011209/build.log.zip)
```
git clone https://github.com/riscv-collab/riscv-gnu-toolchain issue-1761
cd issue-1761
./configure --prefix=`pwd`/installed-tools --enable-multilib
make linux 2>&1 | tee build.log
```