riscv-collab/riscv-gnu-toolchain #1762: objdump crashes by signal SIGABRT with some DSO

onlycparra, 2025-08-31:
Hi, I am working on a tiny program to teach about assembling, linking and loading.

This is how I configured/installed the toolchain:
```
./configure --prefix=/opt/riscv --with-arch=rv64g --with-abi=lp64 --enable-qemu-system
make linux
```

I have this source:
```
// mymath.h
#ifndef MYMATH_H_
#define MYMATH_H_
extern int myglobal_SUB;
int mymath_add(int, int);
int mymath_sub(int, int);
#endif
```
```
// mysecret.c
#include "mymath.h"
static char mysecret[] = "abcdefghijklmnopqrstuvwxyz";
int mysecret_LEN = sizeof(mysecret) - 1;

char get_secret(int x, int y){
    int i = mymath_add(x,y) + myglobal_SUB;
    return mysecret[ i % mysecret_LEN];
}
```

I compile and link a shared library:
```
rv-gcc -Wall -Werror -O2 -save-temps=obj -march=rv64im -mabi=lp64 -MD -ffreestanding -fno-builtin -fno-stack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fpic -c mysecret.c
rv-gcc -shared -nostdlib -nodefaultlibs -nostartfiles -o libmysecret.so mysecret.o
```
(`rv-gcc` is a link to `/opt/riscv/bin/riscv64-unknown-linux-gnu-gcc`)

Finally, I try to `objdump` the newly created `libmysecret.so`, but it crashes with `SIGABRT`:
```
rv-objdump -D libmysecret.so

libmysecret.so:     file format elf64-littleriscv

Disassembly of section .hash:

0000000000000190 <.hash>:
 190:	00000003          	lb	zero,0(zero) # 0 <_PROCEDURE_LINKAGE_TABLE_-0x300>
 194:	0006                	.insn	2, 0x0006
 196:	0000                	.insn	2, 0x0000
 198:	0002                	.insn	2, 0x0002
 19a:	0000                	.insn	2, 0x0000
 19c:	0005                	.insn	2, 0x0005
	...
 1ae:	0000                	.insn	2, 0x0000
 1b0:	0004                	.insn	2, 0x0004
 1b2:	0000                	.insn	2, 0x0000
 1b4:	0000                	.insn	2, 0x0000
 1b6:	0000                	.insn	2, 0x0000
 1b8:	00000003          	lb	zero,0(zero) # 0 <_PROCEDURE_LINKAGE_TABLE_-0x300>

Disassembly of section .gnu.hash:

00000000000001c0 <.gnu.hash>:
 1c0:	0002                	.insn	2, 0x0002
 1c2:	0000                	.insn	2, 0x0000
 1c4:	0004                	.insn	2, 0x0004
 1c6:	0000                	.insn	2, 0x0000
 1c8:	0001                	.insn	2, 0x0001
 1ca:	0000                	.insn	2, 0x0000
 1cc:	0006                	.insn	2, 0x0006
 1ce:	0000                	.insn	2, 0x0000
 1d0:	0050                	.insn	2, 0x0050
 1d2:	0000                	.insn	2, 0x0000
 1d4:	8400                	.insn	2, 0x8400
 1d6:	0000                	.insn	2, 0x0000
 1d8:	0004                	.insn	2, 0x0004
 1da:	0000                	.insn	2, 0x0000
 1dc:	0005                	.insn	2, 0x0005
 1de:	0000                	.insn	2, 0x0000
 1e0:	7d5501ab          	.insn	4, 0x7d5501ab
 1e4:	97d0e12f          	.insn	4, 0x97d0e12f

Disassembly of section .dynsym:

00000000000001e8 <.dynsym>:
	...
 204:	00080003          	lb	zero,0(a6)
 208:	0330                	.insn	2, 0x0330
	...
 216:	0000                	.insn	2, 0x0000
 218:	00000017          	auipc	zero,0x0
 21c:	0010                	.insn	2, 0x0010
	...
 22e:	0000                	.insn	2, 0x0000
 230:	000c                	.insn	2, 0x000c
 232:	0000                	.insn	2, 0x0000
 234:	0010                	.insn	2, 0x0010
	...
 246:	0000                	.insn	2, 0x0000
 248:	0001                	.insn	2, 0x0001
 24a:	0000                	.insn	2, 0x0000
 24c:	0012                	.insn	2, 0x0012
 24e:	0008                	.insn	2, 0x0008
 250:	0330                	.insn	2, 0x0330
 252:	0000                	.insn	2, 0x0000
 254:	0000                	.insn	2, 0x0000
 256:	0000                	.insn	2, 0x0000
 258:	0048                	.insn	2, 0x0048
 25a:	0000                	.insn	2, 0x0000
 25c:	0000                	.insn	2, 0x0000
 25e:	0000                	.insn	2, 0x0000
 260:	0024                	.insn	2, 0x0024
 262:	0000                	.insn	2, 0x0000
 264:	0011                	.insn	2, 0x0011
 266:	000d                	.insn	2, 0x000d
 268:	2008                	.insn	2, 0x2008
 26a:	0000                	.insn	2, 0x0000
 26c:	0000                	.insn	2, 0x0000
 26e:	0000                	.insn	2, 0x0000
 270:	0004                	.insn	2, 0x0004
 272:	0000                	.insn	2, 0x0000
 274:	0000                	.insn	2, 0x0000
	...

Disassembly of section .dynstr:

0000000000000278 <.dynstr>:
 278:	6700                	.insn	2, 0x6700
 27a:	7465                	.insn	2, 0x7465
 27c:	735f 6365 6572      	.insn	6, 0x65726365735f
 282:	0074                	.insn	2, 0x0074
 284:	796d                	.insn	2, 0x796d
 286:	616d                	.insn	2, 0x616d
 288:	6874                	.insn	2, 0x6874
 28a:	615f 6464 6d00      	.insn	6, 0x6d006464615f
 290:	6779                	.insn	2, 0x6779
 292:	6f6c                	.insn	2, 0x6f6c
 294:	6162                	.insn	2, 0x6162
 296:	5f6c                	.insn	2, 0x5f6c
 298:	00425553          	.insn	4, 0x00425553
 29c:	796d                	.insn	2, 0x796d
 29e:	72636573          	.insn	4, 0x72636573
 2a2:	7465                	.insn	2, 0x7465
fish: Job 1, 'rv-objdump -D libmysecret.so' terminated by signal SIGABRT (Abort)
```
(`rv-objdump` is a link to `/opt/riscv/bin/riscv64-unknown-linux-gnu-objdump`)

I am using Fedora on x86 if that is useful.

onlycparra, 2025-08-31:
nevermind... I was linking the wrong objdump... 

I did this **WRONG**:
```
ln -s ../riscv64-unknown-linux-gnu/bin/objdump /opt/riscv/bin/rv-objdump
```

I should have done:
```
ln -s riscv64-unknown-linux-gnu-objdump /opt/riscv/bin/rv-objdump
```

closed 2025-08-31 by onlycparra

onlycparra, 2025-08-31:
I notice that if I specify the SONAME...
```
rv-gcc -shared -nostdlib -nodefaultlibs -nostartfiles -Wl,-soname,libmysecret.so.1 -o libmysecret.so mysecret.o
```

then it does not fail.
```
rv-objdump -D libmysecret.so

libmysecret.so:     file format elf64-littleriscv

Disassembly of section .hash:

0000000000000190 <.hash>:
 190:	00000003          	lb	zero,0(zero) # 0 <_PROCEDURE_LINKAGE_TABLE_-0x310>
 194:	0006                	.insn	2, 0x0006
 196:	0000                	.insn	2, 0x0000
 198:	0002                	.insn	2, 0x0002
 19a:	0000                	.insn	2, 0x0000
 19c:	0005                	.insn	2, 0x0005
	...
 1ae:	0000                	.insn	2, 0x0000
 1b0:	0004                	.insn	2, 0x0004
 1b2:	0000                	.insn	2, 0x0000
 1b4:	0000                	.insn	2, 0x0000
 1b6:	0000                	.insn	2, 0x0000
 1b8:	00000003          	lb	zero,0(zero) # 0 <_PROCEDURE_LINKAGE_TABLE_-0x310>

Disassembly of section .gnu.hash:

00000000000001c0 <.gnu.hash>:
 1c0:	0002                	.insn	2, 0x0002
 1c2:	0000                	.insn	2, 0x0000
 1c4:	0004                	.insn	2, 0x0004
 1c6:	0000                	.insn	2, 0x0000
 1c8:	0001                	.insn	2, 0x0001
 1ca:	0000                	.insn	2, 0x0000
 1cc:	0006                	.insn	2, 0x0006
 1ce:	0000                	.insn	2, 0x0000
 1d0:	0050                	.insn	2, 0x0050
 1d2:	0000                	.insn	2, 0x0000
 1d4:	8400                	.insn	2, 0x8400
 1d6:	0000                	.insn	2, 0x0000
 1d8:	0004                	.insn	2, 0x0004
 1da:	0000                	.insn	2, 0x0000
 1dc:	0005                	.insn	2, 0x0005
 1de:	0000                	.insn	2, 0x0000
 1e0:	7d5501ab          	.insn	4, 0x7d5501ab
 1e4:	97d0e12f          	.insn	4, 0x97d0e12f

Disassembly of section .dynsym:

00000000000001e8 <.dynsym>:
	...
 204:	00080003          	lb	zero,0(a6)
 208:	0340                	.insn	2, 0x0340
	...
 216:	0000                	.insn	2, 0x0000
 218:	00000017          	auipc	zero,0x0
 21c:	0010                	.insn	2, 0x0010
	...
 22e:	0000                	.insn	2, 0x0000
 230:	000c                	.insn	2, 0x000c
 232:	0000                	.insn	2, 0x0000
 234:	0010                	.insn	2, 0x0010
	...
 246:	0000                	.insn	2, 0x0000
 248:	0001                	.insn	2, 0x0001
 24a:	0000                	.insn	2, 0x0000
 24c:	0012                	.insn	2, 0x0012
[... 270 lines cut ...]
  12:	3531                	.insn	2, 0x3531
  14:	312e                	.insn	2, 0x312e
  16:	302e                	.insn	2, 0x302e
	...

Disassembly of section .riscv.attributes:

0000000000000000 <.riscv.attributes>:
   0:	2941                	.insn	2, 0x2941
   2:	0000                	.insn	2, 0x0000
   4:	7200                	.insn	2, 0x7200
   6:	7369                	.insn	2, 0x7369
   8:	01007663          	bgeu	zero,a6,14 <_PROCEDURE_LINKAGE_TABLE_-0x2fc>
   c:	001f 0000 1004      	.insn	6, 0x10040000001f
  12:	7205                	.insn	2, 0x7205
  14:	3676                	.insn	2, 0x3676
  16:	6934                	.insn	2, 0x6934
  18:	7032                	.insn	2, 0x7032
  1a:	5f31                	.insn	2, 0x5f31
  1c:	326d                	.insn	2, 0x326d
  1e:	3070                	.insn	2, 0x3070
  20:	7a5f 6d6d 6c75      	.insn	6, 0x6c756d6d7a5f
  26:	7031                	.insn	2, 0x7031
  28:	0030                	.insn	2, 0x0030
```

cmuellner, 2025-12-18:
This seems to have gone under the radar. Sorry for that.

I tried to reproduce this with a fresh build with the provided instructions (on Fedora 43 / x86_64):
```
rm -rf /opt/riscv-rvi/*
./configure --prefix=/opt/riscv-rvi --with-arch=rv64g --with-abi=lp64 --with-sim=qemu --enable-qemu-system --enable-linux
make -j$(nproc)
make -j$(nproc) build-sim
```

When building the shared library, I can't see the abort:
```
++ rm -rf mysecret.o libmysecret.so libmysecret.so.txt mysecret.i mysecret.d
++ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-gcc -Wall -Werror -O2 -save-temps=obj -march=rv64im -mabi=lp64 -MD -ffreestanding -fno-builtin -fno-stack-protector -fno-unwind-tables -fno-asynchronous-unwind-tables -fpic -c mysecret.c
++ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-gcc -shared -nostdlib -nodefaultlibs -nostartfiles -o libmysecret.so mysecret.o
++ /opt/riscv-rvi/bin/riscv64-unknown-linux-gnu-objdump -D libmysecret.so
++ echo 0
0
```

Last time we updated Binutils was end of July.
You reported end of August.
So, there were no changes to Binutils in between.

@onlycparra, can you still reproduce this with a fresh toolchain build?