riscv-collab/riscv-gnu-toolchain #1763: `O2` flag generates segmentation fault in my code

Joao-Pedro-Cabral, 2025-09-03:
Hello,

I'm using this function from [PQClean](https://github.com/PQClean/PQClean):
```
void
PQCLEAN_FALCON512_CLEAN_prng_init(prng *p, inner_shake256_context *src) {
    /*
     * To ensure reproducibility for a given seed, we
     * must enforce little-endian interpretation of
     * the state words.
     */
    uint8_t tmp[56];
    uint64_t th, tl;
    int i;

    uint32_t *d32 = (uint32_t *) p->state.d;
    uint64_t *d64 = (uint64_t *) p->state.d;

    inner_shake256_extract(src, tmp, 56);
    for (i = 0; i < 14; i ++) {
        uint32_t w;

        w = (uint32_t)tmp[(i << 2) + 0]
            | ((uint32_t)tmp[(i << 2) + 1] << 8)
            | ((uint32_t)tmp[(i << 2) + 2] << 16)
            | ((uint32_t)tmp[(i << 2) + 3] << 24);
        d32[i] = w;
    }
    tl = d32[48 / sizeof(uint32_t)];
     th = d32[52 / sizeof(uint32_t)];
    d64[48 / sizeof(uint64_t)] = tl + (th << 32);
    PQCLEAN_FALCON512_CLEAN_prng_refill(p);
}
```

When running my code in my x86_64 pc (with both `O2` and `O0` flags) everything works fine.

However, when using `O2` flag and running the code in [gem5](https://www.gem5.org/) simulator,  I got a segmentation fault. 
First, I noted that with `O2` flag the `for (i = 0; i < 14; i ++)` is auto-vectorized.
After, I used `remote gdb` in `gem5` to debug my code. Then, I discover that the segmentation fault was caused by this line:
```22768:	02b70187          	vlm.v	v3,(a4)```
This `a4` comes from:
```2275c:	8ec18713          	addi	a4,gp,-1812 # 390ec <__TMC_END__+0x4>```

Additionally, I tried another code with the same `vlm.v` instruction and the same `vtype` and `vl` configurations in `gem5`  (but with a different memory address) and no segmentation fault happened (only to verify if it's not a internal `gem5` error).

What do you think about this issue? If you need any more information from my setup/code, feel free to ask me (I can share anything from this code).

`O0` disassembly code from this function:
```
000000000002f430 <PQCLEAN_FALCON512_CLEAN_prng_init>:
   2f430:	7175                	addi	sp,sp,-144
   2f432:	e506                	sd	ra,136(sp)
   2f434:	e122                	sd	s0,128(sp)
   2f436:	0900                	addi	s0,sp,144
   2f438:	f6a43c23          	sd	a0,-136(s0)
   2f43c:	f6b43823          	sd	a1,-144(s0)
   2f440:	f7843783          	ld	a5,-136(s0)
   2f444:	20878793          	addi	a5,a5,520
   2f448:	fef43023          	sd	a5,-32(s0)
   2f44c:	f7843783          	ld	a5,-136(s0)
   2f450:	20878793          	addi	a5,a5,520
   2f454:	fcf43c23          	sd	a5,-40(s0)
   2f458:	f8840793          	addi	a5,s0,-120
   2f45c:	f7043603          	ld	a2,-144(s0)
   2f460:	03800593          	li	a1,56
   2f464:	853e                	mv	a0,a5
   2f466:	963e40ef          	jal	13dc8 <shake256_inc_squeeze>
   2f46a:	fe042623          	sw	zero,-20(s0)
   2f46e:	a045                	j	2f50e <PQCLEAN_FALCON512_CLEAN_prng_init+0xde>
[... 181 lines cut ...]
   2289c:	0086161b          	slliw	a2,a2,0x8
   228a0:	03c14503          	lbu	a0,60(sp)
   228a4:	8e5d                	or	a2,a2,a5
   228a6:	03f14783          	lbu	a5,63(sp)
   228aa:	0105959b          	slliw	a1,a1,0x10
   228ae:	0086969b          	slliw	a3,a3,0x8
   228b2:	01066633          	or	a2,a2,a6
   228b6:	8ecd                	or	a3,a3,a1
   228b8:	0187171b          	slliw	a4,a4,0x18
   228bc:	8ec9                	or	a3,a3,a0
   228be:	8f51                	or	a4,a4,a2
   228c0:	0187979b          	slliw	a5,a5,0x18
   228c4:	8fd5                	or	a5,a5,a3
   228c6:	1702                	slli	a4,a4,0x20
   228c8:	9301                	srli	a4,a4,0x20
   228ca:	1782                	slli	a5,a5,0x20
   228cc:	8fd9                	or	a5,a5,a4
   228ce:	22f43c23          	sd	a5,568(s0)
   228d2:	8522                	mv	a0,s0
   228d4:	969ff0ef          	jal	2223c <PQCLEAN_FALCON512_CLEAN_prng_refill>
   228d8:	60a6                	ld	ra,72(sp)
   228da:	6406                	ld	s0,64(sp)
   228dc:	6161                	addi	sp,sp,80
   228de:	8082                	ret
```

Joao-Pedro-Cabral, 2025-09-03:
I've created a simpler version of the previous C code, now only simple 20 lines of C code are necessary to generate the issue.
Please, do a pull in my repository.

TommyMurphyTM1234, 2025-09-05:
Sorry - I know nothing about Gem5 so can't get it running in order to reproduce the issue.
If you can advise on exactly how to reproduce the issue then please do so.

Are you sure that it's not a case of Gem5 crashing and it potentially being an issue with the simulator rather than the toolchain/generated code?
Maybe the issue needs to be reported in the Gem5 project?

Joao-Pedro-Cabral, 2025-09-05:
Hello,

I think you can compile my code and run in any simulator (with linux kernel support).

However, now I'm thinking that the problem could be in `gem5`, because now I have a RVV Intrinsics code which generates a segmentation fault, but if I remove the `vslidedown` instruction from my code, the segmentation fault disappears. (I checked the generated assembly code and in this case looks fine).
This new problem I reported in `gem5` project.
Probably, these two problems can be the same, because when `gcc` autovectorizes my code, it adds the  `vslideup` instruction.

But, if you can compile and run my C code in other simulator (to check that's not a gcc problem), I'll be very grateful (since I don't have other simulator setup).